JvanKatwijk/qt-dab #192: Error Linking CXX executable dabMini-2.0

computer-wizard, 2020-12-13:
Error linking dabMini-2.0, see end of the results of 'make' below.
make
[  0%] Automatic RCC for resources.qrc
[  0%] Generating ui_dab-mini.h
[  1%] Generating __/moc_dab-processor.cpp
[  2%] Generating __/includes/output/moc_audio-base.cpp
[  2%] Generating __/includes/output/moc_audiosink.cpp
[  3%] Generating __/includes/ofdm/moc_sample-reader.cpp
[  4%] Generating __/includes/ofdm/moc_ofdm-decoder.cpp
[  4%] Generating __/includes/ofdm/moc_phasereference.cpp
[  5%] Generating __/includes/ofdm/moc_fib-decoder.cpp
[  6%] Generating __/includes/ofdm/moc_fic-handler.cpp
[  6%] Generating __/includes/ofdm/moc_tii_detector.cpp
/home/boudewijn/DATV/DAB/qt-dab/includes/ofdm/tii_detector.h:0: Note: No relevant classes found. No output generated.
[  7%] Generating __/includes/backend/moc_msc-handler.cpp
/home/boudewijn/DATV/DAB/qt-dab/includes/backend/msc-handler.h:0: Note: No relevant classes found. No output generated.
[  8%] Generating __/includes/backend/moc_backend.cpp
/home/boudewijn/DATV/DAB/qt-dab/includes/backend/backend.h:0: Note: No relevant classes found. No output generated.
[  9%] Generating __/includes/backend/audio/moc_mp2processor.cpp
[  9%] Generating __/includes/backend/audio/moc_mp4processor.cpp
[ 10%] Generating __/includes/backend/data/moc_virtual-datahandler.cpp
[ 11%] Generating __/includes/backend/data/moc_pad-handler.cpp
[ 11%] Generating __/includes/backend/data/mot/moc_mot-handler.cpp
/home/boudewijn/DATV/DAB/qt-dab/includes/backend/data/mot/mot-handler.h:0: Note: No relevant classes found. No output generated.
[ 12%] Generating __/includes/backend/data/mot/moc_mot-object.cpp
[ 13%] Generating __/includes/backend/data/mot/moc_mot-dir.cpp
/home/boudewijn/DATV/DAB/qt-dab/includes/backend/data/mot/mot-dir.h:0: Note: No relevant classes found. No output generated.
[ 13%] Generating __/includes/backend/data/moc_ip-datahandler.cpp
[ 14%] Generating __/includes/backend/data/moc_tdc-datahandler.cpp
[ 15%] Generating __/includes/backend/data/moc_journaline-datahandler.cpp
/home/boudewijn/DATV/DAB/qt-dab/includes/backend/data/journaline-datahandler.h:0: Note: No relevant classes found. No output generated.
[ 16%] Generating __/includes/backend/data/moc_data-processor.cpp
[ 16%] Generating __/includes/support/moc_preset-handler.cpp
[ 17%] Generating __/includes/support/moc_presetcombobox.cpp
[ 18%] Generating __/includes/support/moc_smallqlistview.cpp
[ 18%] Generating __/includes/support/moc_smallpushbutton.cpp
[ 19%] Generating __/includes/support/moc_verysmallpushbutton.cpp
[ 20%] Generating __/includes/support/moc_smallcombobox.cpp
[ 20%] Generating __/includes/support/moc_smallspinbox.cpp
[ 21%] Generating moc_radio.cpp
[ 22%] Generating __/includes/backend/audio/moc_faad-decoder.cpp
[ 23%] Generating devices-dab-mini/sdrplay-handler/moc_sdrplay-handler.cpp
[ 23%] Generating devices-dab-mini/sdrplay-handler/moc_sdrplayselect.cpp
[ 24%] Generating devices-dab-mini/airspy-handler/moc_airspy-handler.cpp
[ 25%] Generating devices-dab-mini/rtlsdr-handler/moc_rtlsdr-handler.cpp
Scanning dependencies of target dabMini-2.0
[ 26%] Building CXX object CMakeFiles/dabMini-2.0.dir/main.cpp.o
[ 27%] Building CXX object CMakeFiles/dabMini-2.0.dir/home/boudewijn/DATV/DAB/qt-dab/dab-processor.cpp.o
[ 27%] Building CXX object CMakeFiles/dabMini-2.0.dir/home/boudewijn/DATV/DAB/qt-dab/src/ofdm/sample-reader.cpp.o
[ 28%] Building CXX object CMakeFiles/dabMini-2.0.dir/home/boudewijn/DATV/DAB/qt-dab/src/ofdm/ofdm-decoder.cpp.o
[ 29%] Building CXX object CMakeFiles/dabMini-2.0.dir/home/boudewijn/DATV/DAB/qt-dab/src/ofdm/phasereference.cpp.o
[ 30%] Building CXX object CMakeFiles/dabMini-2.0.dir/home/boudewijn/DATV/DAB/qt-dab/src/ofdm/phasetable.cpp.o
[ 30%] Building CXX object CMakeFiles/dabMini-2.0.dir/home/boudewijn/DATV/DAB/qt-dab/src/ofdm/freq-interleaver.cpp.o
[ 31%] Building CXX object CMakeFiles/dabMini-2.0.dir/home/boudewijn/DATV/DAB/qt-dab/src/ofdm/fib-decoder.cpp.o
/home/boudewijn/DATV/DAB/qt-dab/src/ofdm/fib-decoder.cpp: In member function ‘std::vector<serviceId> fibDecoder::insert(std::vector<serviceId>, serviceId, int)’:
/home/boudewijn/DATV/DAB/qt-dab/src/ofdm/fib-decoder.cpp:1516:23: warning: comparison of integer expressions of different signedness: ‘int’ and ‘uint32_t’ {aka ‘unsigned int’} [-Wsign-compare]
 1516 |               ((baseN < n. SId) && (n. SId <= serv. SId)):
      |                 ~~~~~~^~~~~~~~
[ 32%] Building CXX object CMakeFiles/dabMini-2.0.dir/home/boudewijn/DATV/DAB/qt-dab/src/ofdm/fic-handler.cpp.o
[ 32%] Building CXX object CMakeFiles/dabMini-2.0.dir/home/boudewijn/DATV/DAB/qt-dab/src/ofdm/tii_detector.cpp.o
[ 33%] Building CXX object CMakeFiles/dabMini-2.0.dir/home/boudewijn/DATV/DAB/qt-dab/src/ofdm/timesyncer.cpp.o
[ 34%] Building CXX object CMakeFiles/dabMini-2.0.dir/home/boudewijn/DATV/DAB/qt-dab/src/protection/protTables.cpp.o
[ 34%] Building CXX object CMakeFiles/dabMini-2.0.dir/home/boudewijn/DATV/DAB/qt-dab/src/protection/protection.cpp.o
[ 35%] Building CXX object CMakeFiles/dabMini-2.0.dir/home/boudewijn/DATV/DAB/qt-dab/src/protection/eep-protection.cpp.o
[ 36%] Building CXX object CMakeFiles/dabMini-2.0.dir/home/boudewijn/DATV/DAB/qt-dab/src/protection/uep-protection.cpp.o
[ 37%] Building CXX object CMakeFiles/dabMini-2.0.dir/home/boudewijn/DATV/DAB/qt-dab/src/backend/firecode-checker.cpp.o
[ 37%] Building CXX object CMakeFiles/dabMini-2.0.dir/home/boudewijn/DATV/DAB/qt-dab/src/backend/charsets.cpp.o
[ 38%] Building CXX object CMakeFiles/dabMini-2.0.dir/home/boudewijn/DATV/DAB/qt-dab/src/backend/galois.cpp.o
[ 39%] Building CXX object CMakeFiles/dabMini-2.0.dir/home/boudewijn/DATV/DAB/qt-dab/src/backend/reed-solomon.cpp.o
[ 39%] Building CXX object CMakeFiles/dabMini-2.0.dir/home/boudewijn/DATV/DAB/qt-dab/src/backend/msc-handler.cpp.o
[... 106 lines cut ...]
[ 93%] Building CXX object CMakeFiles/dabMini-2.0.dir/__/includes/support/moc_verysmallpushbutton.cpp.o
[ 93%] Building CXX object CMakeFiles/dabMini-2.0.dir/__/includes/support/moc_smallcombobox.cpp.o
[ 94%] Building CXX object CMakeFiles/dabMini-2.0.dir/__/includes/support/moc_smallspinbox.cpp.o
[ 95%] Building CXX object CMakeFiles/dabMini-2.0.dir/moc_radio.cpp.o
[ 95%] Building CXX object CMakeFiles/dabMini-2.0.dir/__/includes/backend/audio/moc_faad-decoder.cpp.o
[ 96%] Building CXX object CMakeFiles/dabMini-2.0.dir/devices-dab-mini/sdrplay-handler/moc_sdrplay-handler.cpp.o
[ 97%] Building CXX object CMakeFiles/dabMini-2.0.dir/devices-dab-mini/sdrplay-handler/moc_sdrplayselect.cpp.o
[ 97%] Building CXX object CMakeFiles/dabMini-2.0.dir/devices-dab-mini/airspy-handler/moc_airspy-handler.cpp.o
[ 98%] Building CXX object CMakeFiles/dabMini-2.0.dir/devices-dab-mini/rtlsdr-handler/moc_rtlsdr-handler.cpp.o
[ 99%] Building CXX object CMakeFiles/dabMini-2.0.dir/dabMini-2.0_autogen/EWIEGA46WW/qrc_resources.cpp.o
[100%] Linking CXX executable dabMini-2.0
/usr/bin/ld: /tmp/dabMini-2.0.JOdy8H.ltrans2.ltrans.o: in function `bandHandler::~bandHandler()':
<artificial>:(.text+0x72f5): undefined reference to `vtable for bandHandler'
/usr/bin/ld: /tmp/dabMini-2.0.JOdy8H.ltrans2.ltrans.o: in function `RadioInterface::~RadioInterface()':
<artificial>:(.text+0x74a5): undefined reference to `vtable for bandHandler'
/usr/bin/ld: /tmp/dabMini-2.0.JOdy8H.ltrans2.ltrans.o: in function `bandHandler::~bandHandler()':
<artificial>:(.text+0x76f5): undefined reference to `vtable for bandHandler'
/usr/bin/ld: /tmp/dabMini-2.0.JOdy8H.ltrans2.ltrans.o: in function `bandHandler::bandHandler(QString const&, QSettings*)':
<artificial>:(.text+0x94bf): undefined reference to `vtable for bandHandler'
/usr/bin/ld: <artificial>:(.text+0x9551): undefined reference to `bandHandler::staticMetaObject'
/usr/bin/ld: <artificial>:(.text+0x9581): undefined reference to `bandHandler::staticMetaObject'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/dabMini-2.0.dir/build.make:1809: dabMini-2.0] Error 1
make[1]: *** [CMakeFiles/Makefile2:104: CMakeFiles/dabMini-2.0.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

JvanKatwijk, 2020-12-13:
> function ‘std::vector fibDecoder::insert(std::vector, serviceId, int)’:
> /usr/bin/ld: :(.text+0x9551): undefined reference to bandHandler::staticMetaObject'

fixed

computer-wizard, 2020-12-14:
That was fast, thanks!

closed 2021-04-27 by JvanKatwijk