JvanKatwijk/qt-dab #193: Freeze on RPI4

AlexLoes, 2020-12-16:
Hi

First, congrats for this project. I downloaded the App files on Linux Debian. They just work great.
So, I also wanted this on the RPI4. So, I compiled like in the instructions on Raspi4, it compiles without error and an executable is built (qt-dab-3.7)
I installed it (sudo make install). When I type qt-dab-3.7 in a terminal, all I get is a Segmentation fault.
dabMini just works.

Please help :-)

JvanKatwijk, 2020-12-16:
well, does not look good then

Did you generate the executable using qmake or cmake
if qmake then pls send me the ".pro" file
Furthermore, assuming you used the qmake route pls change the
qt-dab.pro file:
current text reads
QMAKE_CFLAGS    +=  -O3 -ffast-math
QMAKE_CXXFLAGS  +=  -O3 -ffast-math
#QMAKE_CFLAGS   +=  -g
#QMAKE_CXXFLAGS +=  -g
#QMAKE_LFLAGS   +=  -g

make it into
#QMAKE_CFLAGS    +=  -O3 -ffast-math
#QMAKE_CXXFLAGS  +=  -O3 -ffast-math
QMAKE_CFLAGS   +=  -g
QMAKE_CXXFLAGS +=  -g
QMAKE_LFLAGS   +=  -g

if you are using the cmake route, then
change these lines in the CMakeLists.txt file
set (CMAKE_CXX_FLAGS "${CMAKE_XCC_FLAGS} -Wall -std=c++11 -flto")
set (CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -flto")

into
set (CMAKE_CXX_FLAGS "${CMAKE_XCC_FLAGS} -Wall -std=c++11 -g")
set (CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -g")

recompile everything and run with gdb
then after a crash do a "bt" command and send me the output

Then we can get an idea of what is happening

AlexLoes, 2020-12-16:
I am pleased to see, how active you are on this project. Amai!!!

The program is now compiled with the new values, but how to run it in gdb? I'm a total newbie with gdb.

Thanx.

AlexLoes, 2020-12-16:
Is as simple as gdb _program_

What bt gives me is:
(gdb) run
Starting program: /home/pi/qt-dab/qt-dab/dab-maxi/build/qt-dab-3.7 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0xb4dfb3c0 in ?? () from /lib/arm-linux-gnueabihf/libQtGui.so.4
(gdb) bt
#0  0xb4dfb3c0 in  () at /lib/arm-linux-gnueabihf/libQtGui.so.4
#1  0xb6fde2e8 in call_init
    (l=<optimized out>, argc=-1224742552, argv=0xbefff344, env=0xbefff34c)
    at dl-init.c:72
#2  0xb6fff978 in  ()
(gdb) Quit
(gdb)

JvanKatwijk, 2020-12-17:
> (l=, argc=-1224742552, argv=0xbefff344, env=0xbefff34c)

What qwt library did you install? I seem to remember that this type of
crash happens when the qwt library is for qt4 rather than for qt5

AlexLoes, 2020-12-17:
I've uninstalled libqwt-dev and installed libqwt-qt5-dev instead, recompiled the package and now everything works fine.

Hartelijk bedankt

Al

JvanKatwijk, 2020-12-17:
as a consolation: you was not the first one with this problem