JvanKatwijk/qt-dab #194: SNR too lo

AlexLoes, 2021-01-03:
Hello

I had an RTL2832 with an E4000 tuner in it, and DAB+ works fine. Now I have sold this stick, because it can not tune on 1090Mhz (dump1090).
I have now an Nooelec NESDR Mini 2+ with an RTL2838 and a R820T2 tuner. I am not able to get DAB+ working again. I can't get a SNR > 2 again. 
Is there an issue with RTLSDR or why doesn't it work?

JvanKatwijk, 2021-01-03:
I am not familiar with the 2838, but I have a question. Is time
synchronization ok (i.e. the indicator labeled "sync")
If so then - at least - time syunchronization is OK.
Can you make a record = just as raw dump - and send me a link, then I see
more of the signal

AlexLoes, 2021-01-03:
The sync Button is always flickering. 
I recorded two files: An xml one and a raw. For that, I let gain in auto mode.
But nethertheless, if I play manually with gain, snr is going from 2 to 1.
I live relatively near the broadcast antenna (~5 km).
You can download the files from http://loes.org.lu/sdr

andimik, 2021-01-04:
Which version are you using? There was a bug in SNR calculator.

JvanKatwijk, 2021-01-04:
I have been looking at the signal, and I am nor sure it is a DAB signal at
all.
The occasionally happening time synchronization may be accidental: the
number of failing time sync attempts exceeds
the number of successful ones by a ratio of over 10 to 1. For a more or
less normal DAB signal - even when the
gain setting is wrong - I never saw such a ratio
If you use e.g. a spectrumviewer, can you confirm then that what you see
are DAB signals on the selected frequency

AlexLoes, 2021-01-04:
Hi.

I don't have a spectrumviewer, but as I said, with the other stick, it worked. 
Perhaps my antenna is not good enough. I use the antenna delivered with the stick, a small one of about 10cm of length.
I am no pro in radiotechnique, so I think my antenna is not good. I order the other stick again and it should work.

andimik, 2021-01-05:
@AlexLoes  

I've downloaded your files, and only then I saw which channel you were using.

> Generic RTL2832U OEM-7D-So--Jan--3-2021-19-44-26.uff

I have an idea. 

Bingo!

> http://loes.org.lu/sdr/

You are living in Luxembourg!! **They have switched off 7D some days ago.** This was just a test. They will start later this year with regular transmissions.

So you only could try to receive 

- 11A SWR RP from Saarburg
- 5C Bundesmux I from Saarbrücken
- 9B Bundesmux II from Saarbrücken
- 9A DR Saarland from Moseltal
- 5B and 6C from Léglise in Belgium

@JvanKatwijk  You can close it. Has nothing to do with Qt-DAB itself.

AlexLoes, 2021-01-05:
Yes, Indeed, as I've done a research on the internet. Luxembourg is far behind the other countries.

closed 2021-01-05 by AlexLoes