JvanKatwijk/qt-dab #195: SNR SHOWN WRONG on 37 pre release ... and HEAVY interrupts with 3.7 release

DG9BFC, 2021-03-06:
title says it all ... snr is shown with roughly half the value what it is in real
as an example ... a local station that shows 18 to 20 db ... is show with 9 db only in main window and even lower sometimes in snr viewer (only 7 db) when i use the 37 pre release
when i use the newest 3.7 i have HEAVY interruptions so buffering seems not to work anymore
??? puzzelt

greetz sigi dg9bfc

ps if it helps ... used the 32 bit version on a core2duo with win7 ... 64bit 3.6 version works fine
as does older 32 bit version

JvanKatwijk, 2021-03-07:
Thanks for the report.
I cannot re-create the phenomenon you are describing with the 3.7 windows
version.
See the attached pciture of my screen.

By toucching the copyright symbol on the main widget with the mouse, the
build date and the commit identifier are shown.
Please tell me the build date/commit id from the version that gives you
troubles

Wrt the SNR: in one of the earlier versions a factor 2 was missing, that
was fixed some time ago.

There is a slighjt difference between the SNR reported on the main GUI and
the SNR as reported in the SNR widget, that is known
and being investigated

all the best
jan

JvanKatwijk, 2021-03-07:
ps
what is the processor load when reception is suffering from interruptions

DG9BFC, 2021-03-09:
the processor load (on an old core2duo) is around 50% ... 
the newest release has bad interruptions (but shows snr correct)
the version before ... works with not breaks (but shows snr with half value)
both show around 50% cpu load 
a bit higher on newest ... after each break the load goes up and down a bit ... trying to resync .. plays half a second or so and then another resync
i did not try that newest (32bit) version on my win 10 tablet yet (and i do not find a 64 bit version)

JvanKatwijk, 2021-03-09:
You should blame windows.
On my laptop the load under Linuix is 1 to 2 procent, under windows 15;
on my (4 year) old laptop the load under Linux was about 15 procent, under
windows 50.
Unfortunately, the W64 cross software was on the old laptop, which had a
severe system crash, so it is not usable right now

DG9BFC, 2021-03-09:
i know ... windows eats up cpu load ... 
fwiw ... the dec1 release works (with wrong snr but no breaks) ... the dec7 release shows those bad breaks (with double snr) ... so ... snr value is fixed ... but it does not play anymore :-(

DG9BFC, 2021-03-09:
i have no idea if that helps ... if i install the newer 3.7 version ... in same folder as the 3.6 version (without deleting the 3.6) and then start the 3.6 ... it also shows an empty station list
so ... between those two versons you changed something in the screen layout that makes now "white text on white background" (only active station is shown in red)

andimik, 2021-03-09:
Jan, the picture will only attached when you logon to GitHub and open #195

JvanKatwijk, 2021-03-09:
[image]

this should be the picture I was mentioning, thanks Andreas!!

JvanKatwijk, 2021-03-09:
I uploaded a version where the snr widget shows a button to record the snr on a user specified file. I hacked the linux
version of the viewer for these files such that it can ne used under Windows again. It is a simple tool where you can
scan through the recorded snr data. 
[image]

the viewer is part of the release (called: dumpViewer). Use is simple, on start up a file menu will appear
This might give some insight in when sudden changes in the snr occur

DG9BFC, 2021-03-09:
i have TWO problems ... on my core2duo heavy interrupts on newest version (version before works fine) ... so you definitely changed someting that now let the core2duo stutter 
and ... ión my win10 the 32 bit version does not work correct (so i stay on version 36 till you can compile a 64 bit version again) ... the 32 bit version shows an "empty" station list (white text on white background)
about cpu looad ... yes the core2duo shows around 50% ... the atom tablet shows around 25 only ...
with a pluto ... core2duo shows 60 and tablet shows 30 ... no stutter (with dec1 version)

JvanKatwijk, 2021-03-10:
Well, you are a luckhy guy with only two problemns. I have many more!!
As long as I cannot recreate the problem, there is not much I can do, I do
not like the tone of "...so you definitely ...",
my advice is: if you do not like the software, ask your moiney back and go
ssomewhgere else

closed 2021-03-10 by JvanKatwijk