JvanKatwijk/qt-dab #196: Please add topics to the repository

mendel5, 2021-03-26:
It would be great if you could add the following Github topics to this repository:

- [https://github.com/topics/sdr](https://github.com/topics/sdr)
- [https://github.com/topics/rtl-sdr](https://github.com/topics/rtl-sdr)
- [https://github.com/topics/dab](https://github.com/topics/dab)
- [https://github.com/topics/dabplus](https://github.com/topics/dabplus)

Thanks!

andimik, 2021-03-27:
What is the benefit?

The 2nd shows 219 hits. The majority has nothing to do with DAB.

mendel5, 2021-03-29:
@JvanKatwijk Thanks for accepting my suggestion. I think you mistyped [https://github.com/topics/dabplus](https://github.com/topics/dabplus) to [https://github.com/topics/dabplus-](https://github.com/topics/dabplus-)?

mendel5, 2021-03-30:
Thanks for fixing it!

closed 2021-03-30 by mendel5