JvanKatwijk/qt-dab #197: SNR shows wrong

DG9BFC, 2021-03-29:
hello jan
version 3.6 shows an snr of 24 db on a local station but it shows way lower on 3.7 (a few db only ?!?)

andimik, 2021-03-29:
The same issue has been discussed in (closed) https://github.com/JvanKatwijk/qt-dab/issues/195

Jan fixed it already.

closed 2021-03-30 by JvanKatwijk

DG9BFC, 2021-03-30:
???????

andimik, 2021-03-30:
> Wrt the SNR: in one of the earlier versions a factor 2 was missing, that was fixed some time ago

I checked it, this was done on 7th December 2021 in 07f5b10da

radio.cpp#1698
dab-processor.cpp#332

DG9BFC, 2021-03-30:
... factor 20 is higher so snr shoud show higher ... not??
but in 3.6 a local is shown with 24 and in 3.7 with 12 db .... maybe that factor 10 slipped back in when compiling the windows installer??

DG9BFC, 2021-03-30:
ok ok ok problem solved ... somehow i have got a 3.7 that showed snr wrong ...
now downloaded it again ... installed again ... and snr is correct ... grmbl grmbl ... sorry for wasting your time
back in my hole
dg9bfc sigi