JvanKatwijk/qt-dab #198: qt-dab maxi sdrplay_v3 probleem bij init onder linux

pe1ayx, 2021-04-24:
Beste een mooi stukje software geschreven !
1 probleem sdrplay_v3 _init failure.
Het probleem heb ik opgelost als volgt:
de sdrplay librarie functies niet private gedeclareerd
dlopen functie weggehaald.
loadFunctions weggehaald
programma gelinkt met += -lsdrplay_api
Nu werkt het als een sneltrein .
Hopelijk heb je iets hier aan.

Kort samengevat libsdrplay niet intern laden maar mee linken bij de build 

Hans

JvanKatwijk, 2021-04-24:
Kan natuurlijk, maar ivm copyrights op o.a. SDRplay driver software worden
de betreffende interface libraries dynamisch geladen ipv meegelinkt,
Speelt uiteraard niet als je zelf een exedcutable bakt maar wel bij de
windows installer en de Linux appImage

pe1ayx, 2021-04-25:
sdrplay_v3 wordt niet gecompileerd met --shared parameter waardoor sdrplay-handler-v3.o geen onderdelen
bevat van libsdrplay_api.so.
De linkparameter -lsdrplay_api geeft aan dat de linux image/programma de missing routines in
een library extern van het programma moet zoeken.
sdrplay geeft niet voor niets een headerfile mee zodat externe programm's er gebruik van mag maken.
Als je de sdrplay library even zou renamen zou je merken dat het rpogramma niet meer werkt ,hij vind de
bibliotheek niet meer.

Hans

closed 2021-04-27 by JvanKatwijk