JvanKatwijk/qt-dab #209: Crash when details window is open and frequency is changed

michaellass, 2021-05-12:
Looks like 804f592c803308edece5d04059268576a8d54d9d introduced a new issue.

When having the details window open, change the frequency. For me it crashes in `motDirectory::~motDirectory()` with and without sanitizer. With sanitizer, this is the trace:

```
==10953==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x60b00026d158 at pc 0x55f17e8702cf bp 0x7ffd17509380 sp 0x7ffd17509370
READ of size 8 at 0x60b00026d158 thread T0
    #0 0x55f17e8702ce in motDirectory::~motDirectory() /home/bevan/git/qt-dab/src/backend/data/mot/mot-dir.cpp:60
    #1 0x55f17e8705b1 in motHandler::~motHandler() /home/bevan/git/qt-dab/src/backend/data/mot/mot-handler.cpp:59
    #2 0x55f17e8706c8 in motHandler::~motHandler() /home/bevan/git/qt-dab/src/backend/data/mot/mot-handler.cpp:60
    #3 0x55f17e859145 in non-virtual thunk to dataProcessor::~dataProcessor() (/home/bevan/git/qt-dab/dab-maxi/qt-dab-3.72+0x12d145)
    #4 0x55f17e841d72 in Backend::~Backend() /home/bevan/git/qt-dab/src/backend/backend-driver.cpp:61
    #5 0x55f17e84221e in mscHandler::reset_Channel() /home/bevan/git/qt-dab/src/backend/msc-handler.cpp:217
    #6 0x55f17e7ec958 in RadioInterface::startChannel(QString const&) /home/bevan/git/qt-dab/dab-processor.cpp:119
    #7 0x55f17e7eeafa in RadioInterface::selectChannel(QString const&) /home/bevan/git/qt-dab/dab-maxi/radio.cpp:2991
    #8 0x55f17e90212b in RadioInterface::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) /home/bevan/git/qt-dab/dab-maxi/moc_radio.cpp:498
    #9 0x7f39e6c22baf  (/usr/lib/libQt5Core.so.5+0x2edbaf)
    #10 0x7f39e7fc8855 in QComboBox::activated(QString const&) (/usr/lib/libQt5Widgets.so.5+0x25d855)
    #11 0x7f39e7fcaa66  (/usr/lib/libQt5Widgets.so.5+0x25fa66)
    #12 0x7f39e6c22baf  (/usr/lib/libQt5Core.so.5+0x2edbaf)
    #13 0x7f39e7fc8a92 in QComboBoxPrivateContainer::itemSelected(QModelIndex const&) (/usr/lib/libQt5Widgets.so.5+0x25da92)
    #14 0x7f39e7fc92bf in QComboBoxPrivateContainer::eventFilter(QObject*, QEvent*) (/usr/lib/libQt5Widgets.so.5+0x25e2bf)
    #15 0x7f39e6beb582 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) (/usr/lib/libQt5Core.so.5+0x2b6582)
    #16 0x7f39e7ec5750 in QApplicationPrivate::notify_helper(QObject*, QEvent*) (/usr/lib/libQt5Widgets.so.5+0x15a750)
    #17 0x7f39e7ecc88a in QApplication::notify(QObject*, QEvent*) (/usr/lib/libQt5Widgets.so.5+0x16188a)
    #18 0x7f39e6beb819 in QCoreApplication::notifyInternal2(QObject*, QEvent*) (/usr/lib/libQt5Core.so.5+0x2b6819)
    #19 0x7f39e7ecb88d in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) (/usr/lib/libQt5Widgets.so.5+0x16088d)
    #20 0x7f39e7f1f89e  (/usr/lib/libQt5Widgets.so.5+0x1b489e)
    #21 0x7f39e7f2270e  (/usr/lib/libQt5Widgets.so.5+0x1b770e)
    #22 0x7f39e7ec5761 in QApplicationPrivate::notify_helper(QObject*, QEvent*) (/usr/lib/libQt5Widgets.so.5+0x15a761)
    #23 0x7f39e6beb819 in QCoreApplication::notifyInternal2(QObject*, QEvent*) (/usr/lib/libQt5Core.so.5+0x2b6819)
    #24 0x7f39e6fc6593 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) (/usr/lib/libQt5Gui.so.5+0x13a593)
    #25 0x7f39e6f9bbb4 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) (/usr/lib/libQt5Gui.so.5+0x10fbb4)
    #26 0x7f39df6f019b  (/usr/lib/libQt5XcbQpa.so.5+0x5f19b)
    #27 0x7f39e531702b in g_main_context_dispatch (/usr/lib/libglib-2.0.so.0+0x5402b)
    #28 0x7f39e536ab58  (/usr/lib/libglib-2.0.so.0+0xa7b58)
    #29 0x7f39e5314780 in g_main_context_iteration (/usr/lib/libglib-2.0.so.0+0x51780)
    #30 0x7f39e6c4433b in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) (/usr/lib/libQt5Core.so.5+0x30f33b)
    #31 0x7f39e6bea17b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) (/usr/lib/libQt5Core.so.5+0x2b517b)
    #32 0x7f39e6bf2633 in QCoreApplication::exec() (/usr/lib/libQt5Core.so.5+0x2bd633)
    #33 0x55f17e7ac150 in main /home/bevan/git/qt-dab/dab-maxi/main.cpp:149
    #34 0x7f39e6451b24 in __libc_start_main (/usr/lib/libc.so.6+0x27b24)
    #35 0x55f17e7ae71d in _start (/home/bevan/git/qt-dab/dab-maxi/qt-dab-3.72+0x8271d)

0x60b00026d158 is located 8 bytes to the left of 112-byte region [0x60b00026d160,0x60b00026d1d0)
allocated by thread T53 (dabProcessor) here:
    #0 0x7f39e84b2f41 in operator new(unsigned long) /build/gcc/src/gcc/libsanitizer/asan/asan_new_delete.cpp:99
    #1 0x55f17e870944 in motDirectory::analyse_theDirectory() /home/bevan/git/qt-dab/src/backend/data/mot/mot-dir.cpp:138
    #2 0x55f17e87365b in motDirectory::directorySegment(unsigned short, unsigned char*, short, int, bool) /home/bevan/git/qt-dab/src/backend/data/mot/mot-dir.cpp:112
    #3 0x55f17e87365b in motDirectory::directorySegment(unsigned short, unsigned char*, short, int, bool) /home/bevan/git/qt-dab/src/backend/data/mot/mot-dir.cpp:88
    #4 0x55f17e87365b in motHandler::add_mscDatagroup(std::vector<unsigned char, std::allocator<unsigned char> >) /home/bevan/git/qt-dab/src/backend/data/mot/mot-handler.cpp:183
    #5 0x55f17e875f08 in dataProcessor::handlePacket(unsigned char*) /home/bevan/git/qt-dab/src/backend/data/data-processor.cpp:173
    #6 0x55f17e928ab2  (/home/bevan/git/qt-dab/dab-maxi/qt-dab-3.72+0x1fcab2)

Thread T53 (dabProcessor) created by T0 here:
    #0 0x7f39e84571c7 in __interceptor_pthread_create /build/gcc/src/gcc/libsanitizer/asan/asan_interceptors.cpp:214
    #1 0x7f39e6a02b22 in QThread::start(QThread::Priority) (/usr/lib/libQt5Core.so.5+0xcdb22)

SUMMARY: AddressSanitizer: heap-buffer-overflow /home/bevan/git/qt-dab/src/backend/data/mot/mot-dir.cpp:60 in motDirectory::~motDirectory()
Shadow bytes around the buggy address:
  0x0c16800459d0: fa fa fa fa fd fd fd fd fd fd fd fd fd fd fd fd
  0x0c16800459e0: fd fd fa fa fa fa fa fa fa fa 00 00 00 00 00 00
  0x0c16800459f0: 00 00 00 00 00 00 00 00 fa fa fa fa fa fa fa fa
  0x0c1680045a00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 fa fa
  0x0c1680045a10: fa fa fa fa fa fa fd fd fd fd fd fd fd fd fd fd
=>0x0c1680045a20: fd fd fd fd fa fa fa fa fa fa fa[fa]00 00 00 00
  0x0c1680045a30: 00 00 00 00 00 00 00 00 00 00 fa fa fa fa fa fa
  0x0c1680045a40: fa fa fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0c1680045a50: fa fa fa fa fa fa fa fa fd fd fd fd fd fd fd fd
  0x0c1680045a60: fd fd fd fd fd fa fa fa fa fa fa fa fa fa fd fd
  0x0c1680045a70: fd fd fd fd fd fd fd fd fd fd fd fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==10953==ABORTING
```

JvanKatwijk, 2021-05-12:
For now I cannnot recreate the error, both with and without the sanitizer
is works fine, can you send me a link to a recording where it - most likely
- would
have happened in your environment

michaellass, 2021-05-12:
Actually, this is pretty hard to reproduce. I created a pull request that is supposed to fix this issue but verifying if it does it pretty hard. Please have a close look on the PR if it seems reasonable to you or not.

michaellass, 2021-05-13:
Fixed via c59f0f924e5808b9f5d9ad9a223c22d22c254872.

closed 2021-05-13 by michaellass