JvanKatwijk/qt-dab #211: i18n (German) broken in dab-maxi since 3.72

andimik, 2021-05-13:
Although there is only one i18n (it's German, maintained by me), it's broken somewhen between 3.71 and 3.72. 

Have tried both, the 3.72 AppImage and have compiled 3.72 on my own. No German.
In the 3.71 AppImage, German expressions and tooltips are working fine.

I could not find out why this happened as the files are still there and the path to i18n/de_DE.ts is declared in the dab-maxi's pro-file.

```
TRANSLATIONS += i18n/de_DE.ts
```

and it shows

```
main: Detected system language "de_AT"
main: Set language "de_DE"
```
right after the start.

I've also tried `lupdate`, it shows all German texts, but it is not used.

Remark: dab-mini's i18n works, indeed.

andimik, 2021-05-13:
I found the error. It's in dab-maxi/main.cpp

see https://github.com/JvanKatwijk/qt-dab/commit/f3f4b9a991313f7d8b5d0f56f47217da9852e3f3

When I revert it, then it works with 3.72.

JvanKatwijk, 2021-05-13:
> I found the error. It's in dab-maxi/main.cpp
>
> see f3f4b9a
>
> When I revert it, then it works with 3.72.

OK, I was in the mood of chaning dynamically allocated stuff to locally
allocated ot changed into std::vectors or so

andimik, 2021-05-13:
Ok, fixed in https://github.com/JvanKatwijk/qt-dab/commit/c59f0f924e5808b9f5d9ad9a223c22d22c254872

closed 2021-05-13 by andimik