JvanKatwijk/qt-dab #215: Wrong font in main window since 3.71

andimik, 2021-05-13:
Regarding `dabradio.ui`:

Although objectName `ensembleDisplay` is still declared to use font DejaVu Sans, 10px and objectName `leftPane` is font Ubuntu, 11, the font is a serif one since 3.71.

See attached screenshot: 3.71 and 3.72
[image]

compared to 3.70  (6299c14)
[image]

But it seems it has nothing to do with Qt-Designer. And one of your pics shows 3.71 in Windows with non-serif font ...

JvanKatwijk, 2021-05-13:
There was an issue with fonts in windows (where else?), to handle that for
once and for all font and font size for the service list can be set by the
user
    theFont                 =
                  dabSettings -> value ("theFont", "Times"). toString ();
        fontSize                =
                  dabSettings -> value ("fontSize", 12). toInt ();

andimik, 2021-05-13:
I see, I found it on https://github.com/JvanKatwijk/qt-dab/dab-maxi/radio.cpp#L289

Thanks!

closed 2021-05-13 by andimik