JvanKatwijk/qt-dab #218: No complete channel reset on channel change

andimik, 2021-05-18:
When I change the DAB channel there is still something in the "cache" from the previous channel. So, there is no real channel reset done, although this is reported on the console (see below).

I saw that both, 

# in the backend 

```
Allocating 15 zero-copy buffers
we start WELLE 1          at 08:09
starting a backend for WELLE 1          (AD58)
Starting hidden service EPG ORS         
starting a backend for EPG ORS          (E0A0AD50)
channel reset: all services will be stopped
channel reset: all services will be stopped
channel reset: all services will be stopped
channel reset: all services will be stopped
channel reset: all services will be stopped
channel reset: all services will be stopped
channel reset: all services will be stopped
channel reset: all services will be stopped
we start Pomurski MMR     at 08:10
starting a backend for Pomurski MMR     (9456)
channel reset: all services will be stopped
Starting hidden service EPG ORS         
starting a backend for EPG ORS          (E0A0AD50)
```
(Pomurski MMR from Slovenia is in the channel 12B I heard previously, whereas WELLE 1 and EPG ORS are from Austria on 6A) 

# and in the GUI (see screenshot)

[image]

WELLE 1 from Austria on 6A (which was in the previous channel) does not belong to the rest (all from Slovenia on 10D).

Tried both, with rtl_tcp and with dabstick.

JvanKatwijk, 2021-05-18:
you are using the airspy as device?

andimik, 2021-05-18:
Hi Jan, thanks for the hint, yes I have an Airspy as well, and this seems to work with an Airspy.

And I successfully tried with a HackRF. 

But just tested it for around half an hour, changing the DAB channels (I can receive 4 at home) all the time.

Let me investigate if this still happens, I'll close this issue in the meanwhile.

So, just letting you know about this strange behaviour ...

closed 2021-05-18 by andimik