JvanKatwijk/qt-dab #219: 64 bit version does not find rtl stick (32 bit does)

DG9BFC, 2021-06-07:
title says all .. the 64 bit version 3.72 does not find thr rtl stick
the older versions did and the 3.72 (32 bit) also does
??
running a new pc since a few days and had to reinstall all kind of stuff
replaced the genuine driver with zadig (like normal) but the 64 bit does not find it
a bit puzzelt dg9bfc sigi

JvanKatwijk, 2021-06-07:
It is a mixup of names. Sometimes the library is called rtlsdr.dll and
sometime librtlsdr.dll
When I have some time I'll look into it

JvanKatwijk, 2021-06-08:
If you rename in the installed folder (C:\Program Files (x86)\qt-dab64) or
copy the file rtlsdr.dll to librtlsdr.dll datsticks will work.

The 64 version is generated on a different - old - computer and later on
I'll generate a new installer

DG9BFC, 2021-06-08:
If you have an updated version.. Keep me informed to try it.. 32 bit version has some tiny faults and i hope the 64 bit works better on this machine (tii does not work and mot slide also does not work) Hmm rename or copy ??I7 quadcore with 32 gig ram running win10 pro (low cpu load even if i run several things in parallel)Ok i will try to get 64 bit runningGreetz sigi dg9bfc








-- 
Jan van Katwijk

DG9BFC, 2021-06-08:
renaming solved it :-) thanks for the hint (and sure you do all next versions with using librtl.dll not rtl.dll) :-)
now 32 and 64 bit work fine ... both show the mot slides now (no idea why in 32 bit it did not work but now it does)
so only thing that now does not work is "TII" .. hmm i am quite sure it worked on the other pc on one of the other older versions
so ... should i make a new issue about tii does not work?? (renaming solved THIS issue and you sure do next versions with correct filenames)?!?
greetz sigi dg9bfc

DG9BFC, 2021-06-08:
keep it open till new version is out :-) but then can be closed

JvanKatwijk, 2021-06-09:
Updated the repository wirh a version that understands both rtlsdr.dll as
librtlsdr.dll
No problems here with tii nor with mot slides

DG9BFC, 2021-06-10:
version before (that needed renaming of file) did open mot slides just fine
this last one doesd not opem mot slide
no tii can be seen

DG9BFC, 2021-06-10:
stepping back one version (and rename the dll) and mot slide works again ... hmm?!?
newest found the stick without rename but no mot slide
no tii in both cases

i remember that in an older version tii did work (but that was on a different pc)

by the way ... could you make the mot slide zoomable?? pulling the mot window bigger or even switch it to fullscreen does not change the size of the picture ... a fullscreen view (on another monitor) or a zoomed up window (depending on screen resolution and windows zoom settings) would be very nice to have
....
i have no idea why tii does not work here :-(

JvanKatwijk, 2021-06-10:
I am not sure the picture I attached in the previous mail is sent when
using the github issue mailing, but there you could see (a) a MOT picture
and (b) tii data

wrt MOT pictures, I cannot resize them

andimik, 2021-06-10:
@DG9BFC 

Sigi: In Barenburg you should receive all necessary muxes with various or even multiple TII codes (for example Bundesmux 1501 from Bremen), but depending on the field strength.

see screenshot, taken from fmscan.org

[image]

Qt-DAB only shows the strongest one. Other tools like qirx ( see https://qirx.softsyst.com/ , Windows only) show all of them.

@JvanKatwijk 

I guess the attachment function only works when you log in to github. There is a link at the bottom of each mail, so no need to search for the correct path.

For example: 

> view it on GitHub
>  <#219 >,

DG9BFC, 2021-06-10:
[image]
as you can see .. mot slide works but no tii
this is the older versions (that needed renaming of dll)
newer version and mot opens but does not show picture (only a white bar .. the top row of the windows box)
yes i can receive a few dab muxes here .. but be it one or ten muxes ... no tii :-(

JvanKatwijk, 2021-06-10:
Since the issue with the dabstick is solved I suggest closing the issue

JvanKatwijk, 2021-06-10:
MOT slides:
for display we are dependent on what Qt provides. For the 32 bits version
there is support for app 10 different formats (i.e. bmp, jpg, etc), for the
64 bit version there are 4.
If that is a problem, then switch over to the 32 bit version. I am already
sorry to have created the 64 bit version

DG9BFC, 2021-06-10:
question is what did you change between last version (no rename of dll needed) and the version before (that needed renaming)
did you maybe "accidently" used a wrong file when making the new kit?? (older qt file used?!?)
as you can see from my screenshot the mot slide works (with the version from a few days ago! and NOT with newest one)
hmm shall i now make a new issue?? mot slides not working in 64 bit in version 3.72?!?

JvanKatwijk, 2021-06-10:
noting was changed

I will delete the 64 bit version since that seems to give trouble, the 32
bit version is THE version

DG9BFC, 2021-06-10:
32 bit and 64 bit do not show tii ... mot works with the older version .. so i guess it was caused by mixing up the different qt versions on your different pc that you used to develop ... I DO NOT WANT TO STEP ON YOUR TOES FOR SURE .. call me a beta tester ... i do not want to moan i just want to report my findings
when i use the version from 25 may is see mot slides do open ... when using version from 27 may mot slides do not open
on every version test i deinstalled the older version before
when i first install 25 may and without deleting i then install 27 may ontop ... guess what .. mot slides do open
so a mix of the two versions seems to be right :-)
files that are not deleted when i then do a deinstall are
libgcc_s_dw-1
libjpeg-62
libportaudio-2
libportaudio-2-xxx
when i manually copy the file libjpeg-62 in the folder (from the older version) then mot slides work ... so
in newest version you missed to include the libjpeg-62 ...
guess i found it ... GRIN

closed 2021-06-10 by DG9BFC

andimik, 2021-06-11:
@DG9BFC 

Sigi, could you please try the 32bit with the TII?

Maybe you could share approx. 10 seconds of a mux which does not show TII.

Jan will investigate then.

DG9BFC, 2021-06-11:
ok i make a recording .. no problem

andimik, 2021-06-11:
Maybe in raw or uff/xml format (see RTLSDR window), not sdr

DG9BFC, 2021-06-11:
i made two recordings (two different muxes) but need a direct email adress (to send them via wetransfer) ... i lost my old pc (where i had a direct email adress from you stored in the emailsoft)

DG9BFC, 2021-06-11:
ok found it :-)

JvanKatwijk, 2021-06-11:
well, you do not have much of a signal, snr 2 or 3.
With a mod in the software I can extract some tii data, but since it is a
matter of signal power the mod opens the possibilities for false indications

From your two fragments I get 15 02 for the one and 04 06 and 03 03 for the
other

JvanKatwijk, 2021-06-11:
sorry read for the last one 20 06 and 19 03

JvanKatwijk, 2021-06-11:
and for the first one 3102

DG9BFC, 2021-06-11:
i remember that on old pc tii did work
and sure if you receive several signals (from different stations) for the same mux a "false" decode may happen
can you find out why i do not see ANY tii at all here?? it is sure not just caused by signal level alone ... or??

JvanKatwijk, 2021-06-11:
Enclosed a picture of the prcessingof the two files
Of course, the approach - essentially by reducing some threshold values -
opens the door for false indications

JvanKatwijk, 2021-06-11:
picture seems to be lost

JvanKatwijk, 2021-06-11:
next week I'll upload a version with the "lowered" threshold

DG9BFC, 2021-06-11:
what about adding threshold in the config window?? so user can set to more sensitivity (but maybe some false decodes) or only show tii with higher snr ?? just thinking
about the signal level and quality ... i plan to put a better antenna higher in the air plus adding a preamp direct at the antenna (feed from bias-t) ... maybe that also helps a bit (especially on channel 5c where i like to hear sunshine live)
isnt the tii also the different peaks that are shown in the "correction" window?? (on 5c i see several peaks while on 7c i only see on major big peak) ... thats the reason why i sent you TWO recordings

DG9BFC, 2021-06-11:
correction .. 7a not 7c :-)

DG9BFC, 2021-06-11:
about the gui ... can you add on option that the soft does not ask every time when i close it if i am sure??
a checkbox in config window (confirm exit yes/no) ... or is there a registry setting where i could change it (so it is user hidden but still available) ?? that pop up window should only come up when doing a recording (Record is on do you really want to stop the soft and exit)
and ... maybe a checkbox with show MOT slides always ontop?? then user can see the mot slides and put them in upper right or left corner of his screen) while doing other work on the pc ...

DG9BFC, 2021-06-11:
a question ... a "good " dab raceiver uses also the other signals (decodes the other stations in parallel and summs signals together) ... are you only using the main (strongest) signal??
receiving a few signals in parallel would give better results in the geographical overlapping area of the signals (at cell border so to speak) ... thats why they have the single frequency network ... you lock on main channel but decode also the other streams and if on main channel some data gets lost it maybe is received from one of the other transmitting sites (aand filled in)
that "multi receive" sure needs more cpu load .. but with todays fast pc that should be no problem

DG9BFC, 2021-06-11:
on channel 5a (as an example) i see 4 peaks ... all with near same height ... summing up those 4 signals should give a super error free receive ... sometimes it lock on this and sometimes on that peak ?!?
most of the time its the third peek
[image]

but sometimes it loks n second peak (or beter say while locked on third peak it sometimes jumps to second and then to third)

[image]

JvanKatwijk, 2021-06-11:
The more peaks you see, the more signal interference is there, you can see
in the spectrum widget - and in the constellation - how tio signal is
polluted
Since one sample takes 1/2048000 seconds, you can compute the delay with
which the other signals are arriving (one seems slightly ahead)

DG9BFC, 2021-06-11:
you even can calculate the distance (or better say distance differences between the transmitter sites)
isn´t that how gps works?? (not exactly but quite similar) ;-)

JvanKatwijk, 2021-06-11:
No since the transmitters are not in a straight line, so the only thing you
know is the difference in arrival time of the data. In my case there are
two strong transmitters in opposite directions, so I
can compute how much farther away the second one is compared to the first
one

DG9BFC, 2021-06-11:
thats what meant with distance difference (the direction does not play a role here)
when they are not in a traight line it does not matter
say you are in centre of a triangle .. all three signals come to receiver at the same time ... now move one at a bigger distance and it will arrive later ... for that third receiver you can calculate the ADDED time and so distance difference to the other two
the first two can be 10km and third 12km away ... or 50 and 52 .. time difference should be the 2km ... not??
(all that is just if all signals on tx sites are well synced and gps locked but you sure know that)
... you did not answer my question ... do you just use a single signal (strongest peak) or do you receive a few in parallel and summ them up ? ... cause the latter would help in the overlapping areas with 3 or 4 signals (and low receive levels)
receive one strong signal is sure a lot easier as summing up 3 or 4 data streams .. 
my back and forth jumping corellation seems to show that you use only one data stream (strongest)

JvanKatwijk, 2021-06-11:
I do not know what you know about DAB (ofdm) signals.
They are digital signals (at least within the computer). There encoding is
such that it is possible - ususally - to extract the strongest signal.
Detection of the
stringest signal is by correlation. Signals from different transmitters
interfere of course, so the first problem is how to identify the stringest
signal,
the second problem is how to synchronize in the time domain with that
signal, the third problem is to correct any frequency offset and the fourth
problem is
how to decode the data you want, if you know that the signal is possluted
by interfering signals from other transmissions (or reflections of the
signal)
see wikipedia for OFDM transmissions

andimik, 2021-06-11:
> a "good " dab raceiver uses also the other signals (decodes the other stations in parallel and summs signals together) ... are you only using the main (strongest) signal??

@DG9BFC 

No, this is not the intention and is not possible at all on receiver side. It cannot pick up only certain signals within an SFN (single frequency network).

If you want to maximize your reception, please adjust the antenna or use a better one.

For your region I'm confident that the SNR should be greater 15, not only 2. Fmscan says 54 dB for 7A, which is quite strong.