JvanKatwijk/qt-dab #220: MOT slides do not open on newest 64 bit version

DG9BFC, 2021-06-10:
with stepping back one version and comparing folders from old and new version i found that in newest 64 bit version there is a file missing ... when i manually copy (from older version) the file libjpeg-62.dll in the 64 bit folder .. THEN mot slides do open
so jan .. please add a new version with this file included
thank you 
ps maybe thats also the cause for the other issue (mot slides not saved anymore) ?!?!?

JvanKatwijk, 2021-06-10:
The 64 version is officially withdrawn., it does not add anything compared
to the 32 bit version and gives me lots of work

DG9BFC, 2021-06-10:
but i found the missing file ... add it again and 64 bit works WITH mot slide .. no need to "kill" the 64 bit version

JvanKatwijk, 2021-06-11:
64 bits version is officially withdrawn and not supported

closed 2021-06-11 by JvanKatwijk