JvanKatwijk/qt-dab #221: SNR compression shows zig zag line with odd settings in config

DG9BFC, 2021-06-10:
when i set even numbers in config for snr compression all seems fine .. when i set odd numbers (1,3,5 etc) it shows a zigzag line
[image]
....
and now with even number
[image]

DG9BFC, 2021-06-10:
higher numbers show less high spikes but it is still going up and down so best would be to only use the even settings?!?

JvanKatwijk, 2021-06-11:
TII data is ytransmitted in each second frame, so there will be a difference in SNR measurements between first and second frames

closed 2021-06-11 by JvanKatwijk

DG9BFC, 2021-06-11:
ok you explained it ... maybe you sould add a hint about setting to even numbers only