JvanKatwijk/qt-dab #223: Feature Request: Log Dynamic Label of Stream

spongioblast, 2021-07-21:
Hi, is there any way to log only the dynamic label of the current stream selected? I would like to log the playlist so I wont have to use soundhound anymore to remember the song. Thanks.

JvanKatwijk, 2021-07-21:
Interesting question.
It is fairly easy to dump just the text lines of the dynamic label,
however. these are transmitted continuously
So you need some kind of postprocessor, removing the duplicates

spongioblast, 2021-07-21:
Postprocessing would easy. Easiest would be then to just have it logged with a new line every time its broadcasted, maybe with the timestamp of the channel.

JvanKatwijk, 2021-07-21:
Wit removing a new line if that is identical to the last line in the
output, you would get something like this
starting a backend for NPO Radio 2      (8202)
Guus Meeuwis - Net Als Dansen 18:55
going to write file /tmp/qt-pictures/MOT02\R2.png
NPO Radio 2 - Wout2day - KRO-NCRV  18:56
NPO Radio 2 - WoV  19:02
NPO Radio 2 - Wout2day - KRO-NCRV  19:02
Rod Stewart - Crazy About Her 19:06
azy About Her 19:06
Rod Stewart - Crazy About Her 19:06
azy About Her 19:07
Rod Stewart - Crazy About Her 19:07
NPO Radio 2 - Wout2day - KRO-NCRV  19:10
ut2day - KRO-NCRV  19:11
NPO Radio 2 - Wout2day - KRO-NCRV  19:11
NPO Radio 2 - WoV  19:11
NPO Radio 2 - Wout2day - KRO-NCRV  19:12
ut2day - KRO-NCRV  19:13
NPO Radio 2 - Wout2day - KRO-NCRV  19:13
NPO Radio 2 - Wout2day - KRO-NCRV
ut2day - KRO-NCRV  19:13

As you can see, this output is over a period on about 20 minutes

Easiest is to print it on stdout (messages like "going to write ..."  are
written to stderr
If you are running Linux it is easy as addon,

spongioblast, 2021-07-21:
That would already be fine, maybe better time before dls. The remaining cleanup could then easily be done post processing and will usually be channel specific. 

It would be better to have it directly save to a text file so the stream doesn't have to be redirected from stdout.

I have it running on a raspberry, so linux based. Sadly my programming knowledge for this is very limited, so I'd really appreciate the integration.

JvanKatwijk, 2021-07-21:
It is a simple operation, I'll add a  command line parameter - as
undocumented feature - and when specified the dll text - as specified (I'll
set time first)
to the file the name of which passed as parameter.

spongioblast, 2021-07-21:
Perfect, thanks. Filename could even be date and time plus channel name if filename is unspecified. That filename would always be unique.

JvanKatwijk, 2021-07-21:
A very first version is uploaded, just give it a try
The command line parameter now is "-t filename", where filename is - as the
name suggests - the filename
It might indeed be an idea to just have a filename generated, something
consisting of servicename + start date and time

JvanKatwijk, 2021-07-21:
The point with automatically generated filenames - as suggested - is that
it involves a lot more:
what happens if another service is selected, what happend when another
channel is selected, what happens if e.g. scanning is selected etc etc
So, while it might be a nice features, first of all semantics have to be
defined. Note that with the current "solution" these issues doe not play:
just what is dlltext is written to the file. A simpler solution might be to
add channel/service name to the text line, then - while
it might not be clear from the filename what the content is, the content
itself is very clear

spongioblast, 2021-07-21:
"A simpler solution might be to add channel/service name to the text line"

I fully agree, this is the simplest solution and would be completely fine. Everything else can then be done easily in post processing. Thanks, I will give it a go.

spongioblast, 2021-07-21:
Fantastic, works like a charm. Really appreciate the feature. It seems the file is only written to on closing qt-dab. Is this on purpose?

spongioblast, 2021-07-21:
On a side note, most of the redundant logs could be reduced by checking if the current line was written in the past three lines from what I've seen by checking a few channels.

JvanKatwijk, 2021-07-22:
well, there is now a check on a new line entering the system being the same
as the last line. Of course this can be expanded but at the
cost of buffering the last few lines

The file is now closed on exiting qt-dab.  There are some open issues
though: Windows users and users of the appImage on Linux cannot easily add
command line parameters, however, they might want to use such a feature.

spongioblast, 2021-07-22:
Yes, the cost would be buffering the last few lines, but its pretty much clean afterwards for most channels. 

If someone on windows or appImage wishes for the feature I guess they can request it. It seems so far that was not the case. As for me I am happy the way it works! Cheers.

spongioblast, 2021-07-22:
One nice addition would be to have the date before the time in the log, if logged over a longer time.

spongioblast, 2021-07-22:
Maybe an easy to implement idea for creating a log via gui -> Have it automatically create the channel log if aac is recorded. Then one has the stream as well as the dls data. Personally however I don't need the stream, I just sometimes like to look up what I listened to.

JvanKatwijk, 2021-07-22:
I uploaded a preliminary version 4. I Added a button on the main GUI - that
is why I changed the version number - with which
the text of the dynamic label is stored.
I added a buffer of the last 4 dl lines and if a new line is already in the
buffer it is not added, not perfect but it might help
The format is now
channel:service:time text

andimik, 2021-07-22:
Cool feature Jan, but one comment:

Could you use the same charset also for the textfile? See screenshot from Qt-DAB 4.0 below and see the special character in DLS:

[image]

```
10D.Radio Student   22:09:  Za narodov blagor!
10D.Radio Student   22:09:  Radio ?tudent - radiostudent.si
```
whereas the word Študent is shown in the qt-dab window.

JvanKatwijk, 2021-07-23:
Hi Andreas

Implementation is not finished yet, I'll look into the language
representation issue

JvanKatwijk, 2021-07-23:
The text is now mapped from an Qt string to an utf8 string, pls give it a
try

JvanKatwijk, 2021-07-23:
@spongioblast: well, this is it. Thanks for the suggestion, seems to be an
interesting feature

andimik, 2021-07-23:
yes, confirmed, now it's Unicode :-)

```
10D.Slo 1 Prvi      21:36:  Slovencem po svetu - Portreti: Marija Ahačič Pollak
```

```
10D.Radio Student   21:39:  Radio Študent - radiostudent.si
10D.Radio Student   21:39:  Radio Študent tudi na DAB+
```

```
10D.Slo 3 ARS       21:40:  Petkov koncertni večer - Posnetek koncerta Orkestra Slovenske filharmonije
```

and from a file (remark: 10D comes from the last received DAB channel)

```
10D.PR Dwójka       21:44:  Program II Polskie Radio S.A 
 tel. (22) 645 2222 
 e-mail:dwojka@polskieradio.pl 
10D.PR Dwójka       21:44:  Poranek Dwójki Piotr Kędziorek 
Teraz gramy: SCHULHOFF Concertino Paul Klee Ensemble Zehnder 
Concertino na flet, altówkę i 
```

spongioblast, 2021-07-23:
Works super well, thanks. Will test some more.

spongioblast, 2021-07-24:
Maybe using ISO 8601 compliant datetime format. Also adding seconds to the log:

Filename: yyyy-MM-ddTHHmmss_Qt-DAB_DLText
Eg 2021-07-24T162015_Qt-DAB_DLText
Log: yyyy-MM-ddTHHmmss | Channel | DLS Text
Eg  2021-07-24T162015 | 10D.PR Dwójka | Program II Polskie Radio S.A 

The filter with remembering the last four rows works quite well already. Easy to filter out the rest in post processing. Another way to improve the filtering might be instead of checking the last 4 written lines, check the last four dls texts streamed and refresh those, that should filter out more. As it is now, the repeating channel information is pushed out of the buffer as soon as 4 songs have passed. Hence every 4 songs are followed by the general channel info in the log.

JvanKatwijk, 2021-07-24:
Wrt time, there is still an issue: should the time recorded in the file be
the actual time or the time as reported from the transmission
(think on playing a recording)

Current filtering on lines is in a simple cache with 5 lines which can be
exanded,

spongioblast, 2021-07-24:
In the log I would say it should be the datetime of the transmission / channel. The filename should be systemdatetime I believe.

The problem with the cache that it takes the last lines written and not the last lines transmitted. Hence after 4 songs, it pushes the repeating channel name / program name out of the buffer. Those should be kept in the buffer, as they continuously come up with every song. It still works jut fine, but would further reduce repeated entries in the log.

JvanKatwijk, 2021-07-24:
No, the cache does not add lines already there

spongioblast, 2021-07-26:
Basically the buffer should be reordered so that the last dls line is the first one in the buffer / so it wont get pushed out of it if it still appears. That's why now the repeating station info with every song now repeats after every fourth song. Eg, With a buffer of four lines and if the station hast two station infos plus the song title, all three repeating a few times until the end of the song, the log looks the following:

ChannelID 16:00: Channel Info 1
ChannelID 16:00: Channel Info 2
ChannelID 16:00: Song 1
ChannelID 16:00: Song 2
ChannelID 16:00: Song 3
ChannelID 16:00: Song 4
ChannelID 16:00: Channel Info 1
ChannelID 16:00: Channel Info 2
ChannelID 16:00: Song 5
ChannelID 16:00: Song 6
ChannelID 16:00: Song 7
ChannelID 16:00: Song 8
ChannelID 16:00: Channel Info 1
ChannelID 16:00: Channel Info 2
and so on.

JvanKatwijk, 2021-07-27:
I am not sure I understand. The approach now is that a cache contains a
copy of the last X (currently X = 10) lines that are emitted.
If a line appears, it is matched against the cache content and only if it
is not in it is send to the output and added to the cache

spongioblast, 2021-07-27:
Hm, I'll quickly compile the latest version and have a look at it. In theory that should work.

spongioblast, 2021-07-27:
So now the channel information is repeated every 10 lines in the log. The problem is, that at some point the repeating elements are pushed out of the vector as only new elements are added. Two possible solutions would be to either alway add every element when broadcasted or rearrange the array. Not sure if this is correct in c++, but something like 

in dl-cache.h
```
bool	isMember	(const QString &s) {
	for (int i = 0; i < size; i ++)
	   if (cache [i] == s)
              std::swap(cache [0],cache [i]);
	      return true;
	return false;
}
};
```

or in radio.cpp

```
	if (the_dlCache. isMember (s))
	   the_dlCache. add (s);
	   return;
```

JvanKatwijk, 2021-07-28:
I do not think you are solving a problem that way.
If the repetition is with a distance of N+1 or larger, you still miss it
since the cache was already filled with other data
The second solurtion is equal to putting everything - or nothing - in the
cache.

I think that you should solve your problem with post processing the data.

spongioblast, 2021-07-29:
The general channel info comes however with every song, multiple times. So its not N+1 (if N = cache size) but repeats constantly during the song played. eg:

Song 1
Channel Info 1
Channel Info 2
Song 1
Channel Info 1
Channel Info 2
Song 1
Channel Info 1
Channel Info 2
Song 2
Channel Info 1
Channel Info 2
Song 2
Channel Info 1
Channel Info 2
Song 2
Channel Info 1
Channel Info 2
Song 3
Channel Info 1
Channel Info 2
and so on. 

Hence it is simply that the way the filter works now, the channel infos are pushed out of the cache after a further N-2 unique songs (where N = cache size) after which the channel info is added again to the cache as its no longer in it. Hence every 8 Songs the channel info is repeated and written to the log. 

Anyway, its only a small beauty issue, as you said it can easily be fixed in post processing.

JvanKatwijk, 2021-07-31:
https://www.dropbox.com/s/oqn0taejwaxg7dd/Qt_DAB-x86_64.AppImage?dl=0
I made a version with your suggestion, i.e. if a label reappears, it is put
as newest in the cache

give it a try

spongioblast, 2021-08-01:
The AppImage runs fine however I cant get it to recognize the rtl-sdr dab stick (rtl2832u) on ubuntu virtualbox. rtl_test -s 2400000 runs fine in commandline

JvanKatwijk, 2021-08-01:
If you start the appImage from a command window, what is the message

spongioblast, 2021-08-01:
I have it working but the dab frames are not transmitted consistently, from what I read virtualbox struggles with the dab sticks. 

`main: Detected system language "en_US"
main: Set language "en_US"
main: Error while loading language specifics "en_US" use English "en_GB" instead
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib setup.c:547:(add_elem) Cannot obtain info for CTL elem (MIXER,'IEC958 Playback Default',0,0,0): No such file or directory
ALSA lib setup.c:547:(add_elem) Cannot obtain info for CTL elem (MIXER,'IEC958 Playback Default',0,0,0): No such file or directory
ALSA lib setup.c:547:(add_elem) Cannot obtain info for CTL elem (MIXER,'IEC958 Playback Default',0,0,0): No such file or directory
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Hostapis: 2

Api 0 is ALSA

Api 1 is OSS

Investigating Device 0

 (output):item 1 wordt stream 0 (Intel 82801AA-ICH: - (hw:0,0))

Investigating Device 1

Investigating Device 2

 (output):item 2 wordt stream 2 (sysdefault)

Investigating Device 3

 (output):item 3 wordt stream 3 (front)

Investigating Device 4

 (output):item 4 wordt stream 4 (surround40)

Investigating Device 5

 (output):item 5 wordt stream 5 (surround41)

Investigating Device 6

 (output):item 6 wordt stream 6 (surround50)

Investigating Device 7
[... 184 lines cut ...]
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
OK, functions seem to be loaded
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Supported gain values (29): 49.6 48.0 44.5 43.9 43.4 42.1 40.2 38.6 37.2 36.4 33.8 32.8 29.7 28.0 25.4 22.9 20.7 19.7 16.6 15.7 14.4 12.5 8.7 7.7 3.7 2.7 1.4 0.9 0.0 
Realtek RTL2838UHIDIR 00000001
channel reset: all services will be stopped
channel reset: all services will be stopped

`

JvanKatwijk, 2021-08-01:
If you want I can generate a windows version this afternoon

JvanKatwijk, 2021-08-01:
https://www.dropbox.com/s/zsj7m5ca4kaksej/setup-qt-dab32.exe?dl=0

This is a link to the windows version, using the latest mods

spongioblast, 2021-08-02:
The repeating channel info is no filtered completely. However I get weird artifacts of song titles logged three time with the title repeating partially on the line. Maybe this is an issue with the windows version or some driver issues? Never had this on raspberry PI.

```

5D.OB   2021-08-02 07:39:06  curated by Baxter
5D.OB   2021-08-02 07:39:36  Coffee Kiss IX
5D.OB   2021-08-02 07:40:19  "Consequence" by Sly5thAve - What It Is
5D.OB   2021-08-02 07:42:37  "Corporate" by Leron Thomas - More Elevator Music
5D.OB   2021-08-02 07:47:06  "Behind The Sun" by Skinshape - Arrogance is the Death of Men
5D.OB   2021-08-02 07:49:20  "Marilou sous la neige" by Serge Gainsbourg - Gainsbourg... Forever
5D.OB   2021-08-02 07:51:36  "Girls Don't Always Sing About Boys" by Ego Ella May - Girls Don't Always Sing About Boys
5D.OB   2021-08-02 07:51:41  "Girls Don't Always Sing About Boys" by Ego Ella May - Girls Don't Always Sing About Boysoys" by Ego Ella May - Girls Don't Always Sing About Boys
5D.OB   2021-08-02 07:53:51  "Girls Don't Always Sing About Boys" by Ego Ella May - Girls Don't Always Sing About Boysbout Boys
5D.OB   2021-08-02 07:56:05  "Strollin’" by Prince - Diamonds and Pearls
5D.OB   2021-08-02 07:56:12  "Strollin’" by Prince - Diamonds and Pearlsds and Pearls
5D.OB   2021-08-02 07:58:19  "The Best Man" by The Nat King Cole Trio - Volume 2
5D.OB   2021-08-02 07:59:49  open broadcast
5D.OB   2021-08-02 08:00:35  "Mother's Love" by The Vernon Spring - Mother's Love 
5D.OB   2021-08-02 08:00:45  "Mother's Love" by The Vernon Spring - Mother's Love Love 
5D.OB   2021-08-02 08:01:20  curated by Pat
5D.OB   2021-08-02 08:02:06  Coffee Kiss - Coffee Kiss #4
5D.OB   2021-08-02 08:05:04  "In Loving Memory" by Gabriels - In Loving Memory
5D.OB   2021-08-02 08:09:36  "Love Song (Helado Negro Remix)" by Devendra Banhart - Love Song (Helado Negro Remix)
5D.OB   2021-08-02 08:11:54  "Love Song (Helado Negro Remix)" by Devendra Banhart - Love Song (Helado Negro Remix) by Devendra Banhart - Love Song (Helado Negro Remix)
5D.OB   2021-08-02 08:14:06  "Sweet Sofa" by ØDYSSEE - Sweet Sofa 
5D.OB   2021-08-02 08:16:19  "Les Fleurs (Live)" by Ivy Sole - Overgrown Live (from Philadelphia)
5D.OB   2021-08-02 08:20:50  "Tierra Querida" by Bruno Hovart - Latin Nostalgia
5D.OB   2021-08-02 08:23:06  "Playground Love" by Air - The Virgin Suicides
5D.OB   2021-08-02 08:27:36  "Berlin" by Blood Orange - Angel's Pulse
5D.OB   2021-08-02 08:29:49  "One More" by Cymande - Cymande

```

Or
```

5D.OB    2021-08-02 02:09:36  "Hold On" by Gus Seyffert - Hold On 
5D.OB    2021-08-02 02:10:12  "Hold On" by Gus Seyffert - Hold On  On 
5D.OB    2021-08-02 02:14:06  "AO Longo Do Rio" by Melodiesinfonie - A Journey to You
5D.OB    2021-08-02 02:14:13  "AO Longo Do Rio" by Melodiesinfonie - A Journey to You to You
5D.OB    2021-08-02 02:18:37  "Paperbag" by Wizards of Ooze - Almost... Bikini
5D.OB    2021-08-02 02:23:05  "To The Moon" by Spirit Fest - Spirit Fest
5D.OB    2021-08-02 02:27:37  "Waltz for Emily" by Gerardo Frisina - Join the Dance
5D.OB    2021-08-02 02:30:17  "Waltz for Emily" by Gerardo Frisina - Join the Dancesina - Join the Dance
5D.OB    2021-08-02 02:32:05  "Pain and Misery" by The Teskey Brothers - Half Mile Harvest
5D.OB    2021-08-02 02:34:23  "Eternal Light" by The Free Nationals - Eternal Light
5D.OB    2021-08-02 02:38:51  "Kominiké" by Erick Cosaque - Chinal K 1973-1995
5D.OB    2021-08-02 02:41:23  "Kominiké" by Erick Cosaque - Chinal K 1973-1995hinal K 1973-1995
5D.OB    2021-08-02 02:43:22  "River River" by Spirit Fest - Spirit Fest
```

[image]

JvanKatwijk, 2021-08-02:
Then the decoding of the dynamic label fragments cannot be done, and two or
more messages are combined.

spongioblast, 2021-08-02:
Thanks, just compiled it for the raspberry, works beautifully.

closed 2021-08-02 by spongioblast