JvanKatwijk/qt-dab #228: Broken constellation diagram

andimik, 2021-07-26:
commit 66d9ed0 (from today) does not show the constellation diagram any more.

[image]

d4d4388 (from yesterday) is working 

[image]

JvanKatwijk, 2021-07-27:
You are running Ubuntu? In my fedorabox I do not have that problem. Later
today I'll have a test on Ububtu

andimik, 2021-07-27:
Yes, its Ubuntu 20.04

JvanKatwijk, 2021-07-27:
Interesting issue
It ran perfectly on my Fedora box and gave - as you said - a blank
constellation diagram on ubuntu 20.
I compiled it also for Ubuntu 16 (the VM on which I make the appImages) and
indeed, the same problem there.
It turned out that in a recent pull request a few lines in a file were
removed. Restoring these lines solved the problem.
The appImage runs without problems on both Fedora and Ubuntu (16 and 20),
and the sources are updated

JvanKatwijk, 2021-07-28:
if the constellation now works as expected, will you close the issue?

andimik, 2021-07-28:
Confirmed, https://github.com/JvanKatwijk/qt-dab/commit/46fd07efb529d42a7dcc6e6ef93239f19b86a5f4 fixes that, so I can close it.

closed 2021-07-28 by andimik