JvanKatwijk/qt-dab #232: Airspy not working on Fedora

AsciiWolf, 2021-08-05:
Fedora 34, latest airspyone_host x86_64 package installed.

Qt-DAB AppImage fails to run when selecting an Airspy device:
```

failed to open libairspy.so
Error = libairspy.so: cannot open shared object file: No such file or directory
```

$ ls -l /usr/lib64/ | grep airspy
lrwxrwxrwx. 1 root root       18 Jan 26  2021 libairspy.so.0 -> libairspy.so.1.0.9
-rwxr-xr-x. 1 root root    45192 Jan 26  2021 libairspy.so.1.0.9

Fedora 34 does not ship libairspy.so, only libairspy.so.0 which I believe is a correct library name - Debian/Ubuntu also ships only libairspy.so.0 in [libairspy0](https://packages.debian.org/sid/libairspy0), libairspy.so is shipped only in libairspy-dev which is a devel library not intended for end-users. Please, consider changing the library name used to libairspy.so.0.

JvanKatwijk, 2021-08-05:
fixed,

AsciiWolf, 2021-08-05:
I can confirm that it works fine now, thanks for the quick fix! :-)

closed 2021-08-05 by AsciiWolf