JvanKatwijk/qt-dab #234: Custom device strings

AsciiWolf, 2021-08-09:
Please, consider supporting passing custom strings to SDR devices. They are used for example to select a correct SDR device if there are multiple ones connected to the same computer, enabling bias tee for preamps/converters etc.

JvanKatwijk, 2021-08-09:
for some devices there is a device selection mechanism, invoked as soon as
the system detects that there are more device candidates
Can you be slightly more specific and give an example? who is interpreting
the "custom" string, when  and howi s it submitted

AsciiWolf, 2021-08-23:
Sorry for the late reply. Good question! They seem to be called "device arguments" and are handled by the gr-osmosdr source, so I am not sure if it will be possible to implement them in qt-dab without using gr-osmosdr. :-/

closed 2024-10-17 by JvanKatwijk