JvanKatwijk/qt-dab #242: Scheduler does not work over night

andimik, 2021-11-10:
The scheduler only works for the same calender day.

If you want to start a recording in the early morning (let's say 3am) and you want to program it in the evening (let's say 9pm) then the timer immediately starts instead of waiting for 6 hours. 

So I fear you need to a day field as well?

JvanKatwijk, 2021-11-11:
the easiest solution would be to make semantics such that if the time
specified is before the actual time,
it is taken as the time of the following day.
Another "solution" would be to allow specification of time beyond 2400
hours (when I was a student
it was not uncommon to leave a party atl 27 or 28 hours of the current day)

In the end, however, the scheduling needs to be extebnded. This obviously
brings the question what
happens if you have a schedule element for say two days later, you switch
off your computer the current day
and restart the next day: will the schedule be saved or not

Short term solution will be the second option, i.e. allowing specifying an
event to occur at e.g. today at 27.30 hours

andimik, 2021-11-13:
Issue fixed with enhanced scheduler in 3bc6a93 which allows a scheduler 7 days in future including re-start function (untested, but sounds great). Thanks Jan!

closed 2021-11-13 by andimik