JvanKatwijk/qt-dab #243: Scanning

WIHDIA, 2021-11-12:
Some users report scanning stops at 8D.

JvanKatwijk, 2021-11-13:
It would be helpful to get some more details:
what is the scan mode, what is the skip file

closed 2021-11-14 by JvanKatwijk

andimik, 2021-11-14:
Well, my skip file is empty and I deleted all qt-dab related files under my home dir. But it always stops after the first found Austrian DAB mux (which I can receive at 6A and 8A) and the message `Scan ended` appears.

see following screenshot
[image]

As soon as I remove the antenna and plug in after this mux (8A) then it scans till 13F.

No matter if I save the scan or not.

Jan, in my opinion you should re-open that issue. **Because when the scan stops the result window keeps empty meaning that the content could not be saved to the file.**

see following screenshot
[image]

I fear that the special character (hash tag or German Umlaut) in the service names cause that issue. But the export using the button `content` works.

Could you check if the scan export differs from the content export?

```
DAB+ Austria    ; ensembleId A101; channel 6A; frequency 181936; time of recording  2021-nov-14  21:09; tii  (9+1)

Audio services
program name;country;serviceId;subchannelId;start address;length (CU); bit rate;DAB/DAB+; prot level; code rate; language; program type

#Technikum ONE  ;Austria;AD2A;7;336;60;80;DAB+;EEP 3-A;1/2;unknown language;Pop Music;
* 88.6 *        ;Austria;AC47;5;228;54;72;DAB+;EEP 3-A;1/2;unknown language;Rock Music;
-AUSTRIA        ;Austria;A3E0;16;726;54;72;DAB+;EEP 3-A;1/2;unknown language;Pop Music;
Antenne Österrei;Austria;AD57;14;618;54;72;DAB+;EEP 3-A;1/2;unknown language;Pop Music;
ENERGY          ;Austria;AC51;2;108;54;72;DAB+;EEP 3-A;1/2;unknown language;Pop Music;
ERF Plus        ;Austria;AD24;8;396;30;40;DAB+;EEP 3-A;1/2;unknown language;Religion;
KLASSIK RADIO   ;Austria;AD53;17;780;54;72;DAB+;EEP 3-A;1/2;unknown language;Light Classical;
Mein Kinderradio;Austria;AD55;12;534;30;40;DAB+;EEP 3-A;1/2;unknown language;Children's;
Radio Flamingo  ;Austria;AD5A;15;672;54;72;DAB+;EEP 3-A;1/2;unknown language;National Music;
Radio Maria     ;Austria;A3DD;1;54;54;72;DAB+;EEP 3-A;1/2;unknown language;Religion;
Rock Antenne    ;Austria;AD27;6;282;54;72;DAB+;EEP 3-A;1/2;unknown language;Rock Music;
Stephans Klassik;Austria;AC52;9;426;54;72;DAB+;EEP 3-A;1/2;unknown language;Classical Music;
WELLE 1         ;Austria;AD58;10;480;54;72;DAB+;EEP 3-A;1/2;unknown language;Pop Music;
arabella HOT    ;Austria;AD59;13;564;54;72;DAB+;EEP 3-A;1/2;unknown language;Pop Music;
arabella RELAX  ;Austria;AD54;0;0;54;72;DAB+;EEP 3-A;1/2;unknown language;Pop Music;
jö.live         ;Austria;AD56;3;162;54;72;DAB+;EEP 3-A;1/2;unknown language;Pop Music;

Data Services
program name;;serviceId;subchannelId;start address;length (CU); bit rate; FEC; prot level; appType ; subService ; 

EPG ORS         ;Austria;E0A0AD50;4;216;12;16;0;EEP 3-A;7;no;;
```

JvanKatwijk, 2021-11-15:
Did you try the most recent sources?
In handling a FIG 0/7 (i.e. nr services in the ensemble) the possibility of
scanning was not taken into
account. However, as far as I can see, it is now

Wrt scan export or content export:
I'll have a look this afternoon, they might differ though

andimik, 2021-11-15:
Now it scans till the end. I will test it in the evening with my roof antenna.

JvanKatwijk, 2021-11-15:
Good to hear. The issue was caused - as said - by being too naive in
handling FIG 0/7 (which does not
appear in the ensembles I receive)
To be sure, I checked the "content" output vs the :scan" output. The only
difference there is
that the period an ensemble is looked at during a scan is so short that not
all data is collected