JvanKatwijk/qt-dab #244: Build problem on RPI4 Buster

DABodr, 2021-11-21:
Error appearing after "make"

home/pi/dab/qt-dab/dab-mini/radio.cpp:1436:42: error: no matching function for call to ‘Scheduler::addRow(QString&, int, int)’
                           targetTime % 60);
                                          ^
In file included from /home/pi/dab/qt-dab/dab-mini/radio.h:46,
                 from /home/pi/dab/qt-dab/dab-mini/radio.cpp:38:
/home/pi/dab/qt-dab/dab-mini/../includes/support/scheduler.h:42:7: note: candidate: ‘void Scheduler::addRow(const QString&, int, int, int)’
  void addRow  (const QString &, int, int, int);
       ^~~~~~
/home/pi/dab/qt-dab/dab-mini/../includes/support/scheduler.h:42:7: note:   candidate expects 4 arguments, 3 provided
make[2]: *** [CMakeFiles/dabMini-2.0.dir/build.make:1048: CMakeFiles/dabMini-2.0.dir/radio.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:104: CMakeFiles/dabMini-2.0.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

JvanKatwijk, 2021-11-21:
Should be fixed, just forgit to upload a change to dab-mini relating to an
extesnion of the scheduler

petrkr, 2021-12-28:
still build problem on different source file

apt-get install qt5-qmake qt5-default libqt5opengl5-dev qtbase5-dev libqwt-qt5-dev libsndfile1-dev libfftw3-dev portaudio19-dev libfaad-dev zlib1g-dev rtl-sdr libusb-1.0-0-dev mesa-common-dev libgl1-mesa-dev libsamplerate-dev libiio-dev

cmake . -DTRY_EPG=ON -DRTLSDR=ON -DRTL_TCP=ON
make

```
[  1%] Building CXX object CMakeFiles/qt-dab-4.12.dir/home/pi/qt-dab/snr-viewer/snr-viewer.cpp.o
/home/pi/qt-dab/snr-viewer/snr-viewer.cpp: In constructor ‘snrViewer::snrViewer(RadioInterface*, QSettings*)’:
/home/pi/qt-dab/snr-viewer/snr-viewer.cpp:60:2: error: ‘snrDumpFile’ was not declared in this scope
  snrDumpFile. store (nullptr);
  ^~~~~~~~~~~
/home/pi/qt-dab/snr-viewer/snr-viewer.cpp:60:2: note: suggested alternative: ‘snrDumpButton’
  snrDumpFile. store (nullptr);
  ^~~~~~~~~~~
  snrDumpButton
/home/pi/qt-dab/snr-viewer/snr-viewer.cpp: In destructor ‘virtual snrViewer::~snrViewer()’:
/home/pi/qt-dab/snr-viewer/snr-viewer.cpp:114:2: error: ‘stopDumping’ was not declared in this scope
  stopDumping  ();
  ^~~~~~~~~~~
/home/pi/qt-dab/snr-viewer/snr-viewer.cpp: In member function ‘void snrViewer::handle_snrDumpButton()’:
/home/pi/qt-dab/snr-viewer/snr-viewer.cpp:243:6: error: ‘snrDumpFile’ was not declared in this scope
  if (snrDumpFile. load () != nullptr) {
      ^~~~~~~~~~~
/home/pi/qt-dab/snr-viewer/snr-viewer.cpp:243:6: note: suggested alternative: ‘snrDumpButton’
  if (snrDumpFile. load () != nullptr) {
      ^~~~~~~~~~~
      snrDumpButton
/home/pi/qt-dab/snr-viewer/snr-viewer.cpp:244:5: error: ‘stopDumping’ was not declared in this scope
     stopDumping ();
     ^~~~~~~~~~~
/home/pi/qt-dab/snr-viewer/snr-viewer.cpp:247:2: error: ‘startDumping’ was not declared in this scope
  startDumping ();
  ^~~~~~~~~~~~
/home/pi/qt-dab/snr-viewer/snr-viewer.cpp:247:2: note: suggested alternative: ‘startTimer’
  startDumping ();
  ^~~~~~~~~~~~
  startTimer
/home/pi/qt-dab/snr-viewer/snr-viewer.cpp: At global scope:
/home/pi/qt-dab/snr-viewer/snr-viewer.cpp:250:6: error: no declaration matches ‘void snrViewer::stopDumping()’
 void snrViewer::stopDumping () {
      ^~~~~~~~~
/home/pi/qt-dab/snr-viewer/snr-viewer.cpp:250:6: note: no functions named ‘void snrViewer::stopDumping()’
In file included from /home/pi/qt-dab/snr-viewer/snr-viewer.cpp:26:
/home/pi/qt-dab/snr-viewer/snr-viewer.h:54:7: note: ‘class snrViewer’ defined here
 class snrViewer: public QObject, Ui_snrWidget {
       ^~~~~~~~~
/home/pi/qt-dab/snr-viewer/snr-viewer.cpp:258:6: error: no declaration matches ‘void snrViewer::startDumping()’
 void snrViewer::startDumping () {
      ^~~~~~~~~
/home/pi/qt-dab/snr-viewer/snr-viewer.cpp:258:6: note: no functions named ‘void snrViewer::startDumping()’
In file included from /home/pi/qt-dab/snr-viewer/snr-viewer.cpp:26:
/home/pi/qt-dab/snr-viewer/snr-viewer.h:54:7: note: ‘class snrViewer’ defined here
 class snrViewer: public QObject, Ui_snrWidget {
       ^~~~~~~~~
make[2]: *** [CMakeFiles/qt-dab-4.12.dir/build.make:1333: CMakeFiles/qt-dab-4.12.dir/home/pi/qt-dab/snr-viewer/snr-viewer.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:104: CMakeFiles/qt-dab-4.12.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

JvanKatwijk, 2021-12-28:
Ybasicaes, I am changing a gew things and from time to time I have to
upload what I have
Basically a simple "ifdef" in the include file that wast not removed yet

I uploaded a version that with CMake will be compiled and linked (when
playing around here
I only use qmake)

petrkr, 2021-12-28:
I see now.  I tried tag 4.11 and that compile fine.

Regards

JvanKatwijk, 2021-12-28:
The 4.11 is stable, the current changes probably on an almost daily basis.
usually the qmake/make
route will work since that is how I test it (almost) daily)

The 4.12 "continuous" should not give no problems now
(few changes compared to 4.11, basically to the various displays)

petrkr, 2021-12-28:
I will test... On that 4.11 I had problem with disappearing config windows for RTL-SDR and RTL-TCP.. That windows just flash and dissapear. It does also on 4.12 AppImage on desktop. But sometimes it stay shown. I will try latest master if you not fix that meanwhile

JvanKatwijk, 2021-12-28:
Be aware that the main gui has a button labeled "device" that controls the
visibility of the device widget
If the "not visible"  mode is chosen, on program startup the widget will
flash

closed 2022-08-28 by JvanKatwijk