JvanKatwijk/qt-dab #245: RX-888

jjcarron, 2021-11-27:
Trying to start the application with ExtIO_sddc.dll for RX-888 issues an error message "cannot handle this rate" 

Any solution to solve it?

JvanKatwijk, 2021-11-27:
I do not know the RX-888, what is the rate?

jjcarron, 2021-11-27:
It is a relatively new 16-bit SDR. 

[image]

I have got the following error.

[image]

[image]

JvanKatwijk, 2021-11-27:
Sampling frequency seems a little too high, if I read it correctly, it is
64000000, while DAB needs something in
the range of 2 MHz (2.048 MHz to be precise, but from 2 .. app 4 or 5 MHz
we can resample)
Cant you change the sampling frequency?

jjcarron, 2021-11-27:
The lowest accepted is 50MHz 
When I try to apply a lower value. It set to 50MHz when I press apply

JvanKatwijk, 2021-11-27:
Interesting device, do you have a description of the library functions?

50Mhz would (almost) cover the whole DAB band (175 .. 240 MHz), so in order
for the data to be
useful you need some filtering.
If there is some filtering on say the center 2 or 3 MHz, then it is just a
matter of decimating the
samplerate to something that is better manageable, if there is no filtering
then a software filtering/decimation
would be needed, starting at 64 MHz, and a 32 times decimation results in 2
MHz, that is easy to
upsample to 2.048 MHz

jjcarron, 2021-11-27:
The github for the driver is :

jjcarron, 2021-11-27:
https://github.com/ik1xpv/ExtIO_sddc?spm=a2g0o.detail.1000023.1.3f982f06dfes3S

I am just starting with SDR and not sure to understand. WHen you say A software filtering is needed, you mean I shoud write a fonction in the driver to divide the actual sampling rate as I have no direct acces to the device itself, don't you?

JvanKatwijk, 2021-11-27:
Well, first of all I try to understand what the features/capabilities of
the device are. Most devices I have seen so far
have options to set the samplerate to a few MHz, some - not all - have the
possibility of filtering

Just thinking loud, if you start with a samplerate of 64 MHz, then
decimating 32 times is doable, though it would
be easier to address (for me) if I knew whether or not to take filtering
into account

If filtering is not provides by the device and you tune to say, a DAB
channel on 205 MHz, then you basically tune
to the whole band (205 - 32 to 205 + 32) and you get data from all channels
in the band, something you really
do not want

Filtering as such is not too complex, especially half band filtering, but
both 32 times decimating and filtering
is now not implremented in the extio handler and - suppose I will be able
to do that - I need that information

closed 2024-10-17 by JvanKatwijk