JvanKatwijk/qt-dab #247: -A paramer to specify user specified bands doesn't work

computer-wizard, 2021-12-25:
I  tried to use the  -A parameter for a user specified band but that doesn't work.
I even extracted the AppImage and executed the following,

 ./qt-dab -A ham
main: Detected system language "en_CA"
main: Set language "en_CA"
main: Error while loading language specifics "en_CA" use English "en_GB" instead
QObject::connect: No such slot snrViewer::setLength (int)
QObject::connect:  (sender name:   'snrLengthSelector')
Segmentation fault (core dumped)

 The 'ham' text file just contains,

tok  434000
   
qt-dab by itself without any parameter runs normally.
OS is Ubuntu 20.04 and qt-dab is the latest version.

JvanKatwijk, 2021-12-25:
fixed in appImage and in 4.12 windows installer

computer-wizard, 2021-12-25:
Thanks Jan!

closed 2021-12-25 by computer-wizard