JvanKatwijk/qt-dab #248: SDRplay support in AppImage

AsciiWolf, 2021-12-29:
I am trying to use the latest Qt-DAB AppImage with my SDRplay RSPdx (using API 3.07) on Fedora 35, however I get an `libsdrplay_api.so: cannot open shared object file: No such file or directory` error every time. The same thing happens even when I try to manually preload the actual libsdrplay_api library (`LD_PRELOAD=/usr/local/lib/libsdrplay_api.so ./Qt_DAB-x86_64.AppImage`). My SDR works fine in other software such as Gqrx. Are SDRplay devices actually supported by Qt-DAB?

JvanKatwijk, 2021-12-29:
SDRplay devices are well supported, in 99 percent of the cases I use an
SDRplay

On my laptop I am using fedora 35 and for supporting API 3.07 I installed
the officlal SDRplay API, which gives libsdrplay_api.so.3.07
in the /usr/local/lib directory.
For supporting the 2.13 interface library, I have libsdrplay_api.so
installed (the installer can still be downloaded from SDRplay.com, under
the heading
Legacy software  Click here to show
If a library such as libsdrplay_api.so,3,07 cannot be found it might be the
case the the /usr/local/lib directory is not in the searchpath, It is not
in the searchpath by default, you have to add an entry in /etc/ld.so.conf.d/

hope this helps

AsciiWolf, 2021-12-29:
Hmm, maybe something is wrong on my system/configuration, although it is strange that other SDR software works fine. Anyway, if SDRplay devices work fine in Qt-DAB, feel free to close this ticket.

AsciiWolf, 2022-02-27:
Ah, I probably just realized where is the problem. I have only the 3.07 api and libsdrplay_api.so is a symlink to libsdrplay_api.so.3.07 on my machine. Any chance Qt-DAB could also support the v3 api? :-)

AsciiWolf, 2022-03-09:
Turns out it was Fedora issue (feature): `/usr/local/lib` not in ldconfig library path. It works fine now after adding:

```
# cat /etc/ld.so.conf.d/local.conf
/usr/local/lib
/usr/local/lib64
```

And running `sudo ldconfig`.

closed 2022-03-09 by AsciiWolf