JvanKatwijk/qt-dab #249: MOT Slide show doesn't save picture or shows in separate widget

computer-wizard, 2021-12-31:
OS: Ubuntu 20.04 & latest qt-dab
I included a MOT Slide show  in my DAB transmission, (Only one jpg) but it isn't  saved in /tmp/qt-pictures.
or when I set 'MOT slides on separate widget' in the config, it doesn't show.  The MOT Decoding bar is 
clearly green when  I include the slide show in DAB.  I'm using the mmb tools to generate DAB.

andimik, 2021-12-31:
In general, this is working like a charme. Qt-DAB saves both, SPI slides and MOT slides.

Can you share a raw file or an eti file? Then I could investigate if the problem is your self created transmission.

Do other programs / DAB radios display the SLS at all?

computer-wizard, 2022-01-01:
OK, thanks. The problem seems to be here as I just checked with dablin_gtk, what also can display slides.
I first will investigate what the problem here is and at least I know that Qt-DAB should display SLS.
I will close this as it is not issue with Qt-DAB. tnx agn.

computer-wizard, 2022-01-02:
The problem was that I did forget to add the signalling in the mux file, Now the picture shows for dab in dablin and in
sdrangel but not Qt-DAB

JvanKatwijk, 2022-01-02:
pls send me (a link to) the file  that you use as input to Qt-DAB, then I
will have a look
jan

computer-wizard, 2022-01-02:
Is in the mail

closed 2022-08-12 by JvanKatwijk