JvanKatwijk/qt-dab #250: libfaad2_2.9.1-1 in Ubuntu 20.04 breaks some DAB+ stations

sjlongland, 2022-01-04:
Noticed this today with Triple M Classic Rock (Brisbane DAB+ Channel 1)…

```
Warning: Unexpected channel configuration change
Warning: Unexpected channel configuration change
Warning: Unexpected channel configuration change
Warning: Unexpected channel configuration change
Warning: Unexpected channel configuration change
Warning: Unexpected channel configuration change
Warning: Unexpected channel configuration change
Warning: Unexpected channel configuration change
```

That gets emitted a lot, you get a quick sub-second burst of audio then silence.  It turns out to be related to https://github.com/knik0/faad2/pull/51 and is actually, not a bug in `qt-dab` itself, but I'll document it here in case others encounter the same problem.

The fix is simple enough.  First, clone the repository that has the fix (you do not need the whole thing, just the last few commits), then check the last commit:

```
stuartl@vk4msl-ws:/tmp$ git clone --depth=10 -b PS https://github.com/fcartegnie/faad2.git
Cloning into 'faad2'...
remote: Enumerating objects: 271, done.
remote: Counting objects: 100% (271/271), done.
remote: Compressing objects: 100% (233/233), done.
remote: Total 271 (delta 94), reused 119 (delta 34), pack-reused 0
Receiving objects: 100% (271/271), 852.14 KiB | 2.27 MiB/s, done.
Resolving deltas: 100% (94/94), done.
stuartl@vk4msl-ws:/tmp$ cd faad2
stuartl@vk4msl-ws:/tmp/faad2$ git log -1
commit c31cd53969e5fe0ef27c32a8316037c5e1b6e2b1 (HEAD -> PS, origin/PS)
Author: Francois Cartegnie <fcvlcdev@free.fr>
Date:   Mon May 4 09:48:30 2020 +0200

    Unbreak PS audio
    
    This reverts commit 3b80a57483a6bc822d3ce3cc640fa81737a87c54.
```

If you see the same commit, you're good to go, dump that one change out to a file:

```
stuartl@vk4msl-ws:/tmp/faad2$ git format-patch HEAD^
0001-Unbreak-PS-audio.patch
```

Now, go up a directory and run `apt-get source libfaad2` to fetch the broken Ubuntu package:
```
stuartl@vk4msl-ws:/tmp$ apt-get source libfaad2     
Reading package lists... Done
Picking 'faad2' as source package instead of 'libfaad2'
NOTICE: 'faad2' packaging is maintained in the 'Git' version control system at:
https://salsa.debian.org/multimedia-team/faad2.git
Please use:
git clone https://salsa.debian.org/multimedia-team/faad2.git
to retrieve the latest (possibly unreleased) updates to the package.
Need to get 813 kB of source archives.
Get:1 http://au.archive.ubuntu.com/ubuntu focal/universe faad2 2.9.1-1 (dsc) [2,097 B]
Get:2 http://au.archive.ubuntu.com/ubuntu focal/universe faad2 2.9.1-1 (tar) [803 kB]
Get:3 http://au.archive.ubuntu.com/ubuntu focal/universe faad2 2.9.1-1 (diff) [7,904 B]
Fetched 813 kB in 2s (511 kB/s)
dpkg-source: info: extracting faad2 in faad2-2.9.1
dpkg-source: info: unpacking faad2_2.9.1.orig.tar.gz
dpkg-source: info: unpacking faad2_2.9.1-1.debian.tar.xz
stuartl@vk4msl-ws:/tmp$ cd faad2-2.9.1
```

Apply the fix:
```
stuartl@vk4msl-ws:/tmp/faad2-2.9.1$ patch -p1 < ../faad2/0001-Unbreak-PS-audio.patch 
patching file libfaad/specrec.c
```

Then build the `.deb` package:
```
stuartl@vk4msl-ws:/tmp/faad2-2.9.1$ dpkg-buildpackage -us -uc -b
dpkg-buildpackage: info: source package faad2
dpkg-buildpackage: info: source version 2.9.1-1
dpkg-buildpackage: info: source distribution unstable
dpkg-buildpackage: info: source changed by Fabian Greffrath <fabian@debian.org>
dpkg-buildpackage: info: host architecture amd64
…
dpkg-deb: building package 'libfaad2' in '../libfaad2_2.9.1-1_amd64.deb'.
dpkg-deb: building package 'faad' in '../faad_2.9.1-1_amd64.deb'.
dpkg-deb: building package 'libfaad-dev' in '../libfaad-dev_2.9.1-1_amd64.deb'.
dpkg-deb: building package 'faad-dbgsym' in 'debian/.debhelper/scratch-space/build-faad/faad-dbgsym_2.9.1-1_amd64.deb'.
dpkg-deb: building package 'libfaad2-dbgsym' in 'debian/.debhelper/scratch-space/build-libfaad2/libfaad2-dbgsym_2.9.1-1_amd64.deb'.
        Renaming faad-dbgsym_2.9.1-1_amd64.deb to faad-dbgsym_2.9.1-1_amd64.ddeb
        Renaming libfaad2-dbgsym_2.9.1-1_amd64.deb to libfaad2-dbgsym_2.9.1-1_amd64.ddeb
 dpkg-genbuildinfo --build=binary
 dpkg-genchanges --build=binary >../faad2_2.9.1-1_amd64.changes
dpkg-genchanges: info: binary-only upload (no source code included)
 dpkg-source --after-build .
dpkg-buildpackage: info: binary-only upload (no source included)
```

Now, you should be able to run `sudo dpkg -i ../faad_2.9.1-1_amd64.deb ../libfaad-dev_2.9.1-1_amd64.deb`.  Re-start `qt-dab` (you do not need to re-compile as the binary should be dynamically linked), and you should find your once unreceivable station is now receivable once more.

closed 2022-01-04 by sjlongland

JvanKatwijk, 2022-01-04:
> silence. It turns out to be related to knik0/faad2#51

Stuart

I just installed 20.10 in a VM and created a Qt-DAB executable (could not
find 20.04)
20.10 installs with faad 2.10.0-1 and that does not seem to give problems,

sjlongland, 2022-01-04:
Yeah, it seems to depend on the particular station.  I had 4KQ playing all morning and no issues, but the moment I switched over to the other network, that's what happened.  Switch back to 4KQ, no problem.

It's related to how the particular station encodes their AAC stream, with Parametric Stereo being problematic.

My desktop is an Ubuntu 18.04 LTS install updated to 20.04 via `do-release-upgrade`.

Images for 20.04 can be found here: http://archive.ubuntu.com/ubuntu/dists/focal/main/installer-amd64/current/legacy-images/

JvanKatwijk, 2022-01-06:
I installed a 20.04 in a VM and ran Qt-DAB with all ensembles but no
problems found here.

Can you send a link to a (fragment of) a recording, then I can see what is
going on and I can test with (a) other versions of faad and )b) fdk-aac

sjlongland, 2022-01-08:
Tried uploading the `.aac` dump here but Github doesn't like raw AAC files.

For now I've put a small snippet here, but for copyright reasons, I won't be able to keep it there.
http://static.vk4msl.id.au/github/JvanKatwijk/qt-dab/issues/250/

As I say, I don't think there's much that `qt-dab` can do, it's a `libfaad2` bug, but I'm just documenting the situation here in case other users run up against the same problem.

andimik, 2022-01-08:
@sjlongland

If you zip the file you can add it up to the limit of - I think - 10 Megabytes.

sjlongland, 2022-01-08:
@andimik Yeah I thought about doing that… but the multiplex dump was ~230MB and likely, not very compressible.

The other hack might've been to simply rename the file and provide an instruction to rename it back… but hosting it elsewhere is probably for the better for copyright reasons (there's about 10~15 seconds of audio for every station in the multiplex, most of which were playing copyrighted music at the time).  If there's a DMCA takedown notice, it'll be directed at me and not the project or Github. :-)

andimik, 2022-01-08:
@sjlongland

Do you have the same issues with these stations? 

- 4KQ Classic Hits
- 97 3 Brisbane
- CW Remix
- News Talk Sport

sjlongland, 2022-01-08:
Okay, I just fired up the latest `qt-dab` with that `.sdr` file… and clicked on each station to see which ones would play.

Log file: [qt-dab.log](https://github.com/JvanKatwijk/qt-dab/files/7832930/qt-dab.log)

These ones were fine:
* 4KQ Classic Hits (1269)
* 4TAB 3           (1260)
* 4TAB 4           (1261)
* 97.3 80s         (126D)
* 97.3 90s         (126B)
* 97 3 Brisbane    (1268)
* Coles CBD        (1259)
* Coles Radio      (1255)
* Coles TAS        (1258)
* Nova1069         (1254)
* RadioTAB         (125E)
* Smooth 80s       (1256)
* Smooth FM        (1257)
* TikTok Trending  (126E)
* Triple A         (12CE)

These ones broke with `libfaad2-2.9.1-1` (http://archive.ubuntu.com/ubuntu/pool/universe/f/faad2/faad_2.9.1-1_amd64.deb):
* 4IP - 4TAB XMAS  (125F)
* 4RPH Digital     (12CF)
* 90s SOFT POP     (12A8)
* Christmas Hope   (12D0)
* CW Remix         (126C)
* Elf Radio        (1270)
* Global           (12CC)
* iHeartCountry    (126F)
* Inspire Digital  (12CD)
* MMM 80s          (12A5)
* MMM CLASSIC ROCK (12A6)
* Nova 90s         (125A)
* SoundCloud Radio (12A7)
* The Edge         (126A)

(To be honest, bit surprised to still see Christmas stations in January… then again, they usually spring up around October… so maybe they're just slow to pull them down.)

andimik, 2022-01-08:
@sjlongland

If yes, we already found it out a while ago.

https://github.com/Opendigitalradio/dablin/issues/62

https://github.com/Opendigitalradio/dablin/issues/61

You find the solution in https://github.com/Opendigitalradio/dablin/tree/next#ubuntu-2004-faad2-and-he-aac-v2-services

sjlongland, 2022-01-08:
@andimik Yep, I think actually that was what tipped me off to `libfaad2` being the culprit.  I did a search on the error message (as you do), and stumbled on that.

As I say, the purpose of filing this issue was to document the problem for others who use `qt-dab` and might hit the problem on Ubuntu 20.04. :-)