JvanKatwijk/qt-dab #251: Compilation error

Dieter9, 2022-01-08:
I have not been able to compile the source code for some time now. 
Both with qmake and with cmake compilation the following error is displayed:

```
In file included from ../src/scopes-qwt6/iqdisplay.cpp:24:
../includes/scopes-qwt6/spectrogramdata.h:81:13: error: virtual function ‘virtual QwtInterval SpectrogramData::interval(Qt::Axis) const’ overriding final function
    81 | QwtInterval interval (Qt::Axis x) const {
       |             ^~~~~~~~
```
My system is an up to date Arch Linux with qwt 6.20.

JvanKatwijk, 2022-01-08:
qwt 6.20 might be the issue. I got several reports on some inconsistencies
between 6.19  and 6.20
I got two conflictijng pull requests to "solve" it, the last one was
applied but apparently did not solve the issue

Tomorrow I'll upload in a VM a qwt6.20 to look into the issue

Dieter9, 2022-01-09:
Thanks for taking care of the problem so quickly.
But I am in doubt if it is really only due to the qwt version.

According to the update history of my computer, the qwt update was on 2021-10-11.
But my last successful translation of qt-dab-maxi was 2021-12-28.

Unfortunately, as Arch Linux has about 10 to 20 updates every day, I can't determine what else might be responsible.

JvanKatwijk, 2022-01-09:
I adapted the file spectrogram according to the first pull request I got,
and added ifdefs for an older version.
The result works fine on a VM with Fedora 35, and a home-built qwt6.2,
as well as on my Fedora 35 box with qwt 6.1.X and on a VM with Ubuntu 16
(from where appimages are built) with an even older qwt version

best
jan

gvanem, 2022-01-14:
> I adapted the file spectrogram 

Still not working. I compile with Qwt 6.3.0 and get these errors compiling `src/scopes-qwt6/iqdisplay.cpp`:
```
scopes-qwt6\spectrogramdata.h(71): error C3861: 'setInterval': identifier not found
scopes-qwt6\spectrogramdata.h(72): error C3861: 'setInterval': identifier not found
scopes-qwt6\spectrogramdata.h(73): error C3861: 'setInterval': identifier not found
```

JvanKatwijk, 2022-01-14:
well, then something went wrong, since the three lines should only be
included for qwt version smaller 6.2

You can comment them out
Tomorrow I'll see that the git file is updated

andimik, 2022-08-12:
@Dieter9

Is the issue still existing? If not, please close it.

I am regularly building under Ubuntu and have not seen the error yet.

JvanKatwijk, 2022-08-12:
I'll close it, as far as I know the code changes I made in january were sufficient to address the qwt version problem

closed 2022-08-12 by JvanKatwijk