JvanKatwijk/qt-dab #252: Release build(qt-dab4.3) is asking for an older libcurl

computer-wizard, 2022-01-21:
OS: Ubuntu 20.04
Release build (qt-dab4.3) (AppImage) is asking for libcurl3, doesn't accept libcurl4

./Qt_DAB-x86_64-4.3.AppImage -A ham.dab
/tmp/.mount_Qt_DABb35hfj/usr/bin/qt-dab: /usr/lib/x86_64-linux-gnu/libcurl.so.4: version `CURL_OPENSSL_3' not found (required by /tmp/.mount_Qt_DABb35hfj/usr/bin/qt-dab)

Note: libcurl4 is the default in Ubuntu 20.04. (libcurl3 is not available as a package in this Ubuntu version)

JvanKatwijk, 2022-01-21:
it works fine on my VM with Ubuntu 20. Did you change the protection to
executable?

JvanKatwijk, 2022-01-21:
ps I did a clean install of 20.04 on a VM

andimik, 2022-01-21:
@computer-wizard 

As far as I remember I also had no problems under 20.04 with the app Image.

computer-wizard, 2022-01-22:
Did set only the file permission to executable as usual after the download and nothing for the rest.
That library issue was not there with the continuous build.

On 2022-01-21 05:43, JvanKatwijk wrote:
> it works fine on my VM with Ubuntu 20. Did you change the protection to
> executable?
> Message ID: ***@***.***> 

-- 

When you change the way you look at things, the things you look at change 

Max Planck

JvanKatwijk, 2022-01-22:
Interesting problem
The required curl library is part of the AppImage with the 4.3 in the name,
so on loading the appImage
there is no need to look external to the appImagw
The continuous build does not have the part configured that needs a curl
library, so it is clear
that that sspecific problem does not appear, but the functionality of
mapping TII data to names
is not implemented there.

This afternoon, I'll install the curl 4 package on my 20.04 VM and see what
is happening

JvanKatwijk, 2022-01-22:
I cannot reproduce the phenomenon you describe, Installing libcurl4 on my
VM with 20.04 did not change the behaviour of the AppImage.

Back to the basics

a. you downloaded the AppImage with 4.3 in its name
b. you changed the protection to executable
c. you ran (tried to run) the AppImage which gave  the result you sketched

Taking these steps on my Fedorabox or Ubuntu VM's running Ubuntu 16 and 20
gave no problems
will you please repeat the steps

computer-wizard, 2022-01-23:
I did repeat it and have exactly the same result as before. Are you sure you don't
have a libcurl3 sitting in /usr/local of Ubuntu?  I tried to solved it by coping the libcurl3 library
and softlink to /usr/local/lib/x86_64-linux-gnu/ and that did solve it.  In practice, I can't use that
for the transmitter DAB software as that demands libcurl4. The problem is that Ubuntu did use
the same softlink name (libcurl.so.4) with libcurl4 or libcurl3. I would suggest to compile the
Release build in the future against libcurl4 for the AppImage.

On 2022-01-22 07:54, JvanKatwijk wrote:
> I cannot reproduce the phenomenon you describe, Installing libcurl4 on my
> VM with 20.04 did not change the behaviour of the AppImage.
>
> Back to the basics
>
> a. you downloaded the AppImage with 4.3 in [...]

When you change the way you look at things, the things you look at change 

Max Planck

JvanKatwijk, 2022-01-23:
Too make things more complicated the following observations
Qt-DAB - in the appImage is linked against libcurl4
libcurl4 itself is included in the appImage,

I removed all traces of libcurl and curl from my Ubuntu 20.04 VM see below

***@***.***:~$ apt list --installed |grep curl

WARNING: apt does not have a stable CLI interface. Use with caution in
scripts.

***@***.***:~$

and the appImage ran without problem

When running the appImage - on this VM - I get as externally loaded libs

***@***.***:~$ sudo pldd 1836
1836: /home/jan/Downloads/Qt_DAB-x86_64-4.3.AppImage
linux-vdso.so.1
/lib/x86_64-linux-gnu/libdl.so.2
/lib/x86_64-linux-gnu/libpthread.so.0
/lib/x86_64-linux-gnu/libz.so.1
/lib/x86_64-linux-gnu/libc.so.6
/lib64/ld-linux-x86-64.so.2
/lib/x86_64-linux-gnu/libfuse.so.2

So, I do not seem to find where the reference to libcurl3 is made

computer-wizard, 2022-01-24:
I see it and I really don't know what is causing it, very weird. If I just remove the softlink libcurl.so.4  of
libcurl4  at '/usr/lib/x86_64-linux-gnu/'  Qt_DAB-x86_64-4.3.AppImage will startup without any problem.
(libcurl3 was already removed). For the time being I just run the continuous build.

JvanKatwijk, 2022-01-24:
If you look at the libraries that are loaded by the appImage, they are:

838: /home/jan/Downloads/Qt_DAB-x86_64-4.3.AppImage
linux-vdso.so.1
/lib/x86_64-linux-gnu/libdl.so.2
/lib/x86_64-linux-gnu/libpthread.so.0
/lib/x86_64-linux-gnu/libz.so.1
/lib/x86_64-linux-gnu/libc.so.6
/lib64/ld-linux-x86-64.so.2
/lib/x86_64-linux-gnu/libfuse.so.2
***@***.***:~$

maybe in your case one of these needs the curl?

The continuous build differs from the 4.3 build is that the latter contains
the curl library for loading a database
apart from that there is no difference

computer-wizard, 2022-01-25:
I already researched those libraries but it has nothing to do with that. I checked
the dynamic section of the qt-dab binary with 'readelf -d qt-dab'

 0x000000000000001d (RUNPATH)            Library runpath: [$ORIGIN/../lib]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libportaudio.so.2]
 0x0000000000000001 (NEEDED)             Shared library: [libdl.so.2]
 0x0000000000000001 (NEEDED)             Shared library: [libqwt-qt5.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libfaad.so.2]
 0x0000000000000001 (NEEDED)             Shared library: [libcurl.so.4]
 0x0000000000000001 (NEEDED)             Shared library: [libfftw3f.so.3]
 0x0000000000000001 (NEEDED)             Shared library: [libsndfile.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libsamplerate.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libQt5Widgets.so.5]
 0x0000000000000001 (NEEDED)             Shared library: [libQt5Gui.so.5]
 0x0000000000000001 (NEEDED)             Shared library: [libQt5Xml.so.5]
 0x0000000000000001 (NEEDED)             Shared library: [libQt5Network.so.5]
 0x0000000000000001 (NEEDED)             Shared library: [libQt5Core.so.5]
 0x0000000000000001 (NEEDED)             Shared library: [libstdc++.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libgcc_s.so.1]

The RUNPATH (top line) setting has lower priority as the LD_LIBRARY_PATH environment variable.

  * *|RPATH|* - a list of directories which is linked into the executable, supported on most UNIX systems. It is ignored if |RUNPATH| is present.
  * *|LD_LIBRARY_PATH|* - an environment variable which holds a list of directories
  * *|RUNPATH|* - same as |RPATH|, but searched /after/ |LD_LIBRARY_PATH|, supported only on most recent UNIX systems, e.g. on most current Linux systems
  * *|/etc/ld.so.conf|* - configuration file for |ld.so| which lists additional library directories
  * builtin directories - basically |/lib| and |/usr/lib|

That is why the OS system libcurl.so.4 was checked first. I can change 'RUNPATH' into 'RPATH' in the binary with pathelf but that is
no solution either, besides RPATH is depreciated.

B

JvanKatwijk, 2022-01-25:
The appImage was built on Ubuntu 16. The packaging is done by linuxdeployqt.
If it would help I can build an AppImage on Ubuntu 18

JvanKatwijk, 2022-01-25:
But your list does not show why libcurl3 was referred to

computer-wizard, 2022-01-25:
You can try but I don't think it will make a difference as the runpath priority
remains the same.

JvanKatwijk, 2022-01-25:
as far as I understand, linuxdeployqt collects all required libraries -
based on the qt-dab executable, then a
patch is made to redirect the dependencies, leading to a directory appdir
The final squashing will make it into a mountable fs

But from where is there a reference to the curl3?

Btw, the non squashed tree contains all requirements to run the program
(you want me to send you such a tree),

computer-wizard, 2022-01-25:
No, as I made that conclusion on the reported error as it couldn't find version 'CURL_OPENSSL_3' in first instance.

version 3 has a symbol called CURL_OPENSSL_3 

version 4 has a symbol called CURL_OPENSSL_4

In the binary you can see that with readelf -s qt-dab | grep CURL

   41: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND ***@***.***_OPENSSL_3 (9)
    61: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND ***@***.***_OPENSSL_3 (9)
    88: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND ***@***.***_OPENSSL_3 (9)
   335: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND ***@***.***_OPENSSL_3 (9)
   338: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND ***@***.***_OPENSSL_3 (9)
   601: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND ***@***.***_OPENSSL_3 (9)
   724: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND ***@***.***_OPENSSL_3 (9)
  1155: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND curl_global_init@@CURL_OP
  1206: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND curl_easy_perform@@CURL_O
  1279: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND curl_global_cleanup@@CURL
  1906: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND curl_easy_setopt@@CURL_OP
  1914: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND curl_easy_init@@CURL_OPEN
  2611: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND curl_easy_strerror@@CURL_
  2922: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND curl_easy_cleanup@@CURL_O

or use nm -gD libcurl.so.4 | grep CURL on the library included with the AppImage,

0000000000000000 A CURL_OPENSSL_3

Ubuntu 20.04 with libcurl4,

 nm -gD /usr/lib/x86_64-linux-gnu/libcurl.so.4 | grep CUR

0000000000000000 A CURL_OPENSSL_4

computer-wizard, 2022-01-26:
Yes, It would me interesting if you can send such a tree.

On 2022-01-25 08:02, JvanKatwijk wrote:
> Btw, the non squashed tree contains all requirements to run the program
> (you want me to send you such a tree),
>
>
When you change the way you look at things, the things you look at change 

Max Planck

computer-wizard, 2022-01-26:
> summarizing:
>
> the problem is caused by instructing the loader to give preference to
> libraries that are installed on the user's system
> rather than to libraries enclosed in the AppImage
>
> Installing a modern curl library on the system where the appImage is built
> might help for this particular problem
> but the fundamental issue, i.e. giving preference to non-enclosed libraries
> may - somewhere - give problems
>
> I'll notify the constructor of the appImage packager, probonopd

Yes, because the RUNPATH setting in the ELF binary has a lower preference as LD_LIBRARY_PATH

Thanks!

JvanKatwijk, 2022-01-26:
https://www.dropbox.com/s/8kl978gfw2w6os5/appdir.tgz?dl=0

This is tree, toplevel is AppRun, but you have to change the protection for
usr/bin/qtdab in that tree to executable

JvanKatwijk, 2022-02-03:
The creator of the appImage tools wrote:

Maybe the user has set LD_LIBRARY_PATH on his system? What happens if he
runs

unset LD_LIBRARY_PATH
./Qt_DAB-x86_64-4.3.AppImage

computer-wizard, 2022-02-04:
It starts up when LD_LIBRARY_PATH is unset, what is pretty logical of course.

Thanks for appdir.tgz by the way!

closed 2022-08-12 by JvanKatwijk