JvanKatwijk/qt-dab #254: Choppy sound

gvanem, 2022-01-26:
I've built DAB-Mini using MSVC and clang-cl on Windows
This is a clip from **The Rolling Stones** "*You Can't Always get What You Want*":
* http://www.watt-32.net/dx/DAB-mini-P6-Rock.mp3

The sound is choppy as hell and sounds like crap.

Captured using Audacity. Tuned to *P6 Rock*, F216 at 229.072 MHz. 
SNR approx. 20 dB. Got a sync after approx. 4sec. Which I believe somewhat is good. 
And BTW, that same station in Welle.IO gives excellent audio using the same RTLSDR stick (lock-in = 1 sec).

So there's some problem with PortAudio or something else.
* I've tried different audio-output devices, but they all sounds like crap.
* I tried building with or without a `-D__MSC_THREAD__`. No change. 

Where can I start digging into this?

gvanem, 2022-01-26:
I can add:
The audio is equally awful in DAB-maxi.
But I suspect this maybe is a libfaad issue since in DAB-maxi, it shows lots of these:
```
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
...
```

So, if my libfaad was built with `-DDRM` could it cause such choppy sound problems?

andimik, 2022-01-27:
Ok, let my try on P6 in the next days. I can receive it as well.

In the past I have listened to RIKS only with dablin_gtk, that is why I might not have noticed that.

The parameters are 64 kBits stereo and HE-AAC v1, which is used in many muxes all over Europe.

andimik, 2022-01-27:
https://github.com/AlbrechtL/welle.io/issues/314

I remember I had similar warnings when I wanted to play with DRMplus (which means https://de.wikipedia.org/wiki/Digital_Radio_Mondiale here, BTW).

gvanem, 2022-01-27:
@andimik You said `The hint on #314 (comment) did NOT help.` 

Meaning building libfaad2 with or without `-DDRM` is **not** the cause?

I built libfaad2 via `vcpkg install faad2:x86-windows`. Wish is really this [repo](https://github.com/knik0/faad2).

 I might have to delete and rebuild it or try the libfaad in Welle.IO.

gvanem, 2022-01-27:
I found the cause:

* I built libfaad2 using `vcpkg install faad2:x86-windows`, which installed a 
`$(VCPKG_ROOT)/installed/x86-windows/include/neaacdec.h`.
* But the old bundled header `../includes/backend/audio/neaacdec.h` was shadowing
my new one.

So, swapping the 2 `-I` paths solved it for both the compiler and the `moc` program.
The audio is now perfect!

closed 2022-01-27 by gvanem

andimik, 2022-01-27:
Oh, good. Thanks for clarifying that.