JvanKatwijk/qt-dab #257: qt-dab-4.0 crashes (Segmentation fault)

computer-wizard, 2022-03-18:
qt-dab-4.0 crashes (Segmentation fault) when the command line option -A 'file name' is used
OS: Ubuntu 20.04 LTS

 ./Qt_DAB-x86_64.AppImage -A ham.dab
main: Detected system language "en_CA"
main: Set language "en_CA"
main: Error while loading language specifics "en_CA" use English "en_GB" instead
Segmentation fault (core dumped)

Is starts normally without this option

JvanKatwijk, 2022-03-19:
the inputfile for -A should have the form
name  freq
name  freq
etc

where freq is something in KHz that the device understands

I do not have a problem with that, so show me the contents of the file used
for the -A option

computer-wizard, 2022-03-19:
This is the contents of my ham.dab file.

ve3tok1  431200
ve3tok2  434000
ve3tok3  440000
ve3tok4  910000
ve3tok5 1260000 
ve3tok6 2410000

computer-wizard, 2022-03-20:
This is how the system calls and signals (strace) dump ends. 

openat(AT_FDCWD, "ham.dab", O_RDONLY)   = 14
fstat(14, {st_mode=S_IFREG|0664, st_size=97, ...}) = 0
read(14, "ve3tok1  431200\nve3tok2  434000\n"..., 4096) = 97
read(14, "", 4096)                      = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
+++ killed by SIGSEGV (core dumped) +++
Segmentation fault (core dumped)

computer-wizard, 2022-03-20:
Sadly there are no debugging symbols otherwise I could run it with the GDP debugger to
give some more answers.

JvanKatwijk, 2022-03-20:
with me a file with that content works fine

JvanKatwijk, 2022-03-20:
The "reading in" of the file with user specified channel description is
simple, just a combination of getline and sscanf,

JvanKatwijk, 2022-03-20:
I ve been playing around with different input files for the -A option, but
so far everything worked fine. Since that does not help you
the following suggestions
a. the probably might fo something wrong when the line is too long.  So,
what linebreaks do you use
As said, the implementstion is straightforward, reading in a line and
scanning the line with sscanf,
the readline is not protected, a line too long might cayse an error

computer-wizard, 2022-03-20:
Just the usual Linux\Unix newline \n character (0A)

JvanKatwijk, 2022-03-21:
As said, there are two functions implementing the reading, a readline and a
sscanf.
Just for my understanding:
If you start the program froim the command line without the -A option,
everything works fine
If you start the program from the command line with the -A filename option,
then a crash occurs
Is the crash immediate, i.e. before the GUI shows or shows the GUI
-whatever short .
During processing the channel file,  for each line gthat is processed some
text is output, i.e.
***@***.*** dab-maxi]$ ./linux-bin/qt-dab-4.351 -A ../testband
adding ve3tok1 431200
adding ve3tok2 434000
adding ve3tok3 440000
adding ve3tok4 910000
adding ve3tok5 1260000
adding ve3tok6 2410000

is there any such output?

I assume you are not compiling the stuff yourself. If needed I'll create a
new instance of the 4.0 version with lots of additional print statements in
it
(I hate crashes that cannot be explained)

computer-wizard, 2022-03-21:
The crash is immediately, no GUI is shown.  I see nothing of the processing of that file with 'adding ve3tok1..;, etc.

main: Detected system language "en_CA"
main: Set language "en_CA"
main: Error while loading language specifics "en_CA" use English "en_GB" instead
Segmentation fault (core dumped)

above is all what I see.  

No, I didn't compile it myself.

On 2022-03-21 08:42, JvanKatwijk wrote:
> As said, there are two functions implementing the reading, a readline and a
> sscanf.
> Just for my understanding:
> If you start the program froim the command line without the -A option,
> everything works [...]
When you change the way you look at things, the things you look at change 

Max Planck

computer-wizard, 2022-03-25:
I'm running now version 4.35 and that runs fine with the -A option, so the crash seems to be solved in the latest Qt-dab version.
I first run unset LD_LIBRARY_PATH to prevent that LD_LIBRARY_PATH overrules the local path for AppImage and so the issue
with curl. You can close this Segmentation fault issue if you want so.

closed 2022-03-27 by JvanKatwijk

JvanKatwijk, 2022-10-11:
https://www.dropbox.com/s/f6sl73gm35b1rhg/appdir.zip?dl=0

As far as I remember I sent you a compressed appdir since there wer issues
with the curl library
Here is a new version with some additional print statements.
call AppRun -A xxxx