JvanKatwijk/qt-dab #258: DAB data question

Morrisman68, 2022-03-20:
Dear Jan,

I am writing some code to create tags for off-air recordings using BBC Radio 3 DAB data, and a couple of issues have arisen - any wise words would be most welcome!

1. Have you any idea of the encoding(s) used, as downstream Python code finds non-aliasable Unicode values?
2. Does DAB cmdline buffer data from the stream prior to output to stdout? In some cases there are delimiting newline characters, while not in others, which makes me wonder if the data stream is perhaps semi-synchronous and designed to work well with scrolling/marquee-style displays as found on many DAB receivers? I can supply logged data should it help.

Many thanks for all your previous help,

Ben

Morrisman68, 2022-03-21:
Applying hexdump to the data which caused Unicode problems I found this:

00000000  4e 6f 77 20 50 6c 61 79  69 6e 67 3a 20 52 69 63  |Now Playing: Ric|
00000010  68 61 72 64 20 57 61 67  6e 65 72 20 2d 20 54 61  |hard Wagner - Ta|
00000020  6e 6e 68 ef bf bd 75 73  65 72 20 28 4f 76 65 72  |nnh...user (Over|
00000030  74 75 72 65 29 0a 0a                              |ture)..|

I suspect that the prior translation of a-ummlaut to EFBFBD may indicate a cast from bytes to string, looking at other posts.

Does this shed any light, I wonder?

Thanks,
Ben

JvanKatwijk, 2022-03-21:
Ben

There are several output sources. If you mean the dynamic label texts, they
are encoded in Utf8, and in Qt-DAB that is handled by Qt, in e.g. command
line version
it is not handled as far as I remember.

DL data is built up in small fractions, there is no waiting for newline
type characters. Qt-DAB passes QStrings from the pad decoder to the GUI,
dab-cmdline
uses callbacks to pass on segments.

Hope this helps

closed 2022-03-27 by JvanKatwijk