JvanKatwijk/qt-dab #259: freeze after writing EPG data

dwarning, 2022-03-23:
OS: Ubuntu 20.04 Mate on an old weak notebook.
Using DLF or DLF Kultur MUX 5C I got an program freeze after writing epg data:
`going to open Dlf             
we have now 2 backends running
writing epgdata to /tmp/w20220326dd230c0.EHB
`
This is replicable with the actual AppImage and with a self compiled binary from actual git.
Same effect starting qt_app with sudo.
No problem with transmitter e.g. Rockantenne from 5D w/o epg data.

JvanKatwijk, 2022-03-24:
I do not think the "old weak notebook" is the cause,
We have here a very limited epg channel (merely sending "0" bytes), so it
might be helpful
to record a few minutes so I can see what causes the behaviour

dwarning, 2022-03-24:
Thanks for answering.
I let him stay for 20 min, but nothing changed. Program freezes, no sound, CPU to 100 percent. Only kill possible.
This is shown in console:

`dirSize 1842, numObjects 39, segmentSize 949
xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? xx? `

Attached is the small file /tmp/w20220324dd230c0.EHB. 
Make it sense compile in debug mode? How?
[w20220324dd230c0.txt](https://github.com/JvanKatwijk/qt-dab/files/8344457/w20220324dd230c0.txt)
.

JvanKatwijk, 2022-03-24:
OK, apparently the data contains stuff not being recognized, it seems to be
something in MOT objects, I;ll have a look into the code where and when
an "xx?"  message is given

JvanKatwijk, 2022-03-24:
It seems you are running an RTLSDR device, the sequence "xx? "  is printed
- actually never seen that before - if the callback function
from the RTLSDR library is not able to collect the amount of samples that
it should.
So it might be a performance issue.

JvanKatwijk, 2022-03-24:
If you compile with EPG off, then no additional background threads are
started for handling EPG

dwarning, 2022-03-24:
Same behavior with an airspy mini and on a Intel® Core™ i5 CPU M 430 @ 2.27GHz × 4 and 8G RAM.
Do you provide an AppImage?

andimik, 2022-03-24:
@JvanKatwijk and @dwarning  

I confirm the freeze with German Bundesmux (I have some samples at home) with latest version 4.351. It's not a crash, because `gdb` shows no abnormalities. And it's definitely **not** the performance.

Hint: **The freeze does not show up in version 4.3 (see below table) or older**

[w20220305dd230c0.zip](https://github.com/JvanKatwijk/qt-dab/files/8345051/w20220305dd230c0.zip)

```
[New Thread 0x7fffb8cbe700 (LWP 970691)]
[New Thread 0x7fffb3fff700 (LWP 970692)]
[New Thread 0x7fffb37fe700 (LWP 970693)]
[New Thread 0x7fffc7fff700 (LWP 970694)]
[New Thread 0x7fffc6a3e700 (LWP 970696)]
[Thread 0x7fffb8cbe700 (LWP 970691) exited]
[Thread 0x7fffb3fff700 (LWP 970692) exited]
fileLength = -1
Period = 8000
[New Thread 0x7fffb3fff700 (LWP 970699)]
channel reset: all services will be stopped
[New Thread 0x7fffb8cbe700 (LWP 970700)]
going to open Dlf Kultur      
we have now 1 backends running
[New Thread 0x7fffb2ffd700 (LWP 970703)]
Starting hidden service EPG Deutschland 
going to open EPG Deutschland 
we have now 2 backends running
[Thread 0x7fffb37fe700 (LWP 970693) exited]
dirSize 1842, numObjects 39, segmentSize 949
going to write file /tmp/qt-pictures/Zeitfragen - Kultur & Geschichte
going to write file /tmp/qt-pictures/Dlf_112x32.png
going to write file /tmp/qt-pictures/Dlf_128x128.png
going to write file /tmp/qt-pictures/Dlf_320x240.png
going to write file /tmp/qt-pictures/Dlf_32x32.png
going to write file /tmp/qt-pictures/Dlf_DokDeb_112x32.png
going to write file /tmp/qt-pictures/Dlf_DokDeb_128x128.png
going to write file /tmp/qt-pictures/Dlf_DokDeb_320x240.png
going to write file /tmp/qt-pictures/Dlf_DokDeb_32x32.png
going to write file /tmp/qt-pictures/Dlf_DokDeb_600x600.png
going to write file /tmp/qt-pictures/Dlf_Kultur_112x32.png
going to write file /tmp/qt-pictures/Dlf_Kultur_128x128.png
going to write file /tmp/qt-pictures/Dlf_Kultur_32x32.png
going to write file /tmp/qt-pictures/Dlf_Kultur_600x600.png
going to write file /tmp/qt-pictures/Dlf_Nova_112x32.png
going to write file /tmp/qt-pictures/Dlf_Nova_128x128.png
going to write file /tmp/qt-pictures/Dlf_Nova_32x32.png
going to write file /tmp/qt-pictures/Dlf_Nova_600x600.png
writing epgdata to /tmp/w20220305dd230c0.EHB
[Thread 0x7fffb8cbe700 (LWP 970700) exited]
[Thread 0x7fffb3fff700 (LWP 970699) exited]
[Thread 0x7fffc4ffc700 (LWP 970675) exited]
[Thread 0x7fffc57fd700 (LWP 970674) exited]
[Thread 0x7fffe8fe9700 (LWP 970672) exited]
[Thread 0x7fffeaffd700 (LWP 970661) exited]
[Thread 0x7fffeb7fe700 (LWP 970660) exited]
[Thread 0x7fffebfff700 (LWP 970659) exited]
[Thread 0x7ffff21f3700 (LWP 970658) exited]
[Thread 0x7ffff342b780 (LWP 970654) exited]
[Thread 0x7fffc6a3e700 (LWP 970696) exited]
[Thread 0x7fffc7fff700 (LWP 970694) exited]
--Type <RET> for more, q to quit, c to continue without paging--

Program terminated with signal SIGKILL, Killed.
The program no longer exists.

```

In my case (using a rawfile as input), there are no `xx` on the console, but the term `xx` is in several handlers.

BTW: I also tried WDR, here it works.

```
going to open WDR-EPG         
we have now 1 backends running
dirSize 1724, numObjects 33, segmentSize 949
going to write file /tmp/qt-pictures/radio_wdr-wdr3_128x128v.png
writing epgdata to /tmp/w20211015dD392c0.EHB
filename = epg file 0
writing epgdata to /tmp/w20211015dD394c0.EHB
filename = epg file 1
writing epgdata to /tmp/w20211015dD496c0.EHB
filename = epg file 2
writing epgdata to /tmp/w20211015dD592c0.EHB
filename = epg file 3
writing epgdata to /tmp/w20211015dD792c0.EHB
filename = epg file 4
writing epgdata to /tmp/w20211015dDA92c0.EHB
filename = epg file 5
writing epgdata to /tmp/w20211015dDB92c0.EHB
filename = epg file 6
writing epgdata to /tmp/w20211015dDC92c0.EHB
filename = epg file 7
writing epgdata to /tmp/w20211015dDF95c0.EHB
filename = epg file 8
writing epgdata to /tmp/w20211015dDF96c0.EHB
filename = epg file 9
```

Please note the line `filename = epg file` which does not appear on the freezing Bundesmux.

And I tried older versions as well:

version | freeze? |
----- | ---
4.351 | yes |
4.35 | yes |
4.3 | no |

In **4.3** the output shows the 3 lines with the EPG SID for 0xD230, 0xD210 and again 0xD230 and continues with `Dlf_600x600.png` without a freeze.

```
going to write file /tmp/qt-pictures/Dlf_Nova_32x32.png
going to write file /tmp/qt-pictures/Dlf_Nova_600x600.png
currentSID = D230
currentSID = D210
currentSID = D230
going to write file /tmp/qt-pictures/Dlf_600x600.png
going to write file /tmp/qt-pictures/Dlf_Kultur_320x240.png
```

@dwarning 

You could download an older appImage in the meanwhile.
For example https://github.com/JvanKatwijk/qt-dab/releases/download/qt-dab-4.0/Qt_DAB-x86_64.AppImage

JvanKatwijk, 2022-03-25:
I'll have a look. EPG interpretation changed in 4.35. A recording would be
helpful

JvanKatwijk, 2022-03-25:
It seemed I was a little lazy when writing  the interpreter for the EPG
data. There were a lot of loop constructions where in the loop the type was
dispatched and, depending on the type axction was taken. Unfortunately if -
in some cases - the type did not match any of the implemented types, the
dispatch did not led to actions and the loop was
restarted with the same input.
You could give this one a try
https://www.dropbox.com/s/s8o05ctsu95p1h3/Qt_DAB-x86_64-4.351.AppImage?dl=0

It is not in github yet since it might generate some unnecessary output

dwarning, 2022-03-25:
OK - I tried:

There is no freeze, program is gathering data.

But unfortunately after few minutes core dump.

Why not open a branch on github?

JvanKatwijk, 2022-03-25:
> I see. Surprised that you make it on master. Would be better create a
> development branch from master e.g.:
>
> git checkout -b epg_issues
>
> Later you can merge the commits into master, in case it works. If not
> the master is untouched.
>
> Outputs are not a problem.
>
> I made an experiment with valgrind to check the memory problem. But no
> luck, because the application is time critical. Another idea?

well, I am the only developer and I have pretty poos experinences with
"cooperation", so I stick to one branch, especially
since all code remains.
I always use the address santizer when running privately

dwarning, 2022-03-29:
OK.
BTW - I can't disable the EPG feature. Along manual it should disabled by default. But it isn't.
In dab-max enabling by `cmake .. -DTRY_EPG=ON` follows in same build (sources and executable) as with default cmake.
Do you plan to repair this program feature?

andimik, 2022-03-29:
@dwarning

Have you tried to delete the `#` in https://github.com/JvanKatwijk/qt-dab/blob/master/dab-maxi/CMakeLists.txt#L91 and  
https://github.com/JvanKatwijk/qt-dab/blob/master/dab-maxi/CMakeLists.txt#L93 ?

JvanKatwijk, 2022-03-29:
Is is still freezing? I sent  you a link to version that rould print out
where items in the EPG stream are not recognized (and some action
is taken then).

JvanKatwijk, 2022-03-29:
... and yes you can switch off the option, just look into the qt-dab.pro
file

JvanKatwijk, 2022-03-29:
Andreas,
Did you try the link I sent to see whether ot not the issue with
unrecognized EPG symbols was solved?

dwarning, 2022-03-29:
> @dwarning
> 
> Have you tried to delete the `#` in https://github.com/JvanKatwijk/qt-dab/blob/master/dab-maxi/CMakeLists.txt#L91 and https://github.com/JvanKatwijk/qt-dab/blob/master/dab-maxi/CMakeLists.txt#L93 ?

No, I haven't. The 3 line say, if TRY_EPG is not defined (default cmake) the macro is false and the epg code is not used. But as I found the code is used anytime.

dwarning, 2022-03-29:
> Is is still freezing? I sent you a link to version that rould print out where items in the EPG stream are not recognized (and some action is taken then). Op di 29 mrt. 2022 om 09:27 schreef dwarning ***@***.***>:
> […](#)
As I answered 4 days before the special AppImage works longer and printed out debug messages, but then core dump crash.
Do you want see the printout on stdout? Sorry, then you must provide this image again - I removed it.

dwarning, 2022-03-29:
> ... and yes you can switch off the option, just look into the qt-dab.pro file Op di 29 mrt. 2022 om 10:41 schreef jan van katwijk ***@***.***
Have these changes in qt-dab.pro also impact to cmake build?
Seems I have to use qmake.

JvanKatwijk, 2022-03-29:
Getting the printout would be very helpful in further development of the
EPG function (as said before, the NPO, i.e. Dutch radio, only sends MOT
objects
with zers as content).
Comparfed to the previously sent appImage, this one has one mod in the EPG
handling

https://www.dropbox.com/s/y4u0i8m7pti0i8m/Qt_DAB-x86_64-4.351.AppImage?dl=0

dwarning, 2022-03-29:
The crash (Speicherabzug) comes after 1-2 min.
Attached my console output in epg.log and the epg files from my tmp folder.
Hope it will help.

[epg.log](https://github.com/JvanKatwijk/qt-dab/files/8371410/epg.log)
[epg_files.zip](https://github.com/JvanKatwijk/qt-dab/files/8371421/epg_files.zip)

JvanKatwijk, 2022-03-29:
Thanks,

gives an idea where to look for the error

dwarning, 2022-03-30:
> Again, thank
>
> The output gave an indication of what caused the crash, together with
> input
> from andimik I used it to rework the parsing of the epg input,
> cutting off some processing of symbols the semantics of which is of no
> importance for the limited goals of an elementary EPG guide
>
> On the (few) input files with DAB data containing EPG data it all works
> fine,
> I would appreciate if you would give the new version a try
>
> https://www.dropbox.com/s/7msb6x58r649e37/Qt_DAB-x86_64-4.351.AppImage?dl=0

OK, the image works much better. No crash nor freeze.

Lots of files are gathered with program information's inside in /tmp 
folder. But I am missing the button in the detail menu. So I can't see 
the true EPG result.

I am wrong in the usage?

Thanks.

andimik, 2022-03-30:
Good news. There is no freeze with this new appImage.

andimik, 2022-03-30:
@dwarning 

The button is still there. I've tried with SWR (which I also receive via Satellite here in Austria)

I guess you have to switch the service resp. back to the service (if you were listening to Dlf, switch to Dlf Kultur and then back to Dlf). But I have not tested with Bundesmux! Note, it takes long to write the EPG on Bundesmux (check for the output `filename = epg file `! And only the three programms from Dlf (0xd210, 0xd220 and 0xd230) have an EPG.

[image]

It's too late to switch the satellite dish to Bundesmux and check it there. My recordings from the evening were too short (I'd better recorded for 10 min). Or maybe recordings do not work (time in the past) and I need to do it with a live signal.

dwarning, 2022-03-31:
Thanks both for your work.
After about 20min the situation is as shown in my screendump. There are tons of files generated in my /tmp folder and in the program starting folder, in this case my desktop. All they have program informations inside.
But as you see no button is generated, also after switching between DLF and DLFKultur.

@andimik : Did you use the same image like me, provided last by @JvanKatwijk ?

[image]

andimik, 2022-03-31:
@dwarning 

Yes, the same from his link

Let me try with a previous version on Bundesmux, but it is not so easy for me (need to run 2 PCs in parallel to get the signal). 

But I can't tell you if EPG on Dlf was ever working before.

andimik, 2022-03-31:
@dwarning

Could you also check the 17 EPG files in your home directory? Do they contain xml with EPG in the future?

dwarning, 2022-03-31:
Made an new trial from 10:50 ... 11:10 with 5C Bundesmux. First DLF, then DLF Kultur and DLFnova started in my home/tmp folder.
I think there are program information's for the future inside, I see 01. and 02.04.22.
Files attached.
[tmp.zip](https://github.com/JvanKatwijk/qt-dab/files/8387837/tmp.zip)
[home_tmp.zip](https://github.com/JvanKatwijk/qt-dab/files/8387839/home_tmp.zip)

JvanKatwijk, 2022-03-31:
Few comments

EPG is in DAN transmissions encoded as binary xml. I read and process the
binaty encoded xml and (try to) make simple time tables for the services
encountered.
EPG contains as such more info, but as said earlier, while last year the
NPO transmitted losts of data, including time tables and service related
data in the EPG labeled service, they now only send MOT objects with zeros
as data. So, using life data as test is currently impossible for me.

The EPG data - as received was saved in files - for some other processing.
Basic processing is done to collect descriptions of the
service items per service, and if a section id complete, a time table can
be made visible in the technical details widget.

The original EPG handler, derived from the one written by Julian Cable of
the BBC and taken from DRM, was used to generate the textual XML files.
To clean up things a little, the binary form of the EPG data is not written
out into a file anymore, andthe original EPG handler is called in to
generate the textual XML files in the /tmp/ directory (or whatever
directory you choose for "epgPath" in the ".ini" file.

Anyway, I am glad that it seems that processing EPG does not interfere with
other parts of the DAB processing.
attached is the link to the version where the above mentioned issues (plus
one related to the text of "long names") were addressed.

https://www.dropbox.com/s/mcb7tbq83smiz05/Qt_DAB-x86_64-4.351.AppImage?dl=0

If there are no more further comments, I'll upload the sources to github

best
jan

dwarning, 2022-04-05:
The AppImage works stable. Would be fine to upload the changes to gihub.
Thanks.

closed 2022-04-05 by dwarning