JvanKatwijk/qt-dab #260: Failed to open LIBIIO

DG9BFC, 2022-03-24:
Hello Jan
i do not know what is different on your pluto ... mine does not work with any version later as 3.72 (64 bit)
4.0, 4.3, 4.35 .. all  show same error 
failed to open libiio
so it seems as either your fw or your libioo (that you used when compiling the windows version) is different
3.72 64 bit (that i kept for safety) works with my pluto just fine
so there is a difference between 3.72 (64 bit) and 4.0 and later that let the pluto stop working
fw on pluto is revision 32 (in fact a 32-dirty by evariste that has additionally datv support plus controls a ptt relais when txing)
so ... with an rtl stick all your versions run fine ... with my pluto not :-( .. sniff
hopefully you have an idea?!?
if needed i can let you lock into my pluto remote... or send you the pluto fw used here (to try on your pluto)
greetz sigi dg9bfc

JvanKatwijk, 2022-03-24:
pluto does not give any problem when running qt-dab on windows (using the
4.35 version) om my computer
I did not change anything wrt devices the last year

JvanKatwijk, 2022-03-24:
qt-dab uses the libuuo.dll that is installed in Windows - and is found
through the registry.
In my windows system, the dll is installed in
***@***.*** OS]$ find . |grep libiio.dll
./Program Files (x86)/qt-dab-64/libiio.dll
./Users/JvanK/Downloads/libiio/libiio-0.21.g565bf68/MinGW32/libiio.dll
./Users/JvanK/Downloads/libiio/libiio-0.21.g565bf68/MinGW32/libiio.dll.a
./Windows/System32/libiio.dll
./Windows/SysWOW64/libiio.dll

I assume you have a libiio installed somewhere, but you can always copy it
to the installation folder for qt-dab (usually C:\Program Files
(x86)\qt-dab-xxx)

DG9BFC, 2022-03-24:
i do find different libioo dll in my pc ?!? 

one is found in sdr console (and sdrc v3 works fine) ... but that does not help with your software ... size 90k
one is found in satsagen ... does also not work with your soft ... size 64k
one is found in qt-dab-64 (version 3.72/64-bit) ... works fine in version 3.72 but does not work any version higher as 3.72 (tested 4.0, 4.3, 4.35) ... size roughly 624k
so ... maybe you can send me a dropbox link to a dll that works with your soft??

the libiio.dll is different in different software packages (as can be seen above)
it is not one single file that is installed in win system (at least i do not find one in win system32)
i do know that you did nothing over the last year 
thats why ii just kept 3.72 when using pluto and used 4.0 and later only with an rtl dongle ... it was the time when you dropped the 32 bit and 64bit thingy (only 64 bit worked here) ...
the 3.72 works fine ... later versins do not ... and in later versions i do not fnd a libiio.dll inside your install folder
depending what your soft does that dll may be different (as seen above there are at least 3 different ones here!!)
hmmm?!?!?
i really would like to have an actual verdion running with my pluto ... i will get a newer pluto soon and then test it with ORIGINAL fw from analog devices (here using evariste fw) ... i do not know where the problem is hiding ... but i can not use any newer version as that last 3.72 64 bit you once compiled for me
HELP PLEASE :-)

JvanKatwijk, 2022-03-25:
well, for pluto there is a standard library that can be installed on a
woindows system, so I assume you did not install that

You could give
https://www.dropbox.com/s/eae6ptzzlnxc178/setup-qt-dab4.35.exe?dl=0
this one a try, I added - as far as a I can see - the required libraries in
the package

DG9BFC, 2022-03-25:
your package (with libiio inside) does not work
also installed the "general" windows libiioo (found the installer in the net) ... installs the same libiio.dll in systems folder
still no joy ... grr
then i tested a few things here:
in the qt dab 3.72 ... i do find a libiio that is much bigger (as already reported) ... but ... now the funny story begins .. i can delete that libiio.dll out of the 3.72 folder ... AND IT STILL WORKS ?!?!? how can THAT be?? seems as windows loads the dll (any of them) and if that is inside the cpu the other dlls are not used?? (there must be a reason why i find several libiio.dll here on my pc)
hmm how can the 3.72 work without that libiio inside its folder???  (i guess there is another dll loaded already)
now i am more then puzzelt :-(
see screenshot ... i deleted (renamed plus saved elsewhere) the libiio.dll out of the 3.72 folder and it still works

[image]

but all of your newer versions ... do not ... not with the libiio inside your pgm folder and not with that libiio in win systems hmmm??? (i hate when such things happen)

JvanKatwijk, 2022-03-25:
Apparently you have the 64 bit version installed in the right place. I have
the libiio installed via the pluto provided installer and everything works
fine here,
I'll do some tests this afternoon

DG9BFC, 2022-03-25:
i did install pluto drivers (usb mk2blabla) ... i did also install also the general libiio (just in case its missing)
only thing that may be is different ... my apps are not in pgm folder but in c:/afu/ qt.....
lucky you add the version number to the foldername (and i could keep that working 3.72)
hopefully you find a solution
greetz sigi dg9bfc

DG9BFC, 2022-03-25:
sidenote ... the dll found in your 4.35 folder and the one that i find in windows are also a bit different in size
[image]

JvanKatwijk, 2022-03-25:
https://www.dropbox.com/s/mhzuc1ilpzulojc/setup-qt-dab4.35.exe?dl=0

OK, what I did was
a. I removed the officially installed "libiio" from my windows system
b. I made a list of all dependencies for libiio.dll and all dependent libs
are now included in the package as well

One thing: I assume you have installed libad9361, at least I have.

And with that qt-dab works like a charm with pluto

DG9BFC, 2022-03-25:
libad9361?? no i do not have that installed
THAT version seems to work (not perfect but works)
why i write not perfect? cause sometimes it suddenly stops ...
do i have to deinstall the libiio dll (the "genuine one" that i just added??

DG9BFC, 2022-03-25:
ok .. a few tests ... 3.72 64 bit works with its libiio (324k in size!!) or with genuine libio in windows (reinstalled libiio genuine dll
your last one works now with genuine libiio.dll in windows or with your dll inside pgm folder
but 3.72 runs a lot more stable (i can browse internet or write emails or whateven and listen to a signal just fine)
4.35 suddenly stops (no idea why) so it is not as good (or say as stable) as the old one
i try a few other things out here and report back ... at least it works now so we are on the right track
greetz sigi dg9bfc

DG9BFC, 2022-03-25:
i just have now the 4.3 running (with the lbioo inside folder and not with genuine libioo installed)
seems to be more stable (does not stop within a minute or two) so i guess we have it ..
will send another report in a few minutes when it still works

DG9BFC, 2022-03-25:
4.35 stops within a minute or two .. 4.3 runs fine

DG9BFC, 2022-07-03:
hello jan
story beginns (AGAIN!)
4.351 works (but i had to manually change the libioo in the folder)
4.4.2 does not
hmpf ... i hate when such things happen
with some try and error i finally could get it running
i had to copy some files from the 4.351 into the 4.4.2 folder
a bit older libiio.dll
libusb-1.0
libxml2
with dropping those files from the 4.351 folder ... pluto runs fine
with the files that were installed from your 442 install it says failed to open libiio

JvanKatwijk, 2022-07-04:
The libiio should have been installed under Windows, see the installtion instructions for libiio

closed 2022-07-04 by JvanKatwijk

DG9BFC, 2022-07-05:
it is ... but your soft still does not work without manually copy a few files in your installation folder ... ?1?

JvanKatwijk, 2022-10-11:
then you did not install the libiio properly. I removed all traces of
libiio from my windows, then reinstallled it and Qt-DAB just works fine

DG9BFC, 2022-10-11:
i do find half a dozen libiio files in my pc ;-)

JvanKatwijk, 2022-10-11:
I do not suppose you want me to debug your windows

JvanKatwijk, 2022-10-11:
OK, one more time

The Qt-DAB driver for the pluto can be part of the executable since in link
time it does not depend on libwhatever.
On selecting the pluto - with or without the transmit option - the software
will load the required functions from
the dll. That means that the Windows load system should be able to locate
the DLL, i.e. the DLL must be
stored in a directory (folders as you call them) the name of which is in
the searchpath of the loader

There are many places for that, installing libiio with the regular install
mechanism, on my system you'll find the dll
at the following places

./Users/JvanK/Downloads/libiio/libiio-0.21.g565bf68/MinGW32/libiio.dll
./Users/JvanK/Downloads/libiio/libiio-0.21.g565bf68/MinGW32/libiio.dll.a
./Windows/System32/libiio.dll
./Windows/SysWOW64/libiio.dll