JvanKatwijk/qt-dab #262: EPG experiences after last git commit

dwarning, 2022-04-10:
I made an new executable based on the last github source 4 days ago. The program runs stable also on german 5C mux:
`going to open Dlf             
we have now 1 backends running
Starting hidden service EPG Deutschland 
going to open EPG Deutschland 
we have now 2 backends running
dirSize 1842, numObjects 39, segmentSize 949
going to write file /tmp/qt-pictures/Die neue Platte
`
The folders /tmp/qt-pictures and /tmp/Temp-a6a581a8-1959-4eba-a8c1-957894d661c3 are created. But last one is empty also after 30min and longer.
In case you need support for further epg developments don't hesitate to contact me.
Thanks.

JvanKatwijk, 2022-04-11:
I uploaded the source sof the latest version, with two - small - errors
cotrrected. The AppImage and the Windows fsetup
are based on this one

JvanKatwijk, 2022-04-21:
> Hello,
> I am afraid that we are at same state as few weeks ago. It is Bundesmux 5C:
> stream opened stream started Buffersize = 5, filterSize 5 OK, functions
> seem to be loaded Found Rafael Micro R820T tuner [R82XX] PLL not locked!
> Supported gain values (29): 49,6 48,0 44,5 43,9 43,4 42,1 40,2 38,6 37,2
> 36,4 33,8 32,8 29,7 28,0 25,4 22,9 20,7 19,7 16,6 15,7 14,4 12,5 8,7 7,7
> 3,7 2,7 1,4 0,9 0,0 [R82XX] PLL not locked! Generic RTL2832U
> 77771111153705700 channel reset: all services will be stopped Allocating 15
> zero-copy buffers we have now as background services channel reset: all
> services will be stopped Starting hidden service EPG Deutschland going to
> open EPG Deutschland we have now 1 backends running going to open Dlf we
> have now 2 backends running dirSize 1665, numObjects 36, segmentSize 949
> going to write file /tmp/qt-pictures/Dlf_128x128.png going to write file
> /tmp/qt-pictures/Dlf_320x240.png going to write file
> /tmp/qt-pictures/Dlf_32x32.png going to write file
> /tmp/qt-pictures/Dlf_DokDeb_112x32.png going to write file
> /tmp/qt-pictures/Dlf_DokDeb_128x128.png going to write file
> /tmp/qt-pictures/Dlf_DokDeb_320x240.png going to write file
> /tmp/qt-pictures/Dlf_DokDeb_32x32.png going to write file
> /tmp/qt-pictures/Aus Kultur und Sozialwissenschaften going to write file
> /tmp/qt-pictures/Dlf_Kultur_112x32.png going to write file
> /tmp/qt-pictures/Dlf_Kultur_128x128.png going to write file
> /tmp/qt-pictures/Dlf_Kultur_320x240.png going to write file
> /tmp/qt-pictures/Dlf_Kultur_32x32.png going to write file
> /tmp/qt-pictures/Dlf_Kultur_600x600.png going to write file
> /tmp/qt-pictures/Dlf_Nova_128x128.png going to write file
> /tmp/qt-pictures/Dlf_Nova_320x240.png going to write file
> /tmp/qt-pictures/Dlf_Nova_32x32.png going to write file
> /tmp/qt-pictures/Nachrichten *** stack smashing detected ***: terminated
> Abgebrochen (Speicherabzug geschrieben)
> Sorry for the format - seems a CR LF problem.

does not sound good.
Just formy recollection, you are able to generate an executable?
Just for another prohect I added as configuration option
DEFINES+=__EPG_TRACE__ that - if enable will show (virtually alll) nodes
passing in the epg decoder.
Alternatively, disable the EPG by commenting out #CONFIG+= try-epg and make
a recording and send me a link

JvanKatwijk, 2022-04-22:
One comment

I assume that you have unset the possibility of generating the xml type
files (setting on the configuration widget). It is known
that in some cases that particular piece of software - imported - causes a
crash

dwarning, 2022-04-22:
I always use git generated builds. But working with cmake I have to add the define into Cmake.txt.
`#add_definitions (-D__MSC_THREAD__)	# uncomment for use for an RPI
add_definitions (-D__EPG_TRACE__)
`
I hope this is the correct output of epg trace, only last before crash:
encountering node 2c
process_time with length 11
encountering node 80
encountering node 81
encountering node 13
in process_mediaDescription with length 605
encountering node 1a
processs_shortName with length 130
encountering node 80
encountering node 1
encountering node 1b
process_longDescription with length 469
encountering node 80
encountering node 1
*** stack smashing detected ***: terminated
Abgebrochen (Speicherabzug geschrieben)

Don't know how to set/unset xml file type with cmake.

JvanKatwijk, 2022-04-22:
A fair chance that is was a buffer overflow on a buffer with an - apparenty
- too small staticallly determined size.
Node 1 handles strings and similar data.
Made it into a dynamic buffer

btw
the configuration widget - i.e. a widget of the running program - contains
a checkbox for setting or unsetting generating an xml file
from the epg data

dwarning, 2022-04-22:
Now it works - stable for longer time.
After activating epg xml the .EHB files are written and for few services of 5C I got the time table button with the expected time table.
Will check if timetable is shown without xml checkbox activation.
Many thanks.

JvanKatwijk, 2022-04-22:
time table and xml files are independent from each other.

closed 2022-08-12 by JvanKatwijk