JvanKatwijk/qt-dab #272: rtl-sdr won't change channels reliably on Ubuntu 22.04

sjlongland, 2022-08-27:
I've noticed lately, that Qt-DAB is unable to switch channels properly using the rtl-sdr unless I literally stop the input service (switch "dabstick" to "select input", then switch back), then change to the desired channel.

Lately, I've been using qt-dab as an alarm clock with automatic recording, and if something disrupts Qt-DAB, it just sits on channel 5A (which is not a DAB multiplex here in Brisbane).

I'm not sure if this is the OS (Ubuntu 22.04) or Qt-DAB right now.  If I start on the command line:

```
stuartl@vk4msl-ws:~$ projects/qt-dab/qt-dab-s5/linux-bin/qt-dab-5.0 
main: Detected system language "en_AU"
main: Set language "en_AU"
main: Error while loading language specifics "en_AU" use English "en_GB" instead
libtii-lib.so: cannot open shared object file: No such file or directory
/usr/local/lib/tii-lib.so: cannot open shared object file: No such file or directory
Library not loaded
init_tii_L not loaded
ALSA lib pcm_dsnoop.c:601:(snd_pcm_dsnoop_open) unable to open slave
ALSA lib pcm_dmix.c:1032:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm.c:2664:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2664:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2664:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm_oss.c:397:(_snd_pcm_oss_open) Cannot open device /dev/dsp
ALSA lib pcm_oss.c:397:(_snd_pcm_oss_open) Cannot open device /dev/dsp
ALSA lib confmisc.c:160:(snd_config_get_card) Invalid field card
ALSA lib pcm_usb_stream.c:482:(_snd_pcm_usb_stream_open) Invalid card 'card'
ALSA lib confmisc.c:160:(snd_config_get_card) Invalid field card
ALSA lib pcm_usb_stream.c:482:(_snd_pcm_usb_stream_open) Invalid card 'card'
ALSA lib pcm_dmix.c:1032:(snd_pcm_dmix_open) unable to open slave
Hostapis: 3

Api 0 is ALSA

Api 1 is OSS

Api 2 is JACK Audio Connection Kit

Investigating Device 0

 (output):item 1 wordt stream 0 (HDA ATI HDMI: 0 (hw:0,3))

Investigating Device 1

 (output):item 2 wordt stream 1 (HDA ATI HDMI: 1 (hw:0,7))

Investigating Device 2

 (output):item 3 wordt stream 2 (HDA ATI HDMI: 2 (hw:0,8))

Investigating Device 3

 (output):item 4 wordt stream 3 (HDA ATI HDMI: 3 (hw:0,9))

Investigating Device 4

 (output):item 5 wordt stream 4 (HDA ATI HDMI: 4 (hw:0,10))

Investigating Device 5

Investigating Device 6

 (output):item 6 wordt stream 6 (hdmi)

Investigating Device 7

 (output):item 7 wordt stream 7 (jack)

Investigating Device 8

 (output):item 8 wordt stream 8 (pipewire)

Investigating Device 9

 (output):item 9 wordt stream 9 (pulse)

Investigating Device 10

 (output):item 10 wordt stream 10 (default)

Investigating Device 11

Investigating Device 12

Investigating Device 13

Investigating Device 14

Investigating Device 15

 (output):item 11 wordt stream 15 (ATH-G1WL Analog Stereo)

Investigating Device 16

Investigating Device 17

Investigating Device 18

 (output):item 12 wordt stream 18 (PulseAudio Volume Control)

Investigating Device 19

added items to combobox
Suggested size for outputbuffer = 0
stream opened
stream started
Buffersize = 5, filterSize 5
OK, functions seem to be loaded
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Supported gain values (29): 49.6 48.0 44.5 43.9 43.4 42.1 40.2 38.6 37.2 36.4 33.8 32.8 29.7 28.0 25.4 22.9 20.7 19.7 16.6 15.7 14.4 12.5 8.7 7.7 3.7 2.7 1.4 0.9 0.0 
[R82XX] PLL not locked!
Realtek RTL2838UHIDIR 00000001
channel reset: all services will be stopped
Allocating 15 zero-copy buffers
```

It starts on channel 5A (which has nothing, not DAB).  Switch to channel 9B:

```
we have now as background services
serving the channel is set to false
rtlsdr_demod_write_reg failed with -9
r82xx_write: i2c wr failed=-9 reg=17 len=1
r82xx_set_freq: failed=-9
channel reset: all services will be stopped
```

Not sure if the `-9` is `-EBADF`.  I see it quite a lot, so not sure what this implies yet.

If I switch the "input" to "select input", then back again:

```
we have now as background services
serving the channel is set to false
device is deleted
unknown device, failing
we have now as background services
serving the channel is set to false
device is deleted
Buffersize = 5, filterSize 5
OK, functions seem to be loaded
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Supported gain values (29): 49.6 48.0 44.5 43.9 43.4 42.1 40.2 38.6 37.2 36.4 33.8 32.8 29.7 28.0 25.4 22.9 20.7 19.7 16.6 15.7 14.4 12.5 8.7 7.7 3.7 2.7 1.4 0.9 0.0 
[R82XX] PLL not locked!
Realtek RTL2838UHIDIR 00000001
channel reset: all services will be stopped
Allocating 15 zero-copy buffers
```

I can then choose a channel:

```
we have now as background services
serving the channel is set to false
rtlsdr_demod_write_reg failed with -9
r82xx_write: i2c wr failed=-9 reg=17 len=1
r82xx_set_freq: failed=-9
channel reset: all services will be stopped
we have now as background services
serving the channel is set to false
channel reset: all services will be stopped
```

I still see the `-9` error code, but _something_ worked because the station list gets populated, and I can choose a station.

Basically, every time I fire up `qt-dab`, I have to go through this dance.  I'm not quite sure where to go looking to troubleshoot this.

JvanKatwijk, 2022-08-27:
Stuart

I ran a quick test and I am afraid the behaviour on my (Fedora) box is
different from yours
When starting with an overaged stick I see the following

.....
suggested size for outputbuffer = 0
stream opened
stream started
Buffersize = 5, filterSize 5
OK, functions seem to be loaded
Detached kernel driver
Found Fitipower FC0013 tuner
Supported gain values (23): 19.7 19.1 18.8 18.6 18.4 18.2 18.1 17.9 7.1 7.0
6.8 6.7 6.5 6.3 6.1 5.8 -5.4 -5.8 -6.0 -6.3 -6.5 -7.3 -9.9
NOXON DAB Stick 0
channel reset: all services will be stopped
Alphen aan den Rijn/Cellnex Toren 19 km 49 °
going to select Radio 10 60s&70s
going to open Radio 10 60s&70s
we have now 1 backends running
Den Haag/Cellnex Toren 9 km 315 °
Alphen aan den Rijn/Cellnex Toren 19 km 49 °
we have now as background services
channel reset: all services will be stopped
Rotterdam/Cellnex Toren Waalhaven 16 km 176 °
Rotterdam/Cellnex Toren Waalhaven 16 km 176 °
going to open NPO Radio 4
we have now 1 backends running
going to write file /tmp/MOT/MOT04/slideshow/broadcast_loop_1661601605.png
Reattached kernel driver
.. end the radio silences
back in main program
It is done

On starting up the system it opens the channel that was recorded from the
previous program instantiation
and (manually) switching over to another channel worked fine.

What I read in your description is that betwen channel selections, the
driver should be completely deleted. Now
the point with the driver is that a small task running to handle the call
back function that is called by the system driver
remains active on closing a channel. The reason for not killing and
restarting that thread  was that that caused problems
while scanning: too often a thread was killed and initiated, leading to a
crash in the rtlsdr software

What I can do is re-create a driver where the support thread is killed
between "stop" and "start" of the driver,
just to experiment

sjlongland, 2022-08-27:
Okay, so it might be something Ubuntu are doing with their RTL-SDR support libraries if it's working okay in Fedora.  It shouldn't be necessary to completely detach the driver, but clearly I'm hitting some sort of edge case.

I haven't fired up in Gentoo for quite some time… but this gives me a good inkling about where to look. :-)

JvanKatwijk, 2022-08-28:
Attached there is the rtlsdr driver where on restart a "workerthread"  is
created and on "stopReader" the thread is killed
Compile it after uncommenting the __X_VERSION_ define somewhere at the top
and see if it changes somethings

closed 2024-10-17 by JvanKatwijk