JvanKatwijk/qt-dab #273: RTL-SDR

kens500, 2022-09-16:
On Fedora 36 I'm using an RTL-SDR without problem with the rtl-sdr driver from the repository with the GQRX application. 

I was hoping to try out qt-dab but that application does not find the RTL-SDR. 

lsusb shows it as "Realtek Semiconductor Corp. RTL2838 DVB-T" with a RTL2838UHIDIR chip. The r820t  and rtl2832 modules are loaded.

Any suggestions??

andimik, 2022-09-16:
Have you tried to use rtl_tcp?

JvanKatwijk, 2022-09-17:
If you are using Linux then some text is written if you start the appimage
from a command window, what does it say

kens500, 2022-09-17:
> Have you tried to use rtl_tcp?

Yes although I suspect I need to be running a programme that makes the rtl-sdr available over IP?

kens500, 2022-09-17:
--- If you are using Linux then some text is written if you start the appimage
--- from a command window, what does it say

SOLVED - It was failing to find librtlsdr.so but librtlsdr.so.0 was there. I simply copied librtlsdr.so.0 to  librtlsdr.so. I'll do it properly with a link later.

kens500, 2022-09-17:
That's an impressive bit of software. Thank you

Jeroen88, 2023-03-05:
@JvanKatwijk could you help me tto with the RTL2832 USB stick on Ubuntu? This is what I get when I start the software in a terminal:

```
main: Detected system language "en_US"
main: Set language "en_US"
main: Error while loading language specifics "en_US" use English "en_GB" instead
ALSA lib pcm_dsnoop.c:641:(snd_pcm_dsnoop_open) unable to open slave
ALSA lib pcm_dmix.c:1089:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_dmix.c:1089:(snd_pcm_dmix_open) unable to open slave
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Hostapis: 2

Api 0 is ALSA

Api 1 is OSS

Investigating Device 0

 (output):item 1 wordt stream 0 (HD-Audio Generic: HDMI 0 (hw:0,3))

Investigating Device 1

Investigating Device 2

Investigating Device 3

 (output):item 2 wordt stream 3 (hdmi)

Investigating Device 4

 (output):item 3 wordt stream 4 (pulse)

Investigating Device 5

 (output):item 4 wordt stream 5 (default)

added items to combobox
Suggested size for outputbuffer = 0
stream opened
stream started
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8526:(snd_pcm_recover) underrun occurred
unknown device, failing

```

The first thing I try is 'Select Input' but the RTL-SDR is not showing. Would be very kind if you can give me some guidance! :)

andimik, 2023-03-06:
What does `rtl_test` tell you? If you even not get any output, you are missing some packages, see manual. 

Are you using the AppImage or have you compiled it from the sources?

Jeroen88, 2023-03-06:
Hi @andimik 

```
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R828D tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
Allocating 15 zero-copy buffers
lost at least 36 bytes
```

I am using the AppImage. Do the ALSA and jack errors mean that something is wrong with the audio setup as well? Or is the software just trying different options? 

I am using a cheap Aliexpress clone and ordered an "official" device yesterday. But the device I have now is working with Universal Radio Hacker: I can get a spectrum and I can decode a FSK-2 packet, so the device seems to work although it does not show up in de 'device' tab of Qt-dab.

JvanKatwijk, 2023-03-06:
have you tried "dabstick"

Jeroen88, 2023-03-06:
Yes I tried it just now. After a few minutes one radio station pops up in the main window, RADIONL2 Zeeland, but no sound. Maybe I have two problems? Bad reception (antenna?????) and no the right audio setup?

Your help is appreciated!

JvanKatwijk, 2023-03-06:
Assuming you are running 5.2 you can see at the control/config widget the
successrate in decoding the FIC/FIG, if that is less than 100 percent
your feception is weak. Setting the "technical widget' visible by clikcing
on the "details" button you see the qualities of the decoding of the
selected service
Again, values less than 100 percent show missing parts in the audio stream

andimik, 2023-03-06:
> After a few minutes one radio station pops up

Where in NL are you living? 5B is transmitted in the South West only.

[image]

Have you tried NPO on 12C? This is available all over the country.

Jeroen88, 2023-03-06:
@andimik I live in the South West. The problem is: I can not select any station because no station s shown. Sometimes that one radio station pops up, after hitting scan, NPO is not shown.

Jeroen88, 2023-03-06:
@JvanKatwijk Frame, RS, AAC all zero.  MOT bar is red. I can not select any radio station...

In the 'history' I now see three stations: 5B Radio Rijnmond, 5B Omroep West, 5B VAHON Radio. If I click each it says: insufficient data for this program.

JvanKatwijk, 2023-03-06:
try channel 12C, that should be good. There are transmitters in Goes en
Toosendaal

andimik, 2023-03-06:
> NPO is not shown.

Even when you select 12C?

Jeroen88, 2023-03-06:
12C? Where can I select it...? The 'service' window is empty.  Although. now, by magic and clicking a few times on 'presets' 10 radio stations are appearing. If I select one, I can see it in the main window too, but no sound. I see a spectrum, also in the tii window.

JvanKatwijk, 2023-03-06:
Channel selector

Jeroen88, 2023-03-06:
I found it, if I select it, all my channels disappear and the window is above 'presets' is empty again.

WIHDIA, 2023-03-07:
Do check the settings.

[image]

73
Kind regards
Met vriendelijke groeten

Herman Wijnants

[My logblog](https://www.wijnants.info/fmdx/)
[Belgium editor fmlist.org](https://www.fmlist.org/)

Jeroen88, 2023-03-07:
@JvanKatwijk yes I do agree that I should read the manual and ofcourse I did. The problem is that I have no reference how it should work and what to expect because I did not have a normal operation yet, e.g. that the services should appear within seconds when you select a channel. In my case they do not appear, or a few after a long time. 

I still have two problems:

- After selecting channel 12C no services appear, not even after an hour
- At channel 5A sometimes a few services appear. If I left click one, the service appears in the main window but I hear no sound.

@WIHDIA I know how to select the device now. rtl_tcp works, and dabstick also I believe.

Thnx again for all the help!

andimik, 2023-03-07:
Please check the spectrum diagram. It will immediately show you if you need to improve a) the antenna or b) the settings (gain).