JvanKatwijk/qt-dab #276: qtdab4 vs qtdab5

yves2, 2022-10-07:
on ubuntu 22.04,
qtdab4 audio output works with «default» and with «pulse»
qtdab5 audio output nothing with «default »with «pulse» some flashs of sound  more and more distant
            i only get sound with «upmix» the level is lower and the delay is bigger or by direct access to the sound card
«upmix» does’nt work with qtdab4
thanks for this nice software

JvanKatwijk, 2022-10-07:
Interesting,
qt-dab4 and 5 shared the whole processing chain, including sound handling.
Anyway, I'll have a look

JvanKatwijk, 2022-10-07:
running some quick tests it shows that after stopping/restarting any of the
two programs the list of devices shown can be
different, and the behaviour of the selected devices differs between the
two invocations

The sound out is delegated to the portaudio library, so on program start
up, the qt-dab software asks the portaudio library
(after initializing of course) for a list of devices that (a) are output
oriented and (b) can handle 48000 samples/second
(portaudio has an API with sufficient functions for doing such an inquiry)
That list is added to the combobox in the qt-dab-x interface

Selecting a device is then merely stopping any previously selected device
and requesting portaudio to open the
selected device.
If portaudio does not give an error message, indicating that for whatever
reason the device could not be opened,
the audio data is just  passed on to portaudio.

So, while I can see differences in behavious between invocations of the two
programs, and even between
subsequent invocations of any of the two programs, there is right now not
much that I can do.

I realize the answer is not very satisfying but it is what it is

best
jan

JvanKatwijk, 2022-10-07:
PS

Having started the program from within a command window, then some data is
printed on that window on selecting an output device.
As you will notice, even when the text says that the "Stream is Opened",
indicating that portaudio sould open it,
there is no garantee that you will hear sound

closed 2024-10-17 by JvanKatwijk