JvanKatwijk/qt-dab #277: Seg fault for Airspy filterDepth value 0

andimik, 2022-11-16:
There is a seg fault when the Airspy filterDepth is set to 0 in the QSpinBox due to a floating point exception.

```
we have 1 devices
2 samplerates are supported
6000000 
3000000 
selected samplerate = 3000000
Buffersize = 9, filterSize 9
vgaGain 6, result 0
currentTab 0
channel reset: all services will be stopped
filter set on
Floating point exception (core dumped)
```
gdb says,

```
vgaGain 6, result 0
[New Thread 0x7fffad242640 (LWP 4735)]
[New Thread 0x7fffaca41640 (LWP 4736)]
currentTab 0
channel reset: all services will be stopped
[New Thread 0x7fff9ffff640 (LWP 4737)]
filter set on
[Thread 0x7fffeb7fe640 (LWP 4715) exited]

Thread 23 "qt-dab-4.4.2" received signal SIGFPE, Arithmetic exception.
[Switching to Thread 0x7fffad242640 (LWP 4735)]
0x00005555555d47ac in LowPassFIR::Pass (this=0x555556fed8e0, z=...) at ../../src/output/fir-filters.cpp:121
121		ip = (ip + 1) % filterSize;
(gdb) bt
#0  0x00005555555d47ac in LowPassFIR::Pass(std::complex<float>) (this=0x555556fed8e0, z=...) at ../../src/output/fir-filters.cpp:121
#1  0x0000555555618e38 in airspyHandler::data_available(void*, int) [clone .isra.0]
    (this=0x555556f9dbe0, buf=<optimized out>, buf_size=<optimized out>) at ../../qt-devices/airspy-handler/airspy-handler.cpp:419
#2  0x00005555555f6644 in airspyHandler::callback(airspy_transfer_t*) (transfer=<optimized out>)
    at ../../qt-devices/airspy-handler/airspy-handler.cpp:397
#3  airspyHandler::callback(airspy_transfer_t*) (transfer=<optimized out>) at ../../qt-devices/airspy-handler/airspy-handler.cpp:387
#4  0x00007fffbc11fcbb in  () at /lib/x86_64-linux-gnu/libairspy.so
```

The following patch will prevent the crash:

```
diff --git a/qt-devices/airspy-handler/airspy-widget.ui b/qt-devices/airspy-handler/airspy-widget.ui
index 3bf5e5d..e13bebe 100644
--- a/qt-devices/airspy-handler/airspy-widget.ui
+++ b/qt-devices/airspy-handler/airspy-widget.ui
@@ -537,6 +537,9 @@
      <height>31</height>
     </rect>
    </property>
+   <property name="minimum">
+    <number>1</number>
+   </property>
   </widget>
   <widget class="QCheckBox" name="filterSelector">
    <property name="geometry">
```

JvanKatwijk, 2022-11-17:
Yes I see,

the range for the filter spec should not include 0, or else if 0 calling
the filter should be prevented

JvanKatwijk, 2022-11-17:
sources are updated, pre-built versions will follow later on

andimik, 2022-11-17:
Fixed in https://github.com/JvanKatwijk/qt-dab/commit/05a341252087d5f66366b86fbe251909bc976a83

closed 2022-11-17 by andimik