JvanKatwijk/qt-dab #279: cmake error due to renamed directories

andimik, 2022-12-20:
Dear Jan,
 
you have renamed some directories, therefore `cmake` does not compile.

```
$ cmake .. -DRTLSDR=ON
CMake Warning (dev) in CMakeLists.txt:

[...]

-- Found PORTAUDIO: /usr/include  
-- Configuring done
CMake Error at CMakeLists.txt:924 (add_executable):
  Cannot find source file:

    ../qt-devices/rtlsdr-handler/rtlsdr-handler.cpp

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc

CMake Error at CMakeLists.txt:924 (add_executable):
  No SOURCES given to target: qt-dab-4.4.2

CMake Generate step failed.  Build files cannot be regenerated correctly.
```

As a workaround I just have renamed the directory `rtlsdr-handler-linux` into `rtlsdr-handler`.

Or the following patch helped:

```
diff --git a/qt-dab-s4/CMakeLists.txt b/qt-dab-s4/CMakeLists.txt
index 7497743..26c3a0e 100644
--- a/qt-dab-s4/CMakeLists.txt
+++ b/qt-dab-s4/CMakeLists.txt
@@ -762,31 +762,31 @@ endif ()
        if (RTLSDR)
           set (${objectName}_UIS
                 ${${objectName}_UIS}
-                ../qt-devices/rtlsdr-handler/rtlsdr-widget.ui
+                ../qt-devices/rtlsdr-handler-linux/rtlsdr-widget.ui
            )
 
           set (${objectName}_MOCS
                ${${objectName}_MOCS}
-               ../qt-devices/rtlsdr-handler/rtl-dongleselect.h
-               ../qt-devices/rtlsdr-handler/rtlsdr-handler.h
+               ../qt-devices/rtlsdr-handler-linux/rtl-dongleselect.h
+               ../qt-devices/rtlsdr-handler-linux/rtlsdr-handler.h
           )
 
           include_directories (
-               ../qt-devices/rtlsdr-handler/
+               ../qt-devices/rtlsdr-handler-linux/
           )
 
           set (${objectName}_HDRS
                ${${objectName}_HDRS}
-               ../qt-devices/rtlsdr-handler/rtlsdr-handler.h
-               ../qt-devices/rtlsdr-handler/rtl-dongleselect.h
-               ../qt-devices/rtlsdr-handler/rtl-sdr.h
-               ../qt-devices/rtlsdr-handler/rtl-sdr_export.h
+               ../qt-devices/rtlsdr-handler-linux/rtlsdr-handler.h
+               ../qt-devices/rtlsdr-handler-linux/rtl-dongleselect.h
+               ../qt-devices/rtlsdr-handler-linux/rtl-sdr.h
+               ../qt-devices/rtlsdr-handler-linux/rtl-sdr_export.h
           )
 
           set (${objectName}_SRCS
                ${${objectName}_SRCS}
-               ../qt-devices/rtlsdr-handler/rtlsdr-handler.cpp
-               ../qt-devices/rtlsdr-handler/rtl-dongleselect.cpp
+               ../qt-devices/rtlsdr-handler-linux/rtlsdr-handler.cpp
+               ../qt-devices/rtlsdr-handler-linux/rtl-dongleselect.cpp
           )
 
           add_definitions (-DHAVE_RTLSDR)
```

JvanKatwijk, 2022-12-20:
Hi Andreas

Yes As often, the windows interface for the dabsticks gave problems, so I
had to adapt it and
make separate versions for Linux and Windows.

I adapted the CMakeLists.txt files and added different entries, so now
RTLSDR_LINUX and RTLSDR_WIN
are available as keys

Thanks for noticing it,

best
jan

andimik, 2023-01-21:
Solved in 4799b64fd8c7a34269f5b43d7c26438c7ace4d91, therefore closed.

closed 2023-01-21 by andimik