JvanKatwijk/qt-dab #280: Repeated fragmented DL text

andimik, 2023-01-23:
Sometimes the DL shows too long texts. It happens when the first text was displayed incompletely.

see screenshot
[image]

```
6A.#Technikum ONE   2023-01-23 00:16:22  SA ETHERIDGE - LIKE THE WAY I DO auf Technikum ONE
6A.#Technikum ONE   2023-01-23 00:16:28  Sie hören: MELISSA ETHERIDGE - LIKE THE WAY I DO auf Technikum ONE
6A.#Technikum ONE   2023-01-23 00:16:37  IKE THE WAY I DO auf Technikum ONE
6A.#Technikum ONE   2023-01-23 00:18:55  Sie hören: MELISSA ETHERIDGE - LIKE THE WAY I DO auf Technikum ONEIKE THE WAY I DO auf Technikum ONE
6A.#Technikum ONE   2023-01-23 00:18:59  Sie hören: MELISSA ETHERIDGE - LIKE THE WAY I DO auf Technikum ONEIKE THE WAY I DO auf Technikum ONESA ETHERIDGE - LIKE THE WAY I DO auf Technikum ONE
6A.#Technikum ONE   2023-01-23 00:19:03  Sie hören: MELISSA ETHERIDGE - LIKE THE WAY I DO auf Technikum ONEIKE THE WAY I DO auf Technikum ONESA ETHERIDGE - LIKE THE WAY I DO auf Technikum ONESA ETHERIDGE - LIKE THE WAY I DO auf Technikum ONE
```

This screenshot shows the analysis in XPADxpert.

[image]

Here there is an sdr-file

https://we.tl/t-VpteykHEPv

JvanKatwijk, 2023-01-25:
I see what you mean.
The question is whether I (i.e. the software) can detect that the text is a
new line.

I have put in on my (not empty) todo list

JvanKatwijk, 2023-01-25:
I cannot read the file, software complains that it is no "wav" file

JvanKatwijk, 2023-01-25:
Sorry, a Fedora issue. I can read the file on Ubuntu

andimik, 2023-09-23:
Good news, I found the root cause why this happens.

This is a video how it looks like. If you cannot see it, open it in github.

[Bildschirmaufzeichnung vom 23.09.2023, 13:28:36.webm](https://github.com/JvanKatwijk/qt-dab/assets/24510556/5b94cbbc-43c5-48de-978c-e29a61d706e3)

The stream (Blondie / Maria) itself was/is error-free. Here a screenshot from XpadXpert (https://www.basicmaster.de/xpadxpert/ )
[-Radio-ONE-------Sa--Sept--23-13-28-21-2023.zip](https://github.com/JvanKatwijk/qt-dab/files/12706321/-Radio-ONE-------Sa--Sept--23-13-28-21-2023.zip)

[image]

**Then I realized that the XPAD was divided into subfields. Qt-DAB will show one of the subfield segment instead of the complete DLS. So you should check how many segments does the current SLS have.**

[image]

BTW: this SLS (which is in the recorded file as well) was never shown in Qt-DAB (= also not shown in the video)

[image]

JvanKatwijk, 2023-09-23:
Looking at the pad handling for local stations, and adding some print
statements in the pad handler, I see all pad fragments appearing in the
text,
the dutch TV uses variable pad fragments
May be you can send me some raw input then I can see  what pad fragments
are coming in and how they are handled

andimik, 2023-09-23:
Here you are.

[https://we.tl/t-HhEE4CaRUR](https://we.tl/t-HhEE4CaRUR?utm_campaign=TRN_TDL_05&utm_source=sendgrid&utm_medium=email&trk=TRN_TDL_05)
(400 MB)

Expectation on Radio ONE: SLS to be displayed and DLS is not split into the fragments.

> Sie hören: FREDDIE MERCURY - LIVING ON MY OWN auf Radio ONE

See attached screenshot from AbracaDABra.

[image]

andimik, 2024-01-01:
This is how it looks like in 6.40, see screenshot.

[image]

JvanKatwijk, 2024-01-01:
> This is how it looks like in 6.40, see screenshot.
>
> grafik.png (view on web)

Dear Andreas

First, best wishes for - again - a new year, 2024.
Right now I m fighting with Windows (trying to make a 64 bit version). As
soon as that is done, I'll have another look at the dls texts being
partial,
apparently I am missing something there. Your picture is absolutely
convincing

All the best
jan

closed 2024-01-20 by JvanKatwijk