JvanKatwijk/qt-dab #287: No tags for recent releases

michaellass, 2023-04-24:
I noticed that a `setup-qt-dab32-4.6.exe` was added to the release entry "Qt-DAB versions 4 and 5". However, there is no corresponding git tag. Also, there seem to be no tags for the 5.x versions. The source code download for the git release entry points to a `qt-dab-45.tar.gz`.

I'm maintaining a qt-dab package for Arch Linux (https://aur.archlinux.org/packages/qt-dab) but I'm not sure how to build version 4.6 from source. Could you please tag the corresponding commits or point me to the one that was used for the 4.6 release?

JvanKatwijk, 2023-04-25:
I know I am lazy, and I am not creating new versions for simple error
repairs and functional additions.
I realize that it might have been time for a new relarease since I changed
version numbers
But, now  I created a new git version with the latest 4.6 and 5.3 versions
best
jan

michaellass, 2023-04-26:
Please don't worry, having no release tags is kind of a luxury problem. Thanks a lot for taking care of this so quickly. For Arch Linux I now packaged the new tag as version 4.6 and it works great! :slightly_smiling_face:

closed 2023-04-26 by michaellass

JvanKatwijk, 2023-04-26:
Please note that today I made already two changes again, one of them
interesting. There was/is an apparently new
british ensemble where thhe service names - as transkitted in the FIG's -
are not padded to 16 characters.
The most  recent change addressed that (well, basically just adding a
"lengtning" operation

best
jan