JvanKatwijk/qt-dab #291: Minor problem in calculation the modulation error?

tomneda, 2023-05-28:
I wonder how you calculate the modulation error (or quality) in the `ofdmDecoder` method `computeQuality()`. While debugging I saw that the input vector `conjVector` has (nearly) zero-valued elements which should not be (imo). I could solve the issue to fill the vector `conjVector` not indexed with `index`
`conjVector [index] = r1;`
but with `i`
`conjVector [i] = r1;`
means using the FFT-bin-frequency  mapping to use only the relevant FFT bins. This means also that the size of `conjVector` can reduced from `T_u` (2048) to `carriers` (1536).

The call to `computeQuality()` needs no change but the call to
`iqBuffer	-> putDataIntoBuffer (&conjVector [T_u / 2 - carriers / 2], carriers);`
has to changed to
`iqBuffer	-> putDataIntoBuffer (conjVector, carriers);`

JvanKatwijk, 2023-05-29:
The computation of the modulation ??error??quality was based on Tr 101 290.
However that does not feel good since
the computation there is based on coherent decoding. It works OK for my DRM
decoders, since there decoding is coherent
so you can relate the received data to the ideal form of that data.
Some time ago, I therefore went back to the original computation: computing
the standard deviation of the phase(differences), since
only the phase(difefrence) is used in decoding.
The label on the spectrum widget was changed accordingly

tomneda, 2023-05-31:
Thanks for explaining your way of calculating the "modulation error".

I looked into your changes in the current commit head https://github.com/JvanKatwijk/qt-dab/commits/master . The intention is clear for me but I still see the problem I described in my opening comment.
The provided input vector `v` in `computeQuality()` is a kind of mapping of the FFT values, but not only the usefull bins of that but also with the outside band bins.

I can see in the debugger that 
`v[0]` is "filled" with zero (or better "not assigned DC value").
`v[1..768]` has values (positive frequency side from >0...768kHz).
`v[769..1279]` are "filled" with zeros (or better "not assigned") for the not used highest frequency parts (positive and negative).
`v[1280..2047] `has values (negative frequency side <0 ... -768kHz).

Then you fill the `XX[]` from `v[]` array but only with the 1536 carrier elements and not the rest (I guess it should at least go to T_u).
After that you access XX[] with index `[T_u / 2 - carriers / 2 + i]` with` i = [0...1535]`, so the index would be in the range
`1024-768 ... 1024-768+1535 = 256 ... 1791`

Is it really intended using the values `XX[256 ... 1791]` for calculation where many elements are zero in the middle and other values are not used?

For me, the easiest way is only provide an input vector v[] which only provides the 1536 used carrier bins like I described  in my opening comment.

closed 2023-05-31 by JvanKatwijk