JvanKatwijk/qt-dab #292: Retrieving Git hash for copyright info is not working (with CMake)

tomneda, 2023-05-28:
In CMakeLists.txt (V4 and V5) is this line defined:
`IF(EXISTS (".git") AND IS_DIRECTORY (".git"))`
but EXISTS is not working with relative paths (at least here on my side).

I changed it to 
`IF(EXISTS "${CMAKE_CURRENT_SOURCE_DIR}/../.git" AND IS_DIRECTORY "${CMAKE_CURRENT_SOURCE_DIR}/../.git")`
and this is working on my side. The copyright information shows the latest git hash now.

JvanKatwijk, 2023-05-29:
fixed. It dates from the time there was a single version of qt-dab. I do
not understand why people are using CMake
for this project, qmake is much, much easier to handle and to comprehend

tomneda, 2023-05-31:
It works!

For me it is clear preferring CMake over QMake: CMake is nearly used everywhere. We used it also in my company for very different things. I can build project files for different compilers and OS. QMake is maybe easier to handle but I guess intended for QT projects only. I see CMake as a kind of industry de-facto-standard and the effort to learn such a tool I will better use my time learning CMake instead of learning QMake.

I guess you were already be familiar with QMake so it is of course easier to use that.

closed 2023-05-31 by tomneda