JvanKatwijk/qt-dab #296: Record ETI

awingender, 2023-08-08:
I have been attempting to record an ETI file of live DAB+ transmissions using QT-DAB 5.3.
I am able to see that an ETI file is generated, however when I attempt to decode or transmit it, I receive errors about missing FIC information. 

Specifically, I am using DABMOD from Open Digital Radio to broadcast ETI files, but believe the error to be with QT-DAB. 
I get the following error from dabmod:  "Exception Caught: FIC must be present to modulate!"

I have 2 devices with same results (HackRF and RTL-SDR).  Both devices are able to scan, receive, and decode the live DAB+ signal.
Currently using Windows 10.

JvanKatwijk, 2023-08-09:
I recorded a few minutes of a local ensemble and - played it without any
problem on an eti interpreter, so that does not
give help in interpreting.
Does the resulting eti file play with e.g. dablin?

andimik, 2023-08-09:
@awingender

Can you play that file in `dablin` resp. `dablin_gtk`

What about re-transmit it with `odr-dabmod`?

Tried to load it into `XPADxpert`?

https://www.basicmaster.de/xpadxpert/

https://github.com/opendigitalradio/dablin

https://github.com/Opendigitalradio/ODR-DabMux

Honesty I have rarely tried that function, because I am creating eti files with `eti-cmdline`

awingender, 2023-08-09:
> Does the resulting eti file play with e.g. dablin?

No.  I get the following error from Dablin when I play my recorded QT-DAB eti files.  This error repeats for each frame: 
"ETIPlayer: ignored ETI frame due to wrong (MST) CRC"       

I have also tested using pre-recorded sample ETI files from WorldDAB org that do play correctly directly from Dablin.

> What about re-transmit it with odr-dabmod?

No.  I cannot retransmit the ETI files that I have made using QT-DAB.   I get this error:
"Exception Caught: FIC must be present to modulate!"

If I use my pre-recorded ETI files from WorldDAB, I am able to successfully transmit using odr-dabmod, and correctly receive DAB radio signal on multiple test radios. 

> Tried to load it into XPADxpert?

XPADxpert gives error when QT-DAB made ETI file is loaded:
"Runtime exception: Wrong FSYNC at frame # 1"

My WorldDAB pre-recorded ETI files, loaded into XPADxpert work as expected.  This is a nice tool.   **Although sometimes it gets stuck at 'extracting all sub-channels' and XPADxpert has to be restarted. 

> Honesty I have rarely tried that function, because I am creating eti files with eti-cmdline

Before today, I had not used eti-tools (eti-cmdline).   I was able to easily get it up and running, pipping my output to dablin_gtk.  Doing this, my live DAB+ transmission fully worked from Dablin_gtk.  How do I use eti-cmdline to record an ETI file?  I tried pipping the output of eti-cmdline directly to a test.eti file, and that did not seem to work. 

------

I have uploaded two ETI files that I recorded using QT-DAB 5.3.   The each 60 second recording was started after I felt the DAB+ signal was correctly being received by QT-DAB. 

https://www.filebig.net/files/ViYNe5mRKx

------

I do have one thought that may be of concern, but hopefully not!
I am located in the USA.  Because of this, I am transmitting my own DAB+ signal using a HackRF and Raspberry Pi, using the working WorldDAB ETI files.  This works 100%.    

My concern is that I am using a seperate PC to then record an ETI file using QT-DAB. (Using a RTLSDR).  Should I be able to record a re-transmission?   QT-DAB is able to play it no problem, just issue with recording ETI.

JvanKatwijk, 2023-08-10:
I have the impression that the problem is caused by a problem in decoding
the FIC by Qt-DAB. Actually I do not know what to do if Qt-DAB rejects the
FIC blocks of a frame.
That is may be a shortcoming of Qt-DAB.
Interesting to read that working with et-cmdline worked fine: the code for
eti-cmdline is the basis for the eti generation in Qt-DAB, the difference
being that FIC/FIB handling in Qt-DAB
is a superset of that in eti-cmdline.

Any suggestion of how to deal with eiti output for frames with an incorrect
FIC is welcome

dh5ym, 2023-08-10:
Hi,
i can confirm the above. ETI recordings created with qt-dab generate messages like
ETIPlayer: ignored ETI frame due to wrong (MST) CRC
EnsembleSource: skipping 14 bytes for sync
in dablin. Recordings done with eti-cmdline can be decoded.
I guess still storing the faulty FIC blocks might be the choice.
regards
Mario

JvanKatwijk, 2023-08-11:
I Might have found something, it seems that the "fibValid" flag is always
set to true in the Qt-DAB version, while in the eti-cmdline version
the setting is derived from the handling of the fic blocks.
It might take a few days before this is handled

JvanKatwijk, 2023-08-11:
Well, I made a quick change in the software, i.e. passing on data on the
correctness of the fics to the part that generated eti.
My question is: do you use windows, or do you build your own executable or
do you use an AppImage?

dh5ym, 2023-08-11:
The friend who supplied me with the recordings was using Windows.
I can only compile on Linux or use Windows on my work PC.

JvanKatwijk, 2023-08-11:
I uploaded a Windows version, with "-test" added to the name, that contains
a few mods, telling the eti-generator the state of the FIC blocks

awingender, 2023-08-11:
> I uploaded a Windows version, with "-test" added to the name, that contains
> a few mods, telling the eti-generator the state of the FIC blocks

I just tested [setup-qt-dab32-5.4-test.exe](https://github.com/JvanKatwijk/qt-dab/releases/download/qt-dab-4.7%2F5.4/setup-qt-dab32-5.4-test.exe) and unfortunately received the same FSYNC errors from xpadxpert. 
"Runtime Exception: Wrong FSYNC at frame # 1"

> My question is: do you use windows, or do you build your own executable or
> do you use an AppImage?

For my use case, Windows is the most helpful.  

Just a follow-up of a previous question I had above:
I have since been successfully in getting eti-cmdline to output a working ETI file of my DAB 're-broadcast'.  
"eti-cmdline-rtlsdr -C 7A -G 20.7 > dump.eti"

JvanKatwijk, 2023-08-11:
Just for my understanding: the result of processing with eti-cmdline does
not seem to contain an error, while the result of processing with Qt-DAB
gives an error?
If so, then I have to compare more closely the differences between the
processing part in the two programs

dh5ym, 2023-08-11:
Hi,

maybe its a problem specific to the Windows version. I just tried with the
64bit Appimage version and 5 different ensembles. Each of the recordings
(even the one with bad SNR) was playing fine with dablin.
I cannot test with the windows version myself.

regards
Mario

awingender, 2023-08-11:
> eti-cmdline does not seem to contain an error, while the result of processing with Qt-DAB gives an error?

Yes, but my testing of QT-DAB has been exclusively on Windows10.
eti-cmdline has been used on raspberry pi (Raspbian 11 - Bullseye) 

I will try the Appimage version of QT-DAB on RPi and report back

JvanKatwijk, 2023-08-12:
I do not know and I do not understand
The relevant code for telling the ETI stream the status of the FIC block
i.e. true, or false, is now the same for Qt-DAB and eti-cmdline
(as made available in the Window version. (a flag is set to true if the crc
check for the fic block succeeds and to false otherwise.)

dh5ym, 2023-08-12:
I can just have a look into the faulty ETI files with some different analysis tool. Maybe this gives some more insight. The fact that it seems to work under Linux suggests that there is some problem specific to the Windows compilation. No clue what this could be.

awingender, 2023-08-14:
> The fact that it seems to work under Linux suggests that there is some problem specific to the Windows compilation

I agree that the problem seems to be Windows specific.
I tested recording ETI file with QT-DAB 5.3 & 5.4 in linux, and the resulting ETI file works correctly.  

I next used [ETIsnoop](https://github.com/Opendigitalradio/etisnoop) from OpenDigitalRadio. 
From what I'm seeing.  FSYNC and CRC are correct at the beginning of FRAME#0, but by FRAME#1, both FSYNC and CRC are wrong. 

https://imgur.com/a/1WRP57k
See image above.   I compared the output of ETISnoop of a known good vs bad ETI file.
For a good ETI file:  I see that all TPLs say EEP "Equal Error Protection"
For a bad ETI file:   I see that some TPLs say UEP "Unequal Error Protection"

For a good ETI file:  After EOF, i see the CRC shows "Value: No Error"
For a bad ETI file:  After EOF, i see the CRC shows "Value: Mismatch: 2da8"

For a good ETI file:  FYSYNC shows "Value: OK"
For a bad ETI file:   FYSNC shows "Value: Wrong FYSNC"

JvanKatwijk, 2023-08-15:
I might have found something: I did not specify the output file to be
written as a binary file.
I uploaded a version (regular 5.4) wihere the file is explcitly opened as a
file for binary output

dh5ym, 2023-08-15:
It seems this did the job. With the 5.4 Windows binary i was able to record a ETI from DR Deutschland and read it into other tooling.
@awingender: can you confirm that?

awingender, 2023-08-15:
@JvanKatwijk  @dh5ym 

That works!!!

I tested recording an ETI file using latest QT-DAB 5.4 on Windows 10.
Dablin was successfully able to playback the recorded file. 
I also inspected the new ETI file with ETIsnoop, and the 3 previous 'errors' i mentions about FSYNC, CRC, and protection levels are all now gone.  

I want to thank both of you for your help.  Your work has and will continue to indirectly/directly help many many people get a better DAB experience in the future.  The ability to record an ETI from Windows, using a low-cost RTLSDR will help many people I work with troubleshoot radio issues we are investigating far from our offices.   Again, thank you for this great tool.

closed 2023-08-15 by awingender

JvanKatwijk, 2023-08-15:
Feel free to ask if you have further questions about qt-dab

awingender, 2023-08-17:
Is it possible that an image/artwork could be received by multiple vehicle radios, but not stored in the recorded ETI?
More specifically, I know MOT can be received and recorded by QT-DAB  (Tested and verified by me today), however **could there be other multimedia/image methods not supported by QT-DAB**.  For example, SLS, or slideshow?   

My scenario.  I had someone in South UK record an ETI for me this morning using QT-DAB 5.4. I received the file of the BBC Ensemble, but it does not show any image data at all when inspected with XPADxpert.  When I myself try to transmit the ETI file, it works completely, except I see no image/artwork from my radio.   So to me, it seems the image data was not recorded by QT-DAB.  The person recording ETI confirmed they could see images on 2 test radios at same time as recording.   Radios are only DAB, not IP/internet connected at all.   

Forgive me if this goes outside the scope of QT-DAB.

dh5ym, 2023-08-17:
I would assume that QT-DAB only supports XPAD MOT. SPI is very unlikely. But its possible that something is not assembled in the way it should be. I have not enough knowledge about the details of ETI to judge that but would not assume that PAD data is lost during recording. If i have enough time i can try to look into the file with some tooling tomorrow and watch out for errors.

awingender, 2023-08-18:
I think @dh5ym is correct or very close

It appears that the artwork is not being transmitted using the traditional MOT method within the XPAD.  Instead, the [BBC National DAB ensemble](https://en.wikipedia.org/wiki/BBC_National_DAB) seems to be transmitting a data service called "BBC Guide, Service ID: E1C79E5E." which contains both EPG (aka [SPI](https://www.worlddab.org/dab/data-applications/service-and-programme-information)) and the MOT data.

https://imgur.com/a/0RfnmzN

> So, the obvious question is: how is the data transmitted, as separate
> service or embedded as MOT in OAD data

It appears that the images are part of a separate data service rather than being embedded within the XPAD as MOT data.  This might explain that the absence of image data in the ETI file. 

From WorldDAB, I have a known-good ETI file from Germany with similar SPI type images.
When I test broadcasting the German ETI file, QT-DAB was able to display the images OK.  (The BBC images do not show in QT-DAB)
I next recorded an ETI of this SPI broadcast, and rebroadcasted that ETI itself again.  Images recorded and played correctly again.  
So it seems QT-DAB can work

JvanKatwijk, 2023-08-19:
I have limited experience with the EPG handling. The NPO (dutch radio and
TV) just sends empty stuff.
If and EPG service is part of the ensemble, it sends it content using a
subchannel. As said, the content of each of the subchannels (is just fed to
the ETI processor - it does the
deconvoluton - and then added - in the right place into the ETI file.
As far as I can see, there is not further processing or interpretation,
only deconvolution, so it seems to me that if such a service is part of the
input,
its payload should end up in the ETI file.

Another thing is that in the "normal" EPG processing as done by Qt-DAB
there is no SPI specific processing, basically because - as I said above -
the
NPO does not give me that.

I would appreciate getting a recording from an ensemble with an EPG channel
containing SPI data, that would help me im proving QT-DAB

best
jan

JvanKatwijk, 2023-08-23:
> Here is the German ETI file that I have. It was not recorded in QT-DAB
> from what I can tell, but seems to contain SPI data, as well as traditional
> MOT. When broadcasted, the images in this ETI file do work with QT-DAB,
> unlike the ETI file from UK.
> http://www.filebig.net/files/sR4tyVgP46
>
> This file appears to contain 5 subchannels of tdc & mot data:
> Content Details

Thanks
Right now I am traveling for a couple of weeks, will pick this subject up
on return

dh5ym, 2023-08-23:
In Germany almost all ensembles use MOT/SLS via XPAD. Some SPI tests with
dedicated data services emerge the recent years but as far as i know are
mainly used to transfer station logos (also for stations outside the
ensemble e.g. FM stations).