JvanKatwijk/qt-dab #299: TII-CODES folder not found

andimik, 2023-09-22:
During qmake I get the following error

```
$ qmake ..
Info: creating stash file /home/andreas/apps/qt-dab/qt-dab-s6/build/.qmake.stash
Project MESSAGE: Current git hash = 6f3224c
WARNING: Failure to find: ../../TII-CODES/tii-codes.cpp
WARNING: Failure to find: ../../TII-CODES/tii-codes.h
```
which comes from the `pro` files:

```
#	INCLUDEPATH	+= /home/jan/curl
	INCLUDEPATH	+= ../../TII-CODES
	HEADERS		+= ../../TII-CODES/tii-codes.h
	SOURCES		+= ../../TII-CODES/tii-codes.cpp
}

tiiLib	{
	INCLUDEPATH	+= ../src/support/tii-library
	HEADERS		+= ../src/support/tii-library/tii-codes.h
	SOURCES		+= ../src/support/tii-library/tii-codes.cpp
}
```
Seems this is your own local directory.

When I comment all three TII-CODES lines, it at least go till 

```
In file included from ../main.cpp:33:
../radio.h:44:17: fatal error: tii-codes.h: Datei oder Verzeichnis nicht gefunden
   44 | #include        "tii-codes.h"
      |                 ^~~~~~~~~~~~~
compilation terminated.
```

I need to copy the two `tii-codes` files from the library to the `qt-dab-s6` folder in order **nearly** to compile. It will stop right at the end

```
/usr/bin/ld: /tmp/ccpt7X6e.ltrans0.ltrans.o: in function `main':
/home/andreas/apps/qt-dab/qt-dab-s6/build/../radio.cpp:226: undefined reference to `tiiHandler::tiiHandler()'
/usr/bin/ld: /tmp/ccpt7X6e.ltrans0.ltrans.o:/home/andreas/apps/qt-dab/qt-dab-s6/build/../radio.cpp:499: undefined reference to `tiiHandler::tiiFile(QString const&)'
/usr/bin/ld: /tmp/ccpt7X6e.ltrans0.ltrans.o:/home/andreas/apps/qt-dab/qt-dab-s6/build/../radio.cpp:652: undefined reference to `tiiHandler::~tiiHandler()'
/usr/bin/ld: /tmp/ccpt7X6e.ltrans1.ltrans.o: in function `RadioInterface::~RadioInterface()':
/home/andreas/apps/qt-dab/qt-dab-s6/build/../radio.cpp:745: undefined reference to `tiiHandler::~tiiHandler()'
/usr/bin/ld: /tmp/ccpt7X6e.ltrans4.ltrans.o: in function `RadioInterface::loadTable()':
/home/andreas/apps/qt-dab/qt-dab-s6/build/../radio.cpp:4092: undefined reference to `tiiHandler::loadTable(QString const&)'
/usr/bin/ld: /home/andreas/apps/qt-dab/qt-dab-s6/build/../radio.cpp:4093: undefined reference to `tiiHandler::valid()'
/usr/bin/ld: /home/andreas/apps/qt-dab/qt-dab-s6/build/../radio.cpp:4096: undefined reference to `tiiHandler::tiiFile(QString const&)'
/usr/bin/ld: /tmp/ccpt7X6e.ltrans4.ltrans.o: in function `RadioInterface::show_tii(int, int)':
/home/andreas/apps/qt-dab/qt-dab-s6/build/../radio.cpp:4407: undefined reference to `tiiHandler::is_black(unsigned short, unsigned char, unsigned char)'
/usr/bin/ld: /home/andreas/apps/qt-dab/qt-dab-s6/build/../radio.cpp:4416: undefined reference to `tiiHandler::get_transmitterName(QString const&, unsigned short, unsigned char, unsigned char)'
/usr/bin/ld: /home/andreas/apps/qt-dab/qt-dab-s6/build/../radio.cpp:4427: undefined reference to `tiiHandler::get_coordinates(float*, float*, float*, QString const&, QString const&)'
/usr/bin/ld: /home/andreas/apps/qt-dab/qt-dab-s6/build/../radio.cpp:4447: undefined reference to `tiiHandler::distance(float, float, float, float)'
/usr/bin/ld: /home/andreas/apps/qt-dab/qt-dab-s6/build/../radio.cpp:4451: undefined reference to `tiiHandler::corner(float, float, float, float)'
/usr/bin/ld: /home/andreas/apps/qt-dab/qt-dab-s6/build/../radio.cpp:4418: undefined reference to `tiiHandler::set_black(unsigned short, unsigned char, unsigned char)'
collect2: error: ld returned 1 exit status
make: *** [Makefile:733: linux-bin/qt-dab-6.0] Fehler 1
```

Then I changed it to `#CONFIG		+= preCompiled` and it still won't compile.

What am I doing wrong?

andimik, 2023-09-22:
I see, seems the line

```
CONFIG		+= tiiLib
``` 
is commented out by default, which causes the error. What is the purpose of have that switched off when it won't compile?

JvanKatwijk, 2023-09-22:
That is absolutely true. It follows from the condition on making an URL
available for the TII database bu Gunther Lorenz.
He asked me to take measures such that - since all sources of Qt-DAB are
visible - the URL would be invisible.
I "solved" it by putting the TII library into a separate file that is
available from the Qt-DAB distribution and creating a
library - that is loaded in dynamically - for Qt-DAB to load and inspect
THAT encoded library.

For precompiled versions, i.e. Windows installers and AppImages I aways
included a set of functions that can both handle the
encoded library and can create such a library by loading a file from
Gunthers URL

The regular option is to comment out "preCompiled" and uncomment "tiiLib".

Since I believe you are a reliable person, I can send you the files in
TII-CODES, obviously with the request not to distribute them further.

Best jan

closed 2023-12-22 by JvanKatwijk