JvanKatwijk/qt-dab #300: SLS are mixed up with SPI pictures

andimik, 2023-09-22:
There is a bug in handling backend. It will show a picture from backend (SPI, EPG service) at the actual service.

If you cannot see the three screenshots, please open it on github.

[image]

In the screenshot you see private station Radio 1, but the "SLS" is not their slideshow, but the backend picture from public service PRVI, see following log:

```
we have now 2 backends running
going to write file /tmp/Qt-DAB-files/357
going to write file /tmp/Qt-DAB-files/e4_9201_32x32_0.png
going to write file /tmp/Qt-DAB-files/e4_9202_32x32_0.png
going to write file /tmp/Qt-DAB-files/e4_9201_112x32_1.png
going to write file /tmp/Qt-DAB-files/e4_9202_112x32_1.png
```

[image]

This will happen when the SLS of the service has not displayed yet AND a picture from backend has been decoded.

After some seconds the correct SLS will be shown:

[image]

JvanKatwijk, 2023-09-22:
If I understand it correctly, then a slide, decoded in the EPG service is
being displayed (and later overruled as soon the current servide sends its
picture?

andimik, 2023-09-22:
yes, EPG pictures are shown in a radio program.

JvanKatwijk, 2023-09-22:
It can be solved in a pretty transparent way. It will require minor mods to
the backend handling and will therefore impact
versions 4, 5, and 6.
I probably need the weekend to handle it

JvanKatwijk, 2023-09-23:
should be fixed now in the sources in github, for v4, v5 and v6

andimik, 2023-09-23:
Good, the EPG pictures are now NOT showing anymore in the audio service.

But there is still a bug. Qt-DAB won't show the own SLS as it does not update this information. It seems that it crashed internally when received garbage (but this is not related to reception, I just opened the frame dump recording window).

In the attached aac file you see DL+ and SLS

[Radio-1-80-a-----Sa--Sept--23-13-11-32-2023.zip](https://github.com/JvanKatwijk/qt-dab/files/12706299/Radio-1-80-a-----Sa--Sept--23-13-11-32-2023.zip)
[image]
[image]

But Qt-DAB did never show it. Please also see "NG BY MYSELF" which was part of the garbage frame. See other (still not solved issue) https://github.com/JvanKatwijk/qt-dab/issues/280

[image]

closed 2023-12-22 by JvanKatwijk