JvanKatwijk/qt-dab #301: 6.1 does not build with gcc-13.2.1

stefantalpalaru, 2023-09-29:
```text
In file included from /var/tmp/portage/media-radio/qt-dab-6.1/work/qt-dab-qt-dab-6.1/qt-dab-s6/../includes/support/http-handler.h:34,
                 from /var/tmp/portage/media-radio/qt-dab-6.1/work/qt-dab-qt-dab-6.1/qt-dab-s6/radio.h:62,
                 from /var/tmp/portage/media-radio/qt-dab-6.1/work/qt-dab-qt-dab-6.1/src/backend/data/tdc-datahandler.cpp:24:
/var/tmp/portage/media-radio/qt-dab-6.1/work/qt-dab-qt-dab-6.1/qt-dab-s6/../src/support/tii-library/tii-codes.h: In constructor ‘position::position(float, float)’:
/var/tmp/portage/media-radio/qt-dab-6.1/work/qt-dab-qt-dab-6.1/qt-dab-s6/../src/support/tii-library/tii-codes.h:34:12: error: ‘longitude’ was not declared in this scope; did you mean ‘longiture’?
```

6.0 also failed to build on my system, so I skipped packaging it. How about setting up GitHub Actions, so you can see these problems quicker?

JvanKatwijk, 2023-09-29:
This was actually a type in the file /src/support/tii-library/tii-codes.h
the longiture -> longitude

andimik, 2023-09-30:
Jan, where can I define the position? I could build 6.0, but not 6.1

```
In file included from ../radio.h:62,
                 from ../main.cpp:33:
../../includes/support/http-handler.h:39:9: error: ‘position’ does not name a type
   39 |         position coords;
      |         ^~~~~~~~
../../includes/support/http-handler.h:55:34: error: ‘position’ has not been declared
   55 |                                  position       address,
      |                                  ^~~~~~~~
../../includes/support/http-handler.h:63:34: error: ‘position’ has not been declared
   63 |                                  position       target,
      |                                  ^~~~~~~~
../../includes/support/http-handler.h:74:9: error: ‘position’ does not name a type
   74 |         position                homeAddress;
      |         ^~~~~~~~
../../includes/support/http-handler.h:84:42: error: ‘position’ has not been declared
   84 |         std::string     theMap          (position address);
      |                                          ^~~~~~~~
In file included from ../main.cpp:33:
../radio.h:132:9: error: ‘position’ does not name a type
  132 |         position        localPos;
      |         ^~~~~~~~
../radio.h:133:9: error: ‘position’ does not name a type
  133 |         position        targetPos;
      |         ^~~~~~~~
../radio.h: In member function ‘void channelDescriptor::cleanChannel()’:
../radio.h:144:9: error: ‘targetPos’ was not declared in this scope; did you mean ‘fgetpos’?
  144 |         targetPos       = position {0, 0};
      |         ^~~~~~~~~
      |         fgetpos
../radio.h:144:27: error: ‘position’ was not declared in this scope
  144 |         targetPos       = position {0, 0};
```

JvanKatwijk, 2023-09-30:
in src/support/tii-library/tii-codes it reads longiture should be longitude

andimik, 2023-09-30:
No, you already have fixed that in https://github.com/JvanKatwijk/qt-dab/commit/c04750429475fef7c66e39c66a2d48a3b3757873

The error comes from another bug.

JvanKatwijk, 2023-09-30:
> No, you already have fixed that in c047504
>
> The error comes from another bug.

geep apologies. It turned out I updated the code in one of my VM's, but -
since I was taling to someone at the same time - did not update the
development version
It compiles without problems now

wkazubski, 2023-10-12:
> It compiles without problems now

Bugfix release?
Version 6.0 does not build well for me also

JvanKatwijk, 2023-10-12:
It was fixed the same day, and for me there was no need to renumber

stefantalpalaru, 2023-10-16:
Fixed in 6.2.

closed 2023-10-16 by stefantalpalaru