JvanKatwijk/qt-dab #302: Spectrum and waterfall diagrams are not on same vertical position

andimik, 2023-09-30:
Jan, I would be great, when the vertical position for the waterfall is the same like in the spectrum. Currently it's a bit shifted to the left.

See screenshot.

[image]

JvanKatwijk, 2023-10-01:
I changed it, the 4 large elements in the widget are now formatted as "grid"

andimik, 2023-11-07:
No, this did not help, it's still not in the same vertical position:

[image]
(see screenshot, taken with 6.25)

Do you really need the label on the 2nd axis?

JvanKatwijk, 2023-11-07:
> I'll have a look tomorrow

That does not solve the problem completely, but would be a reasonable
approximation

closed 2024-01-20 by JvanKatwijk