JvanKatwijk/qt-dab #304: Opening rtlsdr device failed & setting samplerate failed

monsieurcoyote, 2023-10-17:
Good afternoon,
I installed - succesfully - QTab 6.2 ( setup-qt-dab32-6.2.exe ) on my Windows 10 X64 PC.
However, I just can't get it running with any of my 2 USB dongles (maybe not the best buy, but they work on a Windows 7 PC).
I already have Zadig 2.7 installed and of course restarted the PC.
Is there something you can advice to get it running?
Error with the white stick (Astrometa DVB-T/DVB-T2) is 'Opening rtlsdr device failed'
Error with the blue stick (Detected as Generic RTL2832U OEM) is 'Setting samplerate failed'

Thanks,
wile_e

JvanKatwijk, 2023-10-26:
Do you have by any chance a V5 stick? Last week I included a library that can handle V4 sticks in the disribution

monsieurcoyote, 2023-10-27:
The stick I would like tu use is :
Astrometa DVB-T2 device (DVB-T/T2/C+FM+DAB)
Driver date : 27-04-2017
Version 64.6.427.2017
VID_15F4&PID_0131&MI_00\7&5589535&080000
It looks like this : https://nl.aliexpress.com/item/1005005262577441.html

The other stick I have is 
REALTEK 2832U device (DVB-T+FM+DAB) (820T2 & SDR)
Driver date : 30-07-2015
Version : 64.1.730.2015
Digital signing : Astrometa Co. Ltd
oem62.inf:4777d1a3549092c6:RTLUSBDEV:64.1.730.2015:usb\vid_0bda&pid_2838&mi_00
It looks like this : https://nl.aliexpress.com/item/1005004251615318.html?gatewayAdapt=glo2nld
This second stick works well in my Edision Mio+ 4K for DVB-T (but does not decode DVB-T2).

monsieurcoyote, 2023-10-29:
I downgraded the driver 64.6.427.2017 to the version 64.1.801.2012 which is supposed to work on Windows 10. Still no result.
It doesn't work either with dab player or welle.io

closed 2024-01-20 by JvanKatwijk