JvanKatwijk/qt-dab #305: 6.2 not working with RTL-SDR v4

0kti, 2023-10-20:
Even when changed drivers

JvanKatwijk, 2023-10-20:
> newest RTL-SDR v4

well, then it is a driver's issue,

0kti, 2023-10-20:
Sure, I've changed. Newest drivers are for download on github, but still not working :(

JvanKatwijk, 2023-10-20:
did you try the one from https://www.rtl-sdr.com/v4/
Until now I added the regular driver from osmocom in the distro, but from
the description in https://www.rtl-sdr.com/v4/
I read that a special one for V4 is needed

0kti, 2023-10-20:
Yes, I've downloaded latest release from https://github.com/rtlsdrblog/rtl-sdr-blog/releases
With SDRsharp works fine but with qt-dab not. I don't know why
I've try both x86 drivers and x64 just to be sure.

JvanKatwijk, 2023-10-20:
Strange, that driver is supposed to be backwards compatible with the older
drivers.
I'll have a look into the sources of that driver tomorrow

JvanKatwijk, 2023-10-21:
I uploaded a version of 6.2 with a library for V4 (at least that is what
they say)
It turned out that Qt-DAB used in the windows version a separate driver for
rtlsdr sticks
(basically because they always give trouble) that differs from the Linux
version and the
driver library was linked directly to it rather than being read in as is
common for the drivers.

Anyway, with my "old" sticks it works fine, may be it works for V4 sticks

best
jan

WIHDIA, 2023-10-21:
Hi guys,

Latest 6.2 is now working well with RTL V4 and also still with V3! 

73
Herman
[My logblog](https://wijnants.info/fmdx/)
Belgium editor of the [fmlist.org](https://www.fmlist.org)

JvanKatwijk, 2023-10-21:
Ja, het was even pizzelen

0kti, 2023-10-21:
Tested new version but still not woking
Tested also wello.io and there dab+ after copy new drivers is working fine

JvanKatwijk, 2023-10-21:
I can only test with "old" sticks, but according to Herman V4 works, so I
am a little puzzled
Does the library load?
is a device recognized?
or do not you see a device at all

0kti, 2023-10-21:
I hope you understand me, my english is not very good.
QT-DAB starts fine
looks like driver is loaded correctly, but searching channels (scan) finding nothing, not even signal

JvanKatwijk, 2023-10-21:
No problem with your language
So, it seems the driver software is loading, do you see the rtlsdr widget?
what happens if you change the gain or agc

0kti, 2023-10-22:
I'e uninstalled previous 6.2, cleaned everything from folder
Clean registry
Installed again nevest 6.2
Now working fine
Please close this issue

JvanKatwijk, 2023-10-22:
Anyway, good to hear!!

closed 2023-10-22 by 0kti